Koenkk/zigbee2mqtt #25754: Lidl HG08673 not exposing power anymore

Paraman2016, 2025-01-11:
### What happened?

I recently started using a Docker Compose setup for Home Assistant, Zigbee2MQTT and the Eclipse-Mosquitto broker. Since then, I no longer receive any power values from Lidl HG08673. The value is always 0. I actually use the value for some automations and I haven't found a way to fix it yet. I have already re-taught the plug several times. Unfortunately, that hasn't helped either.

### Zigbee2MQTT version

2.0.0

### Adapter firmware version

8.0.2 [GA]

### Adapter

Sonoff ZBDongle-E (V2 model)

### Setup

Containerized on PI4

sammyke007, 2025-01-17:
I can confirm this. 
Devices added before 2.0.0 report power without a problem, but the same devices added after 2.0.0 only report power 0 / energy 0 and current 0. Voltage reports correctly +- 230v.

Anything I can share?

Koenkk, 2025-01-17:
Can you try connecting a load and reading the active power?

[image]

sammyke007, 2025-01-17:
> Can you try connecting a load and reading the active power?
> 
> [image]

I have tried that and it kept reporting 0. I also tried a force removal + a new pair but with the same result. 

I've force removed it again, brought it close to the coordinator and suddenly it started reporting power, even after replugging it in the first location.

I think I saw some warning while pairing the first times with something about a zigbee version. Is that possible?

It's working now however. I can test some more tomorrow with a spare plug if you like.

JimGermany, 2025-01-26:
I can confirm that. I have 16 Tuya OEM Plugs from various manufacturers (Lidl, Aubess, Neo ...) and I use a Lidl HG08673 for two of the three Nest Hubs. The third Nest Hub is connected via an Aubess Tuya OEM Plug. The power value for the two Lidl HG08673s keeps dropping out and 0 watts is measured, 

[image]

whereas the Aubess Tuya OEM Plug consistently delivers the power value.

[image]

I can rule out and delete a problem with my Zigbee mesh setup, and unfortunately re-learning/pairing the Lidl plugs didn't help either.

I'm not 100% sure, but I think the problem with the Lidl plugs didn't exist with Z2M 1.4.x.

sammyke007, 2025-01-26:
> I can confirm that. I have 16 Tuya OEM Plugs from various manufacturers (Lidl, Aubess, Neo ...) and I use a Lidl HG08673 for two of three Nest Hubs. The third Nest Hub is connected via an Aubess Tuya OEM Plug. The power value for the two Lidl HG08673s keeps dropping out and 0 watts is measured, 
> 
> [image]
> 
> whereas the Aubess Tuya OEM Plug consistently delivers the power value.
> 
> [image]
> 
> I can rule out and delete a problem with my Zigbee mesh setup, and unfortunately re-learning/pairing the Lidl plugs didn't help either.
> 
> I'm not 100% sure, but I think the problem with the Lidl plugs didn't exist with Z2M 1.4.x.

No it never happened pre 2.0.0 indeed. Try force removing the device and pair it very close to your coordinator. That did it for me.

JimGermany, 2025-01-27:
> Try force removing the device and pair it very close to your coordinator. That did it for me.

Unfortunately, re-pairing directly via the coordinator (SLZB-06) does not solve the problem either. I had already tested that and then did it again this morning (9:57 a.m.). So remove the Lidl Plug from Z2M (force removal) and then re-pair directly via the coordinator. The problem then persists: the power value is measured for a few minutes and then 0 watts is "measured" again.

[image]

FYI: Z2M Map

[image]

Theoretically, it could be that the Lidl Plug has problems measuring low power values ​​(< 5 watts) correctly and therefore keeps "measuring" 0 watts. I will now test the Lidl Plug with a consumer that then consumes more watts. Let's see whether the plug then measures the power value continuously, or whether it measures 0 watts again in between.

sammyke007, 2025-01-27:
That's not the issue. Ive tested with heavier loads. Strange thing is that my workaround doesn't work anymore. I've tried again with a new power plug and it keeps returning 0 watt and 0 amps. Current is correct.

JimGermany, 2025-01-27:
As announced, I swapped the plugs around this morning. My (provisional) conclusion is: This is not a specific problem with the Lidl plugs, but other plugs have the same problem. Some plugs cannot measure low power values ​​(~ < 5 watts) correctly and thus 0 watts are always "measured".

The following plugs are used here:
1. Lidl HG08673 - _TZ3000_j1v25l17 - Firmware: No information on this at Z2M
2. Aubess _TZ3000_w0qqde0g - Firmware: 500+TZSKT21Z3102
3. Neo - _TZ3000_w0qqde0g - Firmware: No information on this at Z2M
4. NoName - _TZ3000_5f43h46b - Firmware: 500+TZSKT11RM103

The two Lidl Plugs are now connected to a NUC and a monitor and there is no longer a fallback to 0 watts.

Lidl Plug with NUC
[image]

Lidl Plug with Monitor
[image]

My three Nest Hubs are now connected via: Neo, Aubess and NoName Plug.

Neo Plug = Fall back to 0 watts and thus the same problem as with the Lidl plugs.
[image]

NoName Plug = **No** fall back to 0 watts.
[image]

Aubess = **No** fall back to 0 watts.
[image]

Since I have often heard and read in recent years that some plugs have problems measuring low wattage values, I assume that it is not a Z2M (2.0) problem, but simply something to do with the plugs themselves. This means that there are certainly certain manufacturing/quality variations in all of these OEM parts. I don't know whether the problem can be changed by trying to calibrate the plugs. For me, the topic is closed for now.

sammyke007, 2025-01-27:
I've been testing some more today and the plugs added after v2.0.0 do report power with heavier loads, but they report way too low values. I've been using LIDL plugs long before v2.0.0 and they were always quiet accurate. 
I've bought +- 20 pieces of them and the ones paired after v2.0.0 don't report correctly. Eg. my son is playing on a PS5 + an old LED TV + Asus AP + phone charger and it only reports 21W and a current of 0 Amps. I'm 100% sure that something is wrong.

sammyke007, 2025-01-28:
I've added another "new" LIDL plug that I had in stock (bought together with the current perfectly working ones) and it keeps bugging out.

I notice however that energy accumulates. Power and current stay "0", but the energy value adds up. The only other thing that's reporting correctly is voltage (+- 230V).

I've also tried both before and after OTA update. Doesn't make any difference...

Anything you need Koen for more information? I'm 99,9% sure it's because of v2.0.0...

Adding:
[image]
Before waterboiler:
[image]
After waterboiler:
[image]
Log:
[image]
Default settings:
[image]
[image]

JimGermany, 2025-01-28:
> After waterboiler:

That means even if you connect a device with a higher consumption, the power indicator remains at 0? That's not the case for me. As I wrote yesterday, my three Lidl plugs no longer show a 0 value when a device with a higher power consumption - such as a NUC - is connected to it.

NUC this morning
[image]

NUC Z2M current status
[image]

sammyke007, 2025-01-28:
Yes, it keeps showing 0 W... I do see a difference with your device:

Model HG08673 (the same)
Zigbee Manufacturer _TZ3000**_ynmowqk2** (different)

JimGermany, 2025-01-28:
> Zigbee Manufacturer _TZ3000_ynmowqk2 (different)

Yes, Lidl changed suppliers every now and then, so that the plugs were sold under at least three manufacturer names:

_TZ3000_j1v25l17
_TZ3000_nkcobies
_TZ3000_ynmowqk2

I also know that there have been problems with the power display of the plugs over the last few years and you can find a lot of information about this on Google. Just search for the respective manufacturer name. But Koenkk has actually always solved the problem.

sammyke007, 2025-01-28:
Ok. As I've made clear, I have bought +- 20 of these plugs a year ago. The ones added recently stopped reporting power. I can't seem to find any difference in config between the previous working ones and the newly added ones. I hope Koenkk will be able to find a solution.

sammyke007, 2025-01-28:
@Koenkk , I've asked some AI assistant to compare working an non working database.db entries. They come up with two differences:

[image]

[image]

Changing the 32768 value in  the database.db to 0 manually won't help. It keeps reverting to 1.

JimGermany, 2025-01-28:
> I have bought +- 20 of these plugs a year ago.

OK, I've checked now and so FYI: I got my three Lidl plugs (_TZ3000_j1v25l17) in February 2024. I also know that I did a firmware update at some point back then. Originally, a 7x firmware version was installed on the plugs and I was then offered an OTA update to version 192 via Z2M.

sammyke007, 2025-01-28:
Most of my LIDL plugs are on v192 too, but I have some working perfectly fine on the firmware they came with (v68). So not a firmware thing.

closed 2025-04-06 by github-actions[bot]

bjoernmayer, 2026-02-15:
I am facing the same issue. Did anyone find a solution?