Koenkk/zigbee2mqtt #25756: [Wrong device]: Legrand 067797A

KeryanB, 2025-01-11:
### Link

https://www.legrand.fr/pro/catalogue/interrupteur-filaire-connecte-avec-neutre-avec-option-variateur-celiane-with-netatmo-150w-led-blanc-sans-plaque

### Model

067797A

### Description

Dimmer Switch 

### Vendor

Legrand

### Picture (link)

https://www.legrand.fr/sites/default/files/ecat/th_LG-067797A-WEB-F.jpg

### Database entry

{"id":25,"type":"Router","ieeeAddr":"0x0004740000240863","nwkAddr":15642,"manufId":4129,"manufName":" Legrand\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000","powerSource":"Mains (single phase)","modelId":" Dimmer switch with neutral\u0000\u0000\u0000\u0000","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":256,"inClusterList":[0,3,4,8,6,5,15,64513,769],"outClusterList":[6,0,64513,5,25],"clusters":{"genBasic":{"attributes":{"appVersion":0,"dateCode":" \u000020230310\u0000\u0000\u0000\u0000\u0000","swBuildId":"0018"}},"genOnOff":{"attributes":{"onOff":1}},"genLevelCtrl":{"attributes":{"currentLevel":254,"onLevel":254}},"lightingBallastCfg":{"attributes":{"minLevel":20,"maxLevel":254}},"manuSpecificLegrandDevices":{"attributes":{"0":257,"1":1,"2":0}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x84b4dbfffec9d439","endpointID":1},{"cluster":8,"type":"endpoint","deviceIeeeAddress":"0x84b4dbfffec9d439","endpointID":1}],"configuredReportings":[{"cluster":6,"attrId":0,"minRepIntval":0,"maxRepIntval":65000,"repChange":1},{"cluster":8,"attrId":0,"minRepIntval":10,"maxRepIntval":65000,"repChange":1}],"meta":{}},"242":{"profId":41440,"epId":242,"devId":102,"inClusterList":[33],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":0,"stackVersion":66,"hwVersion":16,"dateCode":" \u000020230310\u0000\u0000\u0000\u0000\u0000","swBuildId":"0018","zclVersion":2,"interviewCompleted":true,"meta":{"configured":332242049},"lastSeen":1736628189117}

### Notes

Hello! 
Fully functional device recognized as https://www.zigbee2mqtt.io/devices/4411C_L4411C_N4411C_NT4411C.html#bticino-4411c%252Fl4411c%252Fn4411c%252Fnt4411c instead of a Legrand Wall Switch :)

marco44, 2025-01-20:
I have the exact same thing with the mosaic version instead of the celiane (i suppose they're the same internally)

chpego, 2025-05-18:
https://github.com/Koenkk/zigbee2mqtt/discussions/27264

closed 2025-07-25 by github-actions[bot]