Koenkk/zigbee2mqtt #25757: Zigbee2MQTT container intermittently stopping after upgrading to 2.0.0

gui28347, 2025-01-11:
### What happened?

Container intermittently stopping in HA after upgrading to 2.0

### What did you expect to happen?

Have Z2M running without stop, like before upgrading

### How to reproduce it (minimal and precise)

Install HA with all updates
Install Z2M latest available addon for HA (2.0.0)
Configure [SLZB-06p7] via network to communicate with HA

### Zigbee2MQTT version

2.0.0

### Adapter firmware version

20240716

### Adapter

SLZB-06p7

### Setup

Addon in HomeAssistant Yellow - 2 Intances, Local and Remote (TCP/Socket)

### Debug log

[09:46:15] INFO: Preparing to start...
[09:46:15] INFO: Socat not enabled
[09:46:16] INFO: Starting Zigbee2MQTT...
Starting Zigbee2MQTT without watchdog.
[2025-01-11 09:46:21] info: 	z2m: Logging to console, file (filename: log.log)
[2025-01-11 09:46:21] info: 	z2m: Starting Zigbee2MQTT version 2.0.0 (commit #unknown)
[2025-01-11 09:46:21] info: 	z2m: Starting zigbee-herdsman (3.2.1)
[2025-01-11 09:46:21] info: 	zh:zstack:znp: Opening TCP socket with 192.168.1.7:6638
[2025-01-11 09:46:23] error: 	zh:zstack:znp: Socket error Error: connect EHOSTUNREACH 192.168.1.7:6638
[2025-01-11 09:46:23] error: 	z2m: Error while starting zigbee-herdsman
[2025-01-11 09:46:23] error: 	z2m: Failed to start zigbee-herdsman
[2025-01-11 09:46:23] error: 	z2m: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start_crashes-runtime.html for possible solutions
[2025-01-11 09:46:23] error: 	z2m: Exiting...
[2025-01-11 09:46:23] error: 	z2m: Error: Error while opening socket
    at Socket.<anonymous> (/app/node_modules/.pnpm/zigbee-herdsman@3.2.1/node_modules/zigbee-herdsman/src/adapter/z-stack/znp/znp.ts:158:24)
    at Socket.emit (node:events:530:35)
    at emitErrorNT (node:internal/streams/destroy:170:8)
    at emitErrorCloseNT (node:internal/streams/destroy:129:3)
    at processTicksAndRejections (node:internal/process/task_queues:90:21)
/app/node_modules/.pnpm/readable-stream@3.6.2/node_modules/readable-stream/lib/_stream_writable.js:264
  var er = new ERR_STREAM_WRITE_AFTER_END();
           ^
Error: write after end
    at writeAfterEnd (/app/node_modules/.pnpm/readable-stream@3.6.2/node_modules/readable-stream/lib/_stream_writable.js:264:12)
    at DerivedLogger.Writable.write (/app/node_modules/.pnpm/readable-stream@3.6.2/node_modules/readable-stream/lib/_stream_writable.js:300:21)
    at DerivedLogger.log (/app/node_modules/.pnpm/winston@3.17.0/node_modules/winston/lib/winston/logger.js:231:12)
    at Logger.log (/app/lib/util/logger.ts:198:25)
    at Logger.info (/app/lib/util/logger.ts:211:14)
    at Znp.onPortClose (/app/node_modules/.pnpm/zigbee-herdsman@3.2.1/node_modules/zigbee-herdsman/src/adapter/z-stack/znp/znp.ts:88:16)
    at Object.onceWrapper (node:events:633:26)
    at Socket.emit (node:events:518:28)
    at TCP.<anonymous> (node:net:343:12)
[10:18:42] INFO: Preparing to start...
[10:18:42] INFO: Socat not enabled
[10:18:43] INFO: Starting Zigbee2MQTT...
Starting Zigbee2MQTT without watchdog.
[2025-01-11 10:18:49] info: 	z2m: Logging to console, file (filename: log.log)
[2025-01-11 10:18:49] info: 	z2m: Starting Zigbee2MQTT version 2.0.0 (commit #unknown)

djurny, 2025-01-11:
Hi @gui28347,
Can you post your zigbee2mqtt `configuration.yaml` and a screenshot of the add-on settings?

Looks like you have a "remote" adapter set up that should be reachable at `192.168.1.7:6638`, is that correct?

Are you sure `ser2net` is configured and running properly?

Grt,

gui28347, 2025-01-13:
data_path: /config/zigbee2mqtt2
socat:
  enabled: false
  master: pty,raw,echo=0,link=/tmp/ttyZ2M2,mode=777
  slave: tcp-listen:8486,keepalive,nodelay,reuseaddr,keepidle=1,keepintvl=1,keepcnt=5
  options: "-d -d"
  log: false
mqtt:
  server: mqtt://192.168.1.4:1883
  user: ******
  password: ******
  base_topic: zigbee2mqtt2
serial:
  port: tcp://192.168.1.7:6638
  baudrate: 115200
  adapter: zstack
  advanced:
    transmit_power: 20
advanced:
  homeassistant_legacy_entity_attributes: false
  homeassistant_legacy_triggers: false
  legacy_api: false
  legacy_availability_payload: false
  channel: 12
device_options:
  legacy: false
ota:
  image_block_response_delay: 100
  default_maximum_data_size: 50

How can I check about properly configured ser2net ? I never had this issue prior Zigbee2MQTT v2.0 ...

djurny, 2025-01-13:
Hi @gui28347,
Sorry, I did not catch that you were using a SLZB-06p7, which does not require `ser2net` to run.
Have you tried to check the web UI of the dongle directly?
Grt,

gui28347, 2025-01-14:
It appears something with network dropping packets. I don't believe it has something to do with zigbee2mqtt. Closing report

closed 2025-01-14 by gui28347