Koenkk/zigbee2mqtt #25759: Gledopto GL-G-002P Keeps dissapearing from Zigbee2MQTT

MiranoVerhoef, 2025-01-11:
### What happened?

After the 2.0.0 update one device keeps dissapearing after a while from my device list. I only notice it whenever the automations scheduled on the light doesn't work and i check and its gone from the device list.

It is someting that is occuring since the newest version, it hasn't happend before.

### What did you expect to happen?

Device should not dissapear from the device list so i can keep controlling it

### How to reproduce it (minimal and precise)

Add device through pairing mode, wait for unkown time and it just dissapears from list

### Zigbee2MQTT version

2.0.0

### Adapter firmware version

20240710

### Adapter

CC2652P7

### Setup

Z2M as addon within HA which is a VM inside unpaid

### Debug log

Debug logging is now enabled and device is added back. so i will add debug logging whenever its gone again
[log.log](https://github.com/user-attachments/files/18387669/log.log)

[45df7312_zigbee2mqtt_2025-01-11T23-30-39.603Z.log](https://github.com/user-attachments/files/18387666/45df7312_zigbee2mqtt_2025-01-11T23-30-39.603Z.log)

senna1992, 2025-01-12:
Can you try triggering a few actions on the light right after pairing it?

Had similar behavior / fix on a IKEA symphonic remote. Thought it could be related to your issue.

MiranoVerhoef, 2025-01-12:
Yeah so it’s still here now, it disappears after like a day or 4 so that’s the strange part.

closed 2025-03-21 by github-actions[bot]