Koenkk/zigbee2mqtt #25762: error Wrong direction

Daximino, 2025-01-12:
### What happened?

I have a slzb-06m.
I have update zigbee2mqtt to version 2.00.2.
I have modified configuration.yaml adding these lines:
advanced:
  homeassistant_legacy_entity_attributes: false
  homeassistant_legacy_triggers: false
  legacy_api: false
  legacy_availability_payload: false
device_options:
  legacy: false
Now i have an error that disconnect my z2m network.
[00:18:16] INFO: Preparing to start...
[00:18:16] INFO: Socat not enabled
[00:18:16] INFO: Starting Zigbee2MQTT...
Starting Zigbee2MQTT without watchdog.
[2025-01-12 00:19:11] error: 	zh:ember:ezsp: ERROR Transaction failure; status=ERROR_WRONG_DIRECTION. Last Frame: [FRAME: ID=52:"SEND_UNICAST" Seq=36 Len=34].
[2025-01-12 00:19:11] error: 	zh:ember:ezsp: ERROR Transaction failure; status=ERROR_WRONG_DIRECTION. Last Frame: [FRAME: ID=52:"SEND_UNICAST" Seq=36 Len=34].
[2025-01-12 00:19:38] error: 	zh:controller:device: Handling of poll check-in from 0x28dba7fffe704a9f failed (ZCL command 0x28dba7fffe704a9f/1 genPollCtrl.checkinRsp({"startFastPolling":false,"fastPollTimeout":0}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"reservedBits":0,"writeUndiv":false,"sendPolicy":"immediate"}) failed ({"target":20431,"apsFrame":{"profileId":260,"clusterId":32,"sourceEndpoint":1,"destinationEndpoint":1,"options":4416,"groupId":0,"sequence":58},"zclSequence":5,"commandIdentifier":11} timed out after 10000ms))
[2025-01-12 00:19:43] error: 	zh:ember:ezsp: ERROR Transaction failure; status=ERROR_WRONG_DIRECTION. Last Frame: [FRAME: ID=52:"SEND_UNICAST" Seq=42 Len=26].
[2025-01-12 00:19:43] error: 	zh:ember:ezsp: ERROR Transaction failure; status=ERROR_WRONG_DIRECTION. Last Frame: [FRAME: ID=52:"SEND_UNICAST" Seq=42 Len=26].
[2025-01-12 00:19:47] error: 	zh:ember:ezsp: ERROR Transaction failure; status=ERROR_WRONG_DIRECTION. Last Frame: [FRAME: ID=52:"SEND_UNICAST" Seq=44 Len=26].
[2025-01-12 00:19:47] error: 	zh:ember:ezsp: ERROR Transaction failure; status=ERROR_WRONG_DIRECTION. Last Frame: [FRAME: ID=52:"SEND_UNICAST" Seq=44 Len=26].
[2025-01-12 00:21:35] error: 	zh:ember:ezsp: ERROR Transaction failure; status=ERROR_WRONG_DIRECTION. Last Frame: [FRAME: ID=52:"SEND_UNICAST" Seq=86 Len=30].
[2025-01-12 00:21:35] error: 	zh:ember:ezsp: ERROR Transaction failure; status=ERROR_WRONG_DIRECTION. Last Frame: [FRAME: ID=52:"SEND_UNICAST" Seq=86 Len=30].
[2025-01-12 00:21:46] error: 	zh:ember:ezsp: ERROR Transaction failure; status=ERROR_WRONG_DIRECTION. Last Frame: [FRAME: ID=52:"SEND_UNICAST" Seq=89 Len=26].
[2025-01-12 00:21:46] error: 	zh:ember:ezsp: ERROR Transaction failure; status=ERROR_WRONG_DIRECTION. Last Frame: [FRAME: ID=52:"SEND_UNICAST" Seq=89 Len=26].
[2025-01-12 09:48:34] error: 	zh:ember:uart:ash: Received ERROR from adapter, with code=ERROR_EXCEEDED_MAXIMUM_ACK_TIMEOUT_COUNT.
[2025-01-12 09:48:34] error: 	zh:ember:uart:ash: ASH disconnected | Adapter status: ASH_NCP_FATAL_ERROR
[2025-01-12 09:48:34] error: 	zh:ember:uart:ash: Error while parsing received frame, status=ASH_NCP_FATAL_ERROR.
[2025-01-12 09:48:34] error: 	zh:ember: Adapter fatal error: HOST_FATAL_ERROR
[2025-01-12 09:48:34] error: 	z2m: Adapter disconnected, stopping

### Zigbee2MQTT version

2.0.0.2

### Adapter firmware version

2.6.8 dev25

### Adapter

Slzb-06m

### Setup

Addon

Xander-V, 2025-01-12:
Did you add the line for mDNS Zeroconf autodiscovery:

```
serial:
    port: mdns://slzb-06
```

See [manual](https://www.zigbee2mqtt.io/guide/configuration/adapter-settings.html#mdns-zeroconf-discovery)

Daximino, 2025-01-18:
I have 
Serial: 
     Port: TCP://slzb-06

It's wrong?

closed 2025-03-27 by github-actions[bot]