Koenkk/zigbee2mqtt #25765: Problem Z2M - Error while starting zigbee-herdsman. New install of HA, Mosquitto and Z2M 2.0.0-2

deltabert, 2025-01-12:
### What happened?

I had problems with the upgrade to Z2M to 2.0. And could not find a solution. So I decided to us a spare RPI5 and install everything fresh.    So, I made a new, fresh install of Home Assistant, Mosquitto an Z2M v 2.0.0-2.
I am using the SMLIGHT SLZB-06 witch is an Ethernet Coordinator. and this worked perfect with Z2M v 1.34

But now I have the same errors in Z2M log as I had after the upgrade from 1.34 to 2.0, 
My YAML configuration:
data_path: /config/zigbee2mqtt
socat:
  enabled: false
  master: pty,raw,echo=0,link=/tmp/ttyZ2M,mode=777
  slave: tcp-listen:8485,keepalive,nodelay,reuseaddr,keepidle=1,keepintvl=1,keepcnt=5
  options: "-d -d"
  log: false
mqtt:
  server: mqtt://192.168.178.215:1883
  user: mqtt-user
  password: quest3
serial:
  serial:
    port: tcp://192.168.178.216:6638
    baudrate: 115200
    adapter: zstack
    disable_led: false
  advanced:
    transmit_power: 20
    log_level: debug 
`

### What did you expect to happen?

I was hoping, with a fresh  install of HA, Mosquitto and Z2M that it would work right out of the box!

I checket this issue but that did not solve my problem:
https://github.com/Koenkk/zigbee2mqtt/issues/25619

### How to reproduce it (minimal and precise)

See above resume

### Zigbee2MQTT version

2.0.0-2

### Adapter firmware version

Firmware version:	20240710	v2.6.8.dev26

### Adapter

SMLIGHT SLZB-06  Ethernet 

### Setup

RPi 5 with HAOS Latest version, Mosquiutto, Z2M 2.0.0-2

### Debug log

[11:36:01] INFO: Preparing to start...
[11:36:01] INFO: Socat not enabled
[11:36:01] INFO: Starting Zigbee2MQTT...
Starting Zigbee2MQTT without watchdog.
[2025-01-12 11:36:02] info: 	z2m: Logging to console, file (filename: log.log)
[2025-01-12 11:36:02] info: 	z2m: Starting Zigbee2MQTT version 2.0.0 (commit #unknown)
[2025-01-12 11:36:02] info: 	z2m: Starting zigbee-herdsman (3.2.1)
[2025-01-12 11:36:02] error: 	**z2m: Error while starting zigbee-herdsman**
[2025-01-12 11:36:02] error: 	**z2m: Failed to start zigbee-herdsman**
[2025-01-12 11:36:02] error: 	z2m: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start_crashes-runtime.html for possible solutions
[2025-01-12 11:36:02] error: 	z2m: Exiting...
[2025-01-12 11:36:02] error: 	**z2m: Error: USB adapter discovery error (No valid USB adapter found). Specify valid 'adapter' and 'port' in your configuration.**
    at discoverAdapter (/app/node_modules/.pnpm/zigbee-herdsman@3.2.1/node_modules/zigbee-herdsman/src/adapter/adapterDiscovery.ts:523:15)
    at processTicksAndRejections (node:internal/process/task_queues:105:5)
    at Function.create (/app/node_modules/.pnpm/zigbee-herdsman@3.2.1/node_modules/zigbee-herdsman/src/adapter/adapter.ts:68:33)
    at Controller.start (/app/node_modules/.pnpm/zigbee-herdsman@3.2.1/node_modules/zigbee-herdsman/src/controller/controller.ts:132:24)
    at Zigbee.start (/app/lib/zigbee.ts:69:27)
    at Controller.start (/app/lib/controller.ts:142:13)
    at start (/app/index.js:161:5)
[2025-01-12 11:36:02] error: 	z2m: If this happens after updating to Zigbee2MQTT 2.0.0, see https://github.com/Koenkk/zigbee2mqtt/discussions/24364

deltabert, 2025-01-12:
I am sorry, I used the CODE ticks in the description of the configuration. But it does not show up.

Fyase82, 2025-01-12:
Hi, follow the instruction here: https://www.zigbee2mqtt.io/guide/installation/

Basically you have to specify the adapter type in the configuration.yaml like this one https://www.zigbee2mqtt.io/guide/configuration/adapter-settings.html#basic-configuration

deltabert, 2025-01-12:
Hi Fyase82, Thank you for pointing me to the mDNS settings for the SLZB-06 coordinator.
That was the problem. Tanks again

deltabert, 2025-01-21:
This is the solution i found:

`data_path: /config/zigbee2mqtt
socat:
  enabled: false
  master: pty,raw,echo=0,link=/tmp/ttyZ2M,mode=777
  slave: tcp-listen:8485,keepalive,nodelay,reuseaddr,keepidle=1,keepintvl=1,keepcnt=5
  options: "-d -d"
  log: false
mqtt:
  server: mqtt://192.168.178.221:1883
  user: <user>
  password: <My Pw>
serial:
  port: tcp://192.168.178.216:6638
  adapter: zstack
homeassistant:
  enabled: true
frontend:
  port: 8099
advanced:
  homeassistant_legacy_entity_attributes: false
  homeassistant_legacy_triggers: false
  legacy_api: false
  legacy_availability_payload: false
  last_seen: ISO_8601_local
device_options:
  legacy: false
devices: null`

closed 2025-01-21 by deltabert