Koenkk/zigbee2mqtt #25771: [New device support]: TGM50-ZB-WPB

DylanoDemore, 2025-01-12:
### Link

https://nl.aliexpress.com/item/1005007628099393.html?gatewayAdapt=glo2nld

### Database entry

{"id":58,"type":"Router","ieeeAddr":"0xa4c138076095ab86","nwkAddr":30439,"manufId":4417,"manufName":"_TZE204_cvub6xbb","powerSource":"Mains (single phase)","modelId":"TS0601","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[4,5,61184,0],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"stackVersion":0,"dateCode":""}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":74,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1736689465322}

### Zigbee2MQTT version

2.0.0-2

### Comments

It's the same device as the TGM50-ZB but is not recognized.

### External definition

```shell
const definition = {
    zigbeeModel: ['TS0601'],
    model: 'TS0601',
    vendor: '_TZE204_cvub6xbb',
    description: 'Automatically generated definition',
    extend: [],
    meta: {},
};

module.exports = definition;
```

### What does/doesn't work with the external definition?

It's stil not supported

DylanoDemore, 2025-01-17:
i found out that the one that is supported is
TGM50-ZB-WP (_TZE204_mwomyz5n) for water floor heating
there are two other ones the TGM50-ZB-WPB (_TZE204_cvub6xbb) for gas boiler heating
and the TGM50-ZB-EP for electric floor heating

Dachmeyer, 2025-01-19:
I have the same Problem.

Wilfredus, 2025-01-20:
Hi. I have the same problem here, though mine is a BEOK thermostat, similar to the one supported (TGM50-ZB-WP), and apparently identical to the one you bougth.

The same definition for the already supported one works well as an external converter, but there are some functionalities that are not working (sound on/off, window opening detection, main and auxili reverse selection...). Hope it helps:

`const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const modernExtend = require('zigbee-herdsman-converters/lib/modernExtend');
const e = exposes.presets;
const ea = exposes.access;
const tuya = require('zigbee-herdsman-converters/lib/tuya');

const definition = {
    fingerprint: tuya.fingerprint('TS0601', ['_TZE204_cvub6xbb']),
    zigbeeModel: ['TS0601'],
    model: 'TGM50-ZB-WP',
    vendor: 'BEOK',
    description: 'BEOK wall panel thermostat for water underfloor',
    fromZigbee: [tuya.fz.datapoints],
    toZigbee: [tuya.tz.datapoints],
    onEvent: tuya.onEvent({timeStart: '1970'}),
    configure: tuya.configureMagicPacket,
    exposes: [
        e.child_lock(),
        e.temperature_sensor_select(['internal', 'external', 'both']),
        e
            .climate()
            .withSystemMode(['off', 'heat'], ea.STATE_SET)
            .withPreset(['manual', 'auto', 'eco'])
            .withSetpoint('current_heating_setpoint', 5, 35, 0.5, ea.STATE_SET)
            .withRunningState(['idle', 'heat'], ea.STATE)
            .withLocalTemperature(ea.STATE)
            .withLocalTemperatureCalibration(-9.9, 9.9, 0.1, ea.STATE_SET),
        e.enum('backlight_mode', ea.STATE_SET, ['auto_off', 'always_low', 'always_mid', 'always_high']).withDescription('Intensity of the backlight'),
        e.binary('frost_protection', ea.STATE_SET, 'ON', 'OFF').withDescription('Antifreeze function'),
        e
            .max_temperature_limit()
            .withUnit('°C')
            .withValueMin(15)
            .withValueMax(90)
            .withValueStep(0.5)
            .withPreset('default', 60, 'Default value')
            .withDescription('Maximum upper temperature'),
        e
            .numeric('temperature_delta', ea.STATE_SET)
            .withUnit('°C')
            .withValueMax(10)
            .withValueMin(0.5)
            .withValueStep(0.5)
            .withPreset('default', 1, 'Default value')
            .withDescription('The delta between local_temperature and current_heating_setpoint to trigger Heat'),
        e.binary('factory_reset', ea.STATE_SET, 'ON', 'OFF').withDescription('Full factory reset, use with caution!'),
    ],
    meta: {
        tuyaDatapoints: [
            // functions not discovered: sound on-off, rest day selection, main and auxili (reverses the output, kind of summer/winter modes), window open detection, power outage memory function
            [1, 'system_mode', tuya.valueConverterBasic.lookup({heat: true, off: false})],
            [2, 'current_heating_setpoint', tuya.valueConverter.divideBy10],
            [3, 'local_temperature', tuya.valueConverter.divideBy10],
            [4, 'preset', tuya.valueConverterBasic.lookup({manual: tuya.enum(0), auto: tuya.enum(1), eco: tuya.enum(3)})],
            [9, 'child_lock', tuya.valueConverter.lockUnlock],
            [15, 'max_temperature_limit', tuya.valueConverter.divideBy10],
            [19, 'local_temperature_calibration', tuya.valueConverter.localTempCalibration3],
            [101, 'running_state', tuya.valueConverterBasic.lookup({heat: tuya.enum(1), idle: tuya.enum(0)})],
            [102, 'frost_protection', tuya.valueConverter.onOff],
            [103, 'factory_reset', tuya.valueConverter.onOff],
            [106, 'sensor', tuya.valueConverterBasic.lookup({internal: tuya.enum(0), external: tuya.enum(1), both: tuya.enum(2)})],
            [107, 'temperature_delta', tuya.valueConverter.divideBy10],
            [
                110,
                'backlight_mode',
                tuya.valueConverterBasic.lookup({
                    off: tuya.enum(0),
                    always_low: tuya.enum(1),
                    always_mid: tuya.enum(2),
                    always_high: tuya.enum(3),
                }),
            ],
        ],    
    },
    extend: [
        // A preferred new way of extending functionality.
    ],
};

module.exports = definition;`

DylanoDemore, 2025-01-21:
Thank you so much [Wilfredus](https://github.com/Wilfredus)
The converter is working i just had to create a different folder in a different location.
/homeassistant/zigbee2mqtt/external_converters instead of /homeassistant/zigbee2mqtt/data/external_converters
again big thank to [Wilfredus](https://github.com/Wilfredus)

DylanoDemore, 2025-01-23:
I have a surprise. I have enabled some features auto off backlight was not working now it is plus turn off\on sound is working and now its possible to reverse the relay behavior (when thermostat go off relay turn on)
Here is the external converter if anyone wants to try

`const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const modernExtend = require('zigbee-herdsman-converters/lib/modernExtend');
const e = exposes.presets;
const ea = exposes.access;
const tuya = require('zigbee-herdsman-converters/lib/tuya');

const definition = {
    fingerprint: tuya.fingerprint('TS0601', ['_TZE204_cvub6xbb']),
    zigbeeModel: ['TS0601'],
    model: 'TGM50-ZB-WP',
    vendor: 'BEOK',
    description: 'BEOK wall panel thermostat for water underfloor',
    fromZigbee: [tuya.fz.datapoints],
    toZigbee: [tuya.tz.datapoints],
    onEvent: tuya.onEvent({timeStart: '1970'}),
    configure: tuya.configureMagicPacket,
    exposes: [
        e.child_lock(),
        e.temperature_sensor_select(['internal', 'external', 'both']),
        e
            .climate()
            .withSystemMode(['off', 'heat'], ea.STATE_SET)
            .withPreset(['manual', 'auto', 'eco'])
            .withSetpoint('current_heating_setpoint', 5, 35, 0.5, ea.STATE_SET)
            .withRunningState(['idle', 'heat'], ea.STATE)
            .withLocalTemperature(ea.STATE)
            .withLocalTemperatureCalibration(-9.9, 9.9, 0.1, ea.STATE_SET),
        e.enum('backlight_mode', ea.STATE_SET, ['auto_off', 'always_low', 'always_mid', 'always_high']).withDescription('Intensity of the backlight'),
        e.binary('frost_protection', ea.STATE_SET, 'ON', 'OFF').withDescription('Antifreeze function'),
        e
            .max_temperature_limit()
            .withUnit('°C')
            .withValueMin(15)
            .withValueMax(90)
            .withValueStep(0.5)
            .withPreset('default', 60, 'Default value')
            .withDescription('Maximum upper temperature'),
        e
            .numeric('temperature_delta', ea.STATE_SET)
            .withUnit('°C')
            .withValueMax(10)
            .withValueMin(0.5)
            .withValueStep(0.5)
            .withPreset('default', 1, 'Default value')
            .withDescription('The delta between local_temperature and current_heating_setpoint to trigger Heat'),
        e.binary('sound', ea.STATE_SET, 'ON', 'OFF').withDescription('Switches beep sound when interacting with thermostat'),
        e.binary('inverse', ea.STATE_SET, true, false).withDescription('Invert the main output relay'),
        e.binary('factory_reset', ea.STATE_SET, 'ON', 'OFF').withDescription('Full factory reset, use with caution!'),
    ],
    meta: {
        tuyaDatapoints: [
            // functions not discovered: sound on-off, rest day selection, main and auxili (reverses the output, kind of summer/winter modes), window open detection, power outage memory function
            [1, 'system_mode', tuya.valueConverterBasic.lookup({heat: true, off: false})],
            [2, 'current_heating_setpoint', tuya.valueConverter.divideBy10],
            [3, 'local_temperature', tuya.valueConverter.divideBy10],
            [4, 'preset', tuya.valueConverterBasic.lookup({manual: tuya.enum(0), auto: tuya.enum(1), eco: tuya.enum(3)})],
            [9, 'child_lock', tuya.valueConverter.lockUnlock],
            [15, 'max_temperature_limit', tuya.valueConverter.divideBy10],
            [19, 'local_temperature_calibration', tuya.valueConverter.localTempCalibration3],
            [101, 'running_state', tuya.valueConverterBasic.lookup({heat: tuya.enum(1), idle: tuya.enum(0)})],
            [102, 'frost_protection', tuya.valueConverter.onOff],
            [105, 'sound', tuya.valueConverter.onOff],
            [103, 'factory_reset', tuya.valueConverter.onOff],
            [106, 'sensor', tuya.valueConverterBasic.lookup({internal: tuya.enum(0), external: tuya.enum(1), both: tuya.enum(2)})],
            [107, 'temperature_delta', tuya.valueConverter.divideBy10],
            [110, 'backlight_mode', tuya.valueConverter.backlightModeOffLowMediumHigh],
            [111, 'inverse', tuya.valueConverter.inverse],
        ],    
    },
    extend: [
        // A preferred new way of extending functionality.
    ],
};

module.exports = definition;`

Wilfredus, 2025-01-24:
Thank you very much @DylanoDemore, the datapoints for 'sound' and 'reverse' work both perfectly. However, the original version for the backlight mode works fine for me, so there must be slightly differences between our dispositives.

Arubinu, 2025-01-25:
Hello, I have the same problem on the same fingerprint. Except that when I add it to the extensions tab, I get this error message when saving:

`z2m: Request 'zigbee2mqtt/bridge/request/extension/save' failed with error: 'ConstructorClass is not a constructor'`

Do you know where the problem comes from? It seems to generate an example for me with a constructor.

Schafy, 2025-01-30:
Same here. Bought 5 devices, one was broken (assemble issue). The replacement device came up with the new model name.

closed 2025-04-09 by github-actions[bot]

wikibrikofix, 2025-08-01:
I have received some new TGM50-ZB in substitution of other with zigbee problems and the new one are identified as

_TZE284_cvub6xbb

How can I remap the device as _TZE204_cvub6xbb?

terragady, 2026-03-20:
yeah that is true but there is no custom firmware for that particular Beok thermostat, it is only for MOES ones with Telink processor (unfortunately i have Silabs MOES and I can't do anything with spamming my zigbee network and buying new ones from MOES is a lottery as you newer know what you will get). That is why I am asking if that Beok has any issues with spamming the zigbee network.

Check this out 😄 
[image]