Koenkk/zigbee2mqtt #25772: Tuya TS004F Action "hold" is missing

Mordevolt, 2025-01-12:
### What happened?

In Z2M 2.0, only  "hold" actions are send to the event entity.
It is not producing event for hold action correctly.

### What did you expect to happen?

In HA only legacy sensor detects "4_hold", new event sensor detects just "hold", without button reference:
[image]

### How to reproduce it (minimal and precise)

Even in list for all available events it is just plain "hold", without reference to exact button:
[image]
seems like not all Action (enum) exposed in HA correctly

### Zigbee2MQTT version

 2.0.0

### Adapter firmware version

20240710

### Adapter

zStack3x0

### Setup

Homeassistant OS

### Debug log

info 2025-01-11 14:40:00z2m:mqtt: MQTT publish: topic 'homeassistant/device_automation/0x84fd27fffe803813/action_4_hold/config', payload '{"automation_type":"trigger","device":{"hw_version":1,"identifiers":["zigbee2mqtt_0x84fd27fffe803813"],"manufacturer":"Tuya","model":"Wireless switch with 4 buttons","model_id":"TS004F","name":"4 Button Switch","via_device":"zigbee2mqtt_bridge_0x00124b0024bfc9c6"},"origin":{"name":"Zigbee2MQTT","sw":"2.0.0","url":"[https://www.zigbee2mqtt.io"},"payload":"4_hold","subtype":"4_hold","topic":"zigbee2mqtt/4](https://www.zigbee2mqtt.io"},"payload":"4_hold","subtype":"4_hold","topic":%22zigbee2mqtt/4) Button Switch/action","type":"action"}'
info 2025-01-11 14:40:01z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/4 Button Switch/action', payload '4_hold'

tdvtdv, 2025-01-15:
Same problem for TS0044

TurnrDev, 2025-01-15:
I can't even get hold events to show in Z2M nor HA

mundschenk-at, 2025-01-18:
The issue is that `4_hold` matches a regular expression that splits the event payload into the actual action happening (`hold`) and a separate attribute `button`, but that regex does not consider the actions `single` and `double`. This is something that will be addressed when `event` entities are reworked after https://github.com/Koenkk/zigbee2mqtt/discussions/25061 has been decided.

Mordevolt, 2025-02-22:
> The issue is that `4_hold` matches a regular expression that splits the event payload into the actual action happening (`hold`) and a separate attribute `button`, but that regex does not consider [...]

any progress on this? as i see, there is no active discussion on https://github.com/Koenkk/zigbee2mqtt/discussions/25061

fapgomes, 2025-03-04:
Hi,

I noticed that only the events 1_hold, 2_hold, 3_hold, and 4_hold, despite appearing correctly in Z2M, show up as 'hold' in Home Assistant. The others 1_double/1_single, 2_double/2_single, 3_double/3_single, and 4_double/4_single are working fine. Is there any solution to this?

Best Regards,
./fapg

fapgomes, 2025-05-04:
Hi, 

The problem still exists...

./fapg

wookiesh, 2025-06-06:
Sorry @fapgomes, just saw your commit with the correction and the correct value `button == 3`. 
I replaced the blueprint with your last committed one and it works perfectly. Thanks ! 🙏🏼

closed 2025-08-13 by github-actions[bot]