Koenkk/zigbee2mqtt #25776: Refusing to accept a string as network_key

TheFax, 2025-01-12:
### What happened?

I configured Zigbee2MQTT via its web interface, and I used a string as network_key.
The string is exactly 16 bytes long and it is similar (but not exactly) this: `Zigbee#TESTPass!`.
After that, I applied the changes and reboot the Zigbee2MQTT integration.

### What did you expect to happen?

I expect the integration restarts and use the new network_key.
This is not happening because integration "Refusing to start because network_key should be an array" as wrote in the LOG.

### How to reproduce it (minimal and precise)

1) Use a string as network_key
2) Save
3) Reboot integration

### Zigbee2MQTT version

2.0.0-2

### Adapter firmware version

Unable to retrieve this info now.

### Adapter

Ember - Sonoff Zigbee dongle E

### Setup

Hassio on Raspberry

### Debug log

```
[17:20:32] INFO: Preparing to start...
[17:20:32] INFO: Socat not enabled
[17:20:33] INFO: Starting Zigbee2MQTT...
Starting Zigbee2MQTT without watchdog.
!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!
            READ THIS CAREFULLY
Refusing to start because configuration is not valid, found the following errors:
- advanced.network_key: should be array or 'GENERATE' (is 'Zigbee#TESTPass!')
If you don't know how to solve this, read https://www.zigbee2mqtt.io/guide/configuration
!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!
```

KennethLavrsen, 2025-01-13:
The network key is supposed to be a 128 bit number or 16 hexadecimal numbers. You cannot put an ascii string

See https://www.zigbee2mqtt.io/advanced/zigbee/03_secure_network.html

TheFax, 2025-01-13:
Yes, I have consulted the documentation and understand that the YAML file accepts only a 128-bit numerical value or its hexadecimal representation as a 16-character string.
I had assumed that the frontend would automatically perform the conversion from a string input to its hexadecimal equivalent, given that the "Network_key" configuration menu offers an "string" option.

[image]

If the field does not accept a string, then I find the term "string" misleading. It should be replaced with "16-character hexadecimal string", "Raw YAML" or something else, real useful to users.

closed 2025-03-22 by github-actions[bot]