Koenkk/zigbee2mqtt #25778: zigbee2mqtt

ucakchi, 2025-01-12:
### What happened?

I try to configurate Z2M but I am recieving errors. 

Homeassistant is installed on Raspberry Pi4
Usb dungle is Sonoff 3.0 USB Dongle Plus. Ember firmware is installed.

How can I fix it?

You can find YAML code below.

`data_path: /config/zigbee2mqtt
socat:
  enabled: false
  master: pty,raw,echo=0,link=/tmp/ttyZ2M,mode=777
  slave: tcp-listen:8485,keepalive,nodelay,reuseaddr,keepidle=1,keepintvl=1,keepcnt=5
  options: "-d -d"
  log: false
mqtt:
  base_topic: zigbee2mqtt
  server: mqtt://core-mosquitto:1883
  user: xxxxxx
  password: xxxxxxx
serial:
  port: >-
    /dev/serial/by-id/usb-ITead_Sonoff_Zigbee_3.0_USB_Dongle_Plus_42d1cc8232fded11a3540765024206e6-if00-port0
  adapter: ember
  baudrate: 115200
frontend:
  enabled: true
homeassistant:
  enabled: true
`

This is the log page. You can see errors. below. 

### Zigbee2MQTT version

 2.0.0-2

### Adapter firmware version

7.4.5.0 build 0

### Adapter

Sonoff 3.0 USB dungle ZB dongle-E

### Setup

Home assistant on Pi4. 

### Debug log

[00:11:25] INFO: Preparing to start...
[00:11:25] INFO: Socat not enabled
[00:11:26] INFO: Starting Zigbee2MQTT...
Starting Zigbee2MQTT without watchdog.
[2025-01-13 00:11:30] info: 	z2m: Logging to console, file (filename: log.log)
[2025-01-13 00:11:30] info: 	z2m: Starting Zigbee2MQTT version 2.0.0 (commit #unknown)
[2025-01-13 00:11:30] info: 	z2m: Starting zigbee-herdsman (3.2.1)
[2025-01-13 00:11:30] info: 	zh:ember: Using default stack config.
[2025-01-13 00:11:30] info: 	zh:ember: ======== Ember Adapter Starting ========
[2025-01-13 00:11:30] info: 	zh:ember:ezsp: ======== EZSP starting ========
[2025-01-13 00:11:30] info: 	zh:ember:uart:ash: ======== ASH Adapter reset ========
[2025-01-13 00:11:30] info: 	zh:ember:uart:ash: RTS/CTS config is off, enabling software flow control.
[2025-01-13 00:11:30] info: 	zh:ember:uart:ash: Serial port opened
[2025-01-13 00:11:30] info: 	zh:ember:uart:ash: ======== ASH starting ========
[2025-01-13 00:11:33] info: 	zh:ember:uart:ash: ======== ASH Adapter reset ========
[2025-01-13 00:11:33] info: 	zh:ember:uart:ash: ======== ASH starting ========
[2025-01-13 00:11:35] info: 	zh:ember:uart:ash: ======== ASH Adapter reset ========
[2025-01-13 00:11:35] info: 	zh:ember:uart:ash: ======== ASH starting ========
[2025-01-13 00:11:38] info: 	zh:ember:uart:ash: ======== ASH Adapter reset ========
[2025-01-13 00:11:38] info: 	zh:ember:uart:ash: ======== ASH starting ========
[2025-01-13 00:11:41] info: 	zh:ember:uart:ash: ======== ASH Adapter reset ========
[2025-01-13 00:11:41] info: 	zh:ember:uart:ash: ======== ASH starting ========
[2025-01-13 00:11:43] error: 	z2m: Error while starting zigbee-herdsman
[2025-01-13 00:11:43] error: 	z2m: Failed to start zigbee-herdsman
[2025-01-13 00:11:43] error: 	z2m: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start_crashes-runtime.html for possible solutions
[2025-01-13 00:11:43] error: 	z2m: Exiting...
[2025-01-13 00:11:43] error: 	z2m: Error: Failed to start EZSP layer with status=HOST_FATAL_ERROR.
    at EmberAdapter.initEzsp (/app/node_modules/.pnpm/zigbee-herdsman@3.2.1/node_modules/zigbee-herdsman/src/adapter/ember/adapter/emberAdapter.ts:670:19)
    at EmberAdapter.start (/app/node_modules/.pnpm/zigbee-herdsman@3.2.1/node_modules/zigbee-herdsman/src/adapter/ember/adapter/emberAdapter.ts:1533:24)
    at Controller.start (/app/node_modules/.pnpm/zigbee-herdsman@3.2.1/node_modules/zigbee-herdsman/src/controller/controller.ts:136:29)
    at Zigbee.start (/app/lib/zigbee.ts:69:27)
    at Controller.start (/app/lib/controller.ts:142:13)
    at start (/app/index.js:161:5)
[00:11:45] INFO: Preparing to start...
[00:11:45] INFO: Socat not enabled
[00:11:46] INFO: Starting Zigbee2MQTT...
Starting Zigbee2MQTT without watchdog.
[2025-01-13 00:11:50] info: 	z2m: Logging to console, file (filename: log.log)
[2025-01-13 00:11:50] info: 	z2m: Starting Zigbee2MQTT version 2.0.0 (commit #unknown)
[2025-01-13 00:11:50] info: 	z2m: Starting zigbee-herdsman (3.2.1)
[2025-01-13 00:11:51] info: 	zh:ember: Using default stack config.
[2025-01-13 00:11:51] info: 	zh:ember: ======== Ember Adapter Starting ========
[2025-01-13 00:11:51] info: 	zh:ember:ezsp: ======== EZSP starting ========
[2025-01-13 00:11:51] info: 	zh:ember:uart:ash: ======== ASH Adapter reset ========
[2025-01-13 00:11:51] info: 	zh:ember:uart:ash: RTS/CTS config is off, enabling software flow control.
[2025-01-13 00:11:51] info: 	zh:ember:uart:ash: Serial port opened
[2025-01-13 00:11:51] info: 	zh:ember:uart:ash: ======== ASH starting ========
[2025-01-13 00:11:53] info: 	zh:ember:uart:ash: ======== ASH Adapter reset ========
[2025-01-13 00:11:53] info: 	zh:ember:uart:ash: ======== ASH starting ========
[2025-01-13 00:11:54] info: 	zh:ember:uart:ash: ======== ASH connected ========
[2025-01-13 00:11:54] info: 	zh:ember:uart:ash: ======== ASH started ========
[2025-01-13 00:11:54] info: 	zh:ember:ezsp: ======== EZSP started ========
[2025-01-13 00:11:56] info: 	zh:ember: Adapter EZSP protocol version (13) lower than Host. Switched.
[2025-01-13 00:12:10] error: 	zh:ember:ezsp: ERROR Transaction failure; status=ASH_ERROR_TIMEOUTS. Last Frame: [FRAME: ID=0:"VERSION" Seq=1 Len=9].
[2025-01-13 00:12:10] error: 	zh:ember: Adapter fatal error: ASH_ERROR_TIMEOUTS
[2025-01-13 00:12:10] error: 	z2m: Error while starting zigbee-herdsman
[2025-01-13 00:12:10] error: 	z2m: Failed to start zigbee-herdsman
[2025-01-13 00:12:10] error: 	z2m: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start_crashes-runtime.html for possible solutions
[2025-01-13 00:12:10] error: 	z2m: Exiting...
[2025-01-13 00:12:10] error: 	z2m: Error: ASH_ERROR_TIMEOUTS
    at Ezsp.ezspGetValue (/app/node_modules/.pnpm/zigbee-herdsman@3.2.1/node_modules/zigbee-herdsman/src/adapter/ember/ezsp/ezsp.ts:1719:19)
    at Ezsp.ezspGetVersionStruct (/app/node_modules/.pnpm/zigbee-herdsman@3.2.1/node_modules/zigbee-herdsman/src/adapter/ember/ezsp/ezsp.ts:1214:52)
    at EmberAdapter.emberVersion (/app/node_modules/.pnpm/zigbee-herdsman@3.2.1/node_modules/zigbee-herdsman/src/adapter/ember/adapter/emberAdapter.ts:1411:41)
    at EmberAdapter.initEzsp (/app/node_modules/.pnpm/zigbee-herdsman@3.2.1/node_modules/zigbee-herdsman/src/adapter/ember/adapter/emberAdapter.ts:674:9)
    at EmberAdapter.start (/app/node_modules/.pnpm/zigbee-herdsman@3.2.1/node_modules/zigbee-herdsman/src/adapter/ember/adapter/emberAdapter.ts:1533:24)
    at Controller.start (/app/node_modules/.pnpm/zigbee-herdsman@3.2.1/node_modules/zigbee-herdsman/src/controller/controller.ts:136:29)
    at Zigbee.start (/app/lib/zigbee.ts:69:27)
    at Controller.start (/app/lib/controller.ts:142:13)
    at start (/app/index.js:161:5)
    at start (/app/index.js:161:5)

brucealdridge, 2025-01-13:
I'm seeing a similar issue on the most recent release. This looks to be a duplicate of #25779

brucealdridge, 2025-01-13:
@ucakchi try https://github.com/Koenkk/zigbee2mqtt/discussions/24364 and see if this resolves for you. It did for me.

ucakchi, 2025-01-13:
hello adaptor is written. Do I miss something?

Webserve, 2025-01-13:
Adapter EZSP protocol version (13) lower than Host. Switched.

I think you have to update the stick to a newer firmware ?

ucakchi, 2025-01-13:
I updated Adaptor. Version is 7.4.5.0 build 0

ucakchi, 2025-01-14:
I deleted ZHE and delete Sonoff usb, reload Yaml configuration. After reboot is done. 

Then Z2M has started without any trouble.

closed 2025-01-14 by ucakchi