Koenkk/zigbee2mqtt #25781: Failed to start zigbee-herdsman after update (HA and zigbee2mqtt)

wassmurf, 2025-01-12:
### What happened?

Dear Koenkk,
After the recent updates of HA and zigbee2mqtt, the system repeatedly failed to start zigbee2mqtt:
[23:46:07] INFO: Preparing to start...
[23:46:07] INFO: Socat not enabled
[23:46:08] INFO: Starting Zigbee2MQTT...
Starting Zigbee2MQTT without watchdog.
[2025-01-12 23:46:12] info: 	z2m: Logging to console, file (filename: log.log)
[2025-01-12 23:46:12] info: 	z2m: Starting Zigbee2MQTT version 2.0.0 (commit #unknown)
[2025-01-12 23:46:12] info: 	z2m: Starting zigbee-herdsman (3.2.1)
[2025-01-12 23:46:12] info: 	zh:zigate: Connected to ZiGate adapter successfully.
[2025-01-12 23:46:17] error: 	zh:zigate:driver: sendCommand error Error: {"ziGateObject":{"_code":17,"_payload":{},"_parameters":[]},"rules":[{"receivedProperty":"code","value":32774}]} after 5000ms
[2025-01-12 23:46:17] error: 	z2m: Error while starting zigbee-herdsman
[2025-01-12 23:46:17] error: 	z2m: Failed to start zigbee-herdsman
[2025-01-12 23:46:17] error: 	z2m: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start_crashes-runtime.html for possible solutions
[2025-01-12 23:46:17] error: 	z2m: Exiting...
[2025-01-12 23:46:17] error: 	z2m: Error: failed to connect to zigate adapter sendCommand error: Error: {"ziGateObject":{"_code":17,"_payload":{},"_parameters":[]},"rules":[{"receivedProperty":"code","value":32774}]} after 5000ms

Adding the adapter (zigate) to the configuration.yaml did not help.
I am using HA on a virtual machine on a synology. The configuration is:
version: 4
homeassistant:
  enabled: true
advanced:
  network_key:
    - 13
    - 197
    - 203
    - 101
    - 189
    - 83
    - 127
    - 155
    - 39
    - 48
    - 239
    - 140
    - 202
    - 198
    - 233
    - 220
  pan_id: 62885
  ext_pan_id:
    - 102
    - 11
    - 220
    - 81
    - 208
    - 239
    - 104
    - 161
mqtt:
  server: mqtt://core-mosquitto:1883
  user: addons
  password: nnn
serial:
  port: tcp://192.168.178.77:6638
  adapter: zigate
frontend:
  enabled: true
  port: 8099

As I am rather a user and not an expert, I would appreciate your help figuring out what need to be changed. For now I am using the zigbee home automation as a substitute. 
The adapter is a Zigbee-Gateway-LAN-PRO adapter connecte via LAN.
Thank you very much in advance for your kind assistance.
Best regards 
Ralf

### Zigbee2MQTT version

2.0.0-2

### Adapter firmware version

ESP32-D0WDQ5 / 0.6.10

### Adapter

Zigbee-Gateway-LAN-PRO

### Setup

HA on Virtual machine on Synology

brucealdridge, 2025-01-13:
@wassmurf try https://github.com/Koenkk/zigbee2mqtt/discussions/24364

wassmurf, 2025-01-13:
Thank Your for the hint! Unfortunately, zigbee2mqtt did not start:
[2025-01-13 14:49:10] error: 	zh:adapter:zstack:manager: Configuration is not consistent with adapter state/backup!
[2025-01-13 14:49:10] error: 	zh:adapter:zstack:manager: - PAN ID: configured=62885, adapter=6470
and
[2025-01-13 14:49:10] error: 	z2m: Error: startup failed - configuration-adapter mismatch
Using adapter: zigate allowed connection with the adpater.
So another paramter needs to be adjusted, I think. Suggestions are welcome.
Best regards
Ralf

KennethLavrsen, 2025-01-13:
Is your indentation in configuration.yaml right? You have pasted without putting it inside the code tags so we cannot see if your indentation is correct

closed 2025-03-25 by github-actions[bot]