Koenkk/zigbee2mqtt #25785: No action with Ikea Somrig

Aiace88, 2025-01-13:
### What happened?

I use the sonoff model p firmware 20230507 as a dongle
I paired the ikea switch SOMRIG, which is correct.

but at this point I only have battery, lqi and if I press the buttons no action appears in the logs.

even analyzing with mqtt explorer, I can't find the actions anywhere.
I tried to redo the pairing several times, but the actions don't arrive.

could I solve it with an upgrade of the coordinator firmware to 20240710 or something else that I'm missing?

### Zigbee2MQTT version

1.42.0

### Adapter firmware version

20230507

### Adapter

Sonoff ZBDongle-P

### Setup

Home Assistant Core

### Debug log

info 2025-01-12 11:05:28z2m:mqtt: MQTT publish: topic 'homeassistant/sensor/0x5cc7c1fffe3fdbb9/action/config', payload ''
info 2025-01-12 11:05:28z2m:mqtt: MQTT publish: topic 'homeassistant/sensor/0x5cc7c1fffe3fdbb9/linkquality/config', payload ''
info 2025-01-12 11:05:28z2m:mqtt: MQTT publish: topic 'homeassistant/sensor/0x5cc7c1fffe3fdbb9/update_state/config', payload ''
info 2025-01-12 11:05:28z2m:mqtt: MQTT publish: topic 'homeassistant/binary_sensor/0x5cc7c1fffe3fdbb9/update_available/config', payload ''
info 2025-01-12 11:05:28z2m:mqtt: MQTT publish: topic 'homeassistant/update/0x5cc7c1fffe3fdbb9/update/config', payload ''
info 2025-01-12 11:05:43zh:controller: Interview for '0x5cc7c1fffe3fdbb9' started
info 2025-01-12 11:05:43z2m: Device '0x5cc7c1fffe3fdbb9' joined
info 2025-01-12 11:05:43z2m: Starting interview of '0x5cc7c1fffe3fdbb9'
info 2025-01-12 11:05:43z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x5cc7c1fffe3fdbb9","ieee_address":"0x5cc7c1fffe3fdbb9"},"type":"device_joined"}'
info 2025-01-12 11:05:43z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x5cc7c1fffe3fdbb9","ieee_address":"0x5cc7c1fffe3fdbb9","status":"started"},"type":"device_interview"}'
info 2025-01-12 11:05:44z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x5cc7c1fffe3fdbb9","ieee_address":"0x5cc7c1fffe3fdbb9"},"type":"device_announce"}'
info 2025-01-12 11:05:51z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x5cc7c1fffe3fdbb9","ieee_address":"0x5cc7c1fffe3fdbb9"},"type":"device_announce"}'
info 2025-01-12 11:05:54zh:controller: Succesfully interviewed '0x5cc7c1fffe3fdbb9'
info 2025-01-12 11:05:54z2m: Successfully interviewed '0x5cc7c1fffe3fdbb9', device has successfully been paired
info 2025-01-12 11:05:54z2m: Device '0x5cc7c1fffe3fdbb9' is supported, identified as: IKEA SOMRIG shortcut button (E2213)
info 2025-01-12 11:05:54z2m:mqtt: MQTT publish: topic 'homeassistant/select/0x5cc7c1fffe3fdbb9/identify/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state","value_template":"{{ value_json.state }}"}],"command_topic":"zigbee2mqtt/0x5cc7c1fffe3fdbb9/set/identify","device":{"identifiers":["zigbee2mqtt_0x5cc7c1fffe3fdbb9"],"manufacturer":"IKEA","model":"SOMRIG shortcut button (E2213)","name":"0x5cc7c1fffe3fdbb9","sw_version":"1.0.20","via_device":"zigbee2mqtt_bridge_0x00124b0029e8b084"},"device_class":"identify","enabled_by_default":false,"entity_category":"config","object_id":"0x5cc7c1fffe3fdbb9_identify","options":["identify"],"origin":{"name":"Zigbee2MQTT","sw":"1.42.0","url":"https://www.zigbee2mqtt.io"},"unique_id":"0x5cc7c1fffe3fdbb9_identify_zigbee2mqtt"}'
info 2025-01-12 11:05:54z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"definition":{"description":"SOMRIG shortcut button","exposes":[{"access":2,"category":"config","description":"Initiate device identification. This device is asleep by default.You may need to wake it up first before sending the identify command.","label":"Identify","name":"identify","property":"identify","type":"enum","values":["identify"]},{"access":5,"category":"diagnostic","description":"Remaining battery in %","label":"Battery","name":"battery","property":"battery","type":"numeric","unit":"%","value_max":100,"value_min":0},{"access":1,"category":"diagnostic","description":"Triggered action (e.g. a button click)","label":"Action","name":"action","property":"action","type":"enum","values":["1_initial_press","1_long_press","1_short_release","1_long_release","1_double_press","2_initial_press","2_long_press","2_short_release","2_long_release","2_double_press"]},{"access":1,"category":"diagnostic","description":"Link quality (signal strength)","label":"Linkquality","name":"linkquality","property":"linkquality","type":"numeric","unit":"lqi","value_max":255,"value_min":0}],"model":"E2213","options":[{"access":2,"description":"Sets the duration of the identification procedure in seconds (i.e., how long the device would flash).The value ranges from 1 to 30 seconds (default: 3).","label":"Identify timeout","name":"identify_timeout","property":"identify_timeout","type":"numeric","value_max":30,"value_min":1}],"supports_ota":true,"vendor":"IKEA"},"friendly_name":"0x5cc7c1fffe3fdbb9","ieee_address":"0x5cc7c1fffe3fdbb9","status":"successful","supported":true},"type":"device_interview"}'
info 2025-01-12 11:05:54z2m:mqtt: MQTT publish: topic 'homeassistant/button/0x5cc7c1fffe3fdbb9/identify/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state","value_template":"{{ value_json.state }}"}],"command_topic":"zigbee2mqtt/0x5cc7c1fffe3fdbb9/set/identify","device":{"identifiers":["zigbee2mqtt_0x5cc7c1fffe3fdbb9"],"manufacturer":"IKEA","model":"SOMRIG shortcut button (E2213)","name":"0x5cc7c1fffe3fdbb9","sw_version":"1.0.20","via_device":"zigbee2mqtt_bridge_0x00124b0029e8b084"},"device_class":"identify","entity_category":"config","object_id":"0x5cc7c1fffe3fdbb9_identify","origin":{"name":"Zigbee2MQTT","sw":"1.42.0","url":"https://www.zigbee2mqtt.io"},"payload_press":"identify","unique_id":"0x5cc7c1fffe3fdbb9_identify_zigbee2mqtt"}'
info 2025-01-12 11:05:54z2m:mqtt: MQTT publish: topic 'homeassistant/sensor/0x5cc7c1fffe3fdbb9/battery/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state","value_template":"{{ value_json.state }}"}],"device":{"identifiers":["zigbee2mqtt_0x5cc7c1fffe3fdbb9"],"manufacturer":"IKEA","model":"SOMRIG shortcut button (E2213)","name":"0x5cc7c1fffe3fdbb9","sw_version":"1.0.20","via_device":"zigbee2mqtt_bridge_0x00124b0029e8b084"},"device_class":"battery","enabled_by_default":true,"entity_category":"diagnostic","object_id":"0x5cc7c1fffe3fdbb9_battery","origin":{"name":"Zigbee2MQTT","sw":"1.42.0","url":"https://www.zigbee2mqtt.io"},"state_class":"measurement","state_topic":"zigbee2mqtt/0x5cc7c1fffe3fdbb9","unique_id":"0x5cc7c1fffe3fdbb9_battery_zigbee2mqtt","unit_of_measurement":"%","value_template":"{{ value_json.battery }}"}'
info 2025-01-12 11:05:54z2m:mqtt: MQTT publish: topic 'homeassistant/sensor/0x5cc7c1fffe3fdbb9/action/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state","value_template":"{{ value_json.state }}"}],"device":{"identifiers":["zigbee2mqtt_0x5cc7c1fffe3fdbb9"],"manufacturer":"IKEA","model":"SOMRIG shortcut button (E2213)","name":"0x5cc7c1fffe3fdbb9","sw_version":"1.0.20","via_device":"zigbee2mqtt_bridge_0x00124b0029e8b084"},"enabled_by_default":true,"entity_category":"diagnostic","icon":"mdi:gesture-double-tap","name":"Action","object_id":"0x5cc7c1fffe3fdbb9_action","origin":{"name":"Zigbee2MQTT","sw":"1.42.0","url":"https://www.zigbee2mqtt.io"},"state_topic":"zigbee2mqtt/0x5cc7c1fffe3fdbb9","unique_id":"0x5cc7c1fffe3fdbb9_action_zigbee2mqtt","value_template":"{{ value_json.action }}"}'
info 2025-01-12 11:05:54z2m:mqtt: MQTT publish: topic 'homeassistant/sensor/0x5cc7c1fffe3fdbb9/linkquality/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state","value_template":"{{ value_json.state }}"}],"device":{"identifiers":["zigbee2mqtt_0x5cc7c1fffe3fdbb9"],"manufacturer":"IKEA","model":"SOMRIG shortcut button (E2213)","name":"0x5cc7c1fffe3fdbb9","sw_version":"1.0.20","via_device":"zigbee2mqtt_bridge_0x00124b0029e8b084"},"enabled_by_default":false,"entity_category":"diagnostic","icon":"mdi:signal","name":"Linkquality","object_id":"0x5cc7c1fffe3fdbb9_linkquality","origin":{"name":"Zigbee2MQTT","sw":"1.42.0","url":"https://www.zigbee2mqtt.io"},"state_class":"measurement","state_topic":"zigbee2mqtt/0x5cc7c1fffe3fdbb9","unique_id":"0x5cc7c1fffe3fdbb9_linkquality_zigbee2mqtt","unit_of_measurement":"lqi","value_template":"{{ value_json.linkquality }}"}'
[... cut ...] mqtt/0x5cc7c1fffe3fdbb9","latest_version_template":"{{ value_json['update']['latest_version'] }}","latest_version_topic":"zigbee2mqtt/0x5cc7c1fffe3fdbb9","name":null,"object_id":"0x5cc7c1fffe3fdbb9","origin":{"name":"Zigbee2MQTT","sw":"1.42.0","url":"https://www.zigbee2mqtt.io"},"payload_install":"{\"id\": \"0x5cc7c1fffe3fdbb9\"}","state_topic":"zigbee2mqtt/0x5cc7c1fffe3fdbb9","unique_id":"0x5cc7c1fffe3fdbb9_update_zigbee2mqtt","value_template":"{{ value_json['update']['installed_version'] }}"}'

jaymemaurice, 2025-01-13:
In my experience with action on trådfri remote, I didn't get it working before doing the OTA update.

It wakes, updates very fast, and when you press the button you only see lqi update in states on z2m... but it does in fact update action and if you actually log out your MQTT topic by subscribing to the message bus topic, you will see an action update.  I did not see any logs in z2m at the debug level.

In the later home assistant, the open/close etc. became a device action and is not a "state"; eg you select device in the automation, the device id,

```
domain: mqtt
device_id: <your device id>
type: action
subtype: close <-- on tradfri open/close/stop
trigger: device
``` 

Now of course, I am stating my experience hoping it may be relevant...

juppw, 2025-01-13:
Same problem here, actions are missing. OTA update is done.

Aiace88, 2025-01-14:
> In my experience with action on trådfri remote, I didn't get it working before doing the OTA update.
>
> It wakes, updates very fast, and when you press the button you only see lqi update in states on [...]

I managed to do the ota update
so you say that maybe the buttons work but I don't see them in the logs?
I have other buttons from various brands, and I see all the actions in the logs..

Mitsszz, 2025-01-23:
Same problem here, cannot do OTA update as the error is "Device does not respond to OTA request".

UsefulVid, 2025-02-11:
I have 2 somrigs here. One is working (long time in use an OTA 1.0.20). and the one I just unpacked is not working. I was able to do the ota update to 1.0.21 but I still see no actions.

UsefulVid, 2025-02-12:
I may have a solution:

1. Force remove the device from z2m
2. Reboot HA
3. Factory reset somrig (4x click on the pairing button) 
4. Allow pairing only on the coordinator 
If this fails try a router device after repeating 1 - 3.

[image]

[image]

Mitsszz, 2025-02-12:
Looks like this does not work for me

UsefulVid, 2025-02-12:
Did you try it with a router?

Mitsszz, 2025-02-12:
Yes i tried with several routers and the coordinator

Cinigua, 2025-02-21:
Nice to hear 👌🏼that I could help you

gregrow, 2025-02-21:
Thanks @Cinigua ! "Home Assistant legacy action sensors" helped my SOMRIG case, as well.
BTW I've had to rejoin SOMRIG after switching "Home Assistant legacy action sensors" on.

zigbee2mqtt 2.0.0 here, SLZB-06 coordinator, SOMRIG firmware 1.0.21

Mitsszz, 2025-03-05:
Works for me to, Thanks!

EDIflyer, 2025-05-07:
Not having much luck with my new SOMRIG (E2213 FW 1.0.20) here.

Initially paired it to zigbee2mqtt OK - battery status showing and under Exposes it would instantly show button presses. However button presses then didn't show up in MQTT.

I tried activating the legacy HA devices referenced above - no change.

I then force removed it from Zigbee2MQTT and re-paired - it says it interviews OK, I get LQI showing but no battery % and N/A for the rest:

[image]

I've tried removing and re-pairing again with no luck.

As I say initially it was paired and sending info, the problem was it wasn't going across to MQTT but now I seem a set further back in that I'm not even seeing the button presses on the Exposes tab - any help much appreciated!

yorickvP, 2025-05-15:
I also had this issue on a new Somrig, but the OTA update to 1.0.21 fixed it for me. (Z2M 1.42.0)

BenjaminHae, 2025-06-14:
I'm on 1.0.21 and it is not fixed._

PoivronPoivreux, 2025-07-09:
Hello, hello. I have the same problem on my side. For me, the Action field first goes through the action performed and then changes to “Empty string ("")” (See GIF)

[image]

Here are the debug logs:

info 2025-07-09 21:11:30z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/0xecf64cfffe327f0a', payload '{"battery":100,"linkquality":90,"update":{"installed_version":16777249,"latest_version":16777249,"state":"idle"},"voltage":1600}'
info 2025-07-09 21:15:57z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/health', payload '{"response_time":1752088557601,"os":{"load_average":[0.44,0.48,0.48],"memory_used_mb":592.77,"memory_percent":34.6601},"process":{"uptime_sec":613,"memory_used_mb":106.5,"memory_percent":11.7388},"mqtt":{"connected":true,"queued":0,"published":221,"received":101},"devices":{"0xecf64cfffe327f0a":{"messages":67,"messages_per_sec":0.1162,"leave_count":1,"network_address_changes":0}}}'
info 2025-07-09 21:17:44z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/0xecf64cfffe327f0a', payload '{"action":"1_initial_press","battery":100,"linkquality":93,"update":{"installed_version":16777249,"latest_version":16777249,"state":"idle"},"voltage":1600}'
info 2025-07-09 21:17:44z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/0xecf64cfffe327f0a', payload '{"action":"","battery":100,"linkquality":93,"update":{"installed_version":16777249,"latest_version":16777249,"state":"idle"},"voltage":1600}'
info 2025-07-09 21:17:44z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/0xecf64cfffe327f0a/action', payload '1_initial_press'
info 2025-07-09 21:17:45z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/0xecf64cfffe327f0a', payload '{"action":"1_short_release","battery":100,"linkquality":93,"update":{"installed_version":16777249,"latest_version":16777249,"state":"idle"},"voltage":1600}'
info 2025-07-09 21:17:45z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/0xecf64cfffe327f0a', payload '{"action":"","battery":100,"linkquality":93,"update":{"installed_version":16777249,"latest_version":16777249,"state":"idle"},"voltage":1600}'
info 2025-07-09 21:17:45z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/0xecf64cfffe327f0a/action', payload '1_short_release'
info 2025-07-09 21:17:46z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/0xecf64cfffe327f0a', payload '{"action":"2_initial_press","battery":100,"linkquality":93,"update":{"installed_version":16777249,"latest_version":16777249,"state":"idle"},"voltage":1600}'
info 2025-07-09 21:17:46z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/0xecf64cfffe327f0a', payload '{"action":"","battery":100,"linkquality":93,"update":{"installed_version":16777249,"latest_version":16777249,"state":"idle"},"voltage":1600}'
info 2025-07-09 21:17:46z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/0xecf64cfffe327f0a/action', payload '2_initial_press'
info 2025-07-09 21:17:47z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/0xecf64cfffe327f0a', payload '{"action":"2_short_release","battery":100,"linkquality":93,"update":{"installed_version":16777249,"latest_version":16777249,"state":"idle"},"voltage":1600}'
info 2025-07-09 21:17:47z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/0xecf64cfffe327f0a', payload '{"action":"","battery":100,"linkquality":93,"update":{"installed_version":16777249,"latest_version":16777249,"state":"idle"},"voltage":1600}'
info 2025-07-09 21:17:47z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/0xecf64cfffe327f0a/action', payload '2_short_release'

acsel91, 2025-08-30:
[45df7312_zigbee2mqtt_2025-08-30T16-43-05.302Z.log](https://github.com/user-attachments/files/22059371/45df7312_zigbee2mqtt_2025-08-30T16-43-05.302Z.log)

@Koenkk 

let me know if this logfile is correct format 

i activated the legacy sensor mode, actions are showed

davidsmoker, 2025-09-05:
[5-9-2025, 06:52:57] z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"definition":{"description":"SOMRIG shortcut button","exposes":[{"access":2,"category":"config","description":"Initiate device identification. This device is asleep by default.You may need to wake it up first before sending the identify command.","label":"Identify","name":"identify","property":"identify","type":"enum","values":["identify"]},{"access":5,"category":"diagnostic","description":"Remaining battery in %","label":"Battery","name":"battery","property":"battery","type":"numeric","unit":"%","value_max":100,"value_min":0},{"access":5,"category":"diagnostic","description":"Reported battery voltage in millivolts","label":"Voltage","name":"voltage","property":"voltage","type":"numeric","unit":"mV"},{"access":1,"category":"diagnostic","description":"Triggered action (e.g. a button click)","label":"Action","name":"action","property":"action","type":"enum","values":["1_initial_press","1_long_press","1_short_release","1_long_release","1_double_press","2_initial_press","2_long_press","2_short_release","2_long_release","2_double_press"]},{"access":1,"category":"diagnostic","description":"Link quality (signal strength)","label":"Linkquality","name":"linkquality","property":"linkquality","type":"numeric","unit":"lqi","value_max":255,"value_min":0}],"model":"E2213","options":[{"access":2,"description":"Sets the duration of the identification procedure in seconds (i.e., how long the device would flash).The value ranges from 1 to 30 seconds (default: 3).","label":"Identify timeout","name":"identify_timeout","property":"identify_timeout","type":"numeric","value_max":30,"value_min":1}],"source":"native","supports_ota":true,"vendor":"IKEA"},"friendly_name":"0x8c8b48fffe1bd61d","ieee_address":"0x8c8b48fffe1bd61d","status":"successful","supported":true},"type":"device_interview"}'
[5-9-2025, 06:52:57] z2m:mqtt: MQTT publish: topic 'homeassistant/sensor/0x8c8b48fffe1bd61d/battery/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state","value_template":"{{ value_json.state }}"},{"topic":"zigbee2mqtt/0x8c8b48fffe1bd61d/availability","value_template":"{{ value_json.state }}"}],"availability_mode":"all","device":{"hw_version":1,"identifiers":["zigbee2mqtt_0x8c8b48fffe1bd61d"],"manufacturer":"IKEA","model":"SOMRIG shortcut button","model_id":"E2213","name":"0x8c8b48fffe1bd61d","sw_version":"1.0.20","via_device":"zigbee2mqtt_bridge_0x00124b00258d3d34"},"device_class":"battery","enabled_by_default":true,"entity_category":"diagnostic","object_id":"0x8c8b48fffe1bd61d_battery","origin":{"name":"Zigbee2MQTT","sw":"2.6.1","url":"https://www.zigbee2mqtt.io"},"state_class":"measurement","state_topic":"zigbee2mqtt/0x8c8b48fffe1bd61d","unique_id":"0x8c8b48fffe1bd61d_battery_zigbee2mqtt","unit_of_measurement":"%","value_template":"{{ value_json.battery }}"}'
[5-9-2025, 06:52:57] z2m:mqtt: MQTT publish: topic 'homeassistant/sensor/0x8c8b48fffe1bd61d/voltage/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state","value_template":"{{ value_json.state }}"},{"topic":"zigbee2mqtt/0x8c8b48fffe1bd61d/availability","value_template":"{{ value_json.state }}"}],"availability_mode":"all","device":{"hw_version":1,"identifiers":["zigbee2mqtt_0x8c8b48fffe1bd61d"],"manufacturer":"IKEA","model":"SOMRIG shortcut button","model_id":"E2213","name":"0x8c8b48fffe1bd61d","sw_version":"1.0.20","via_device":"zigbee2mqtt_bridge_0x00124b00258d3d34"},"device_class":"voltage","enabled_by_default":true,"entity_category":"diagnostic","object_id":"0x8c8b48fffe1bd61d_voltage","origin":{"name":"Zigbee2MQTT","sw":"2.6.1","url":"https://www.zigbee2mqtt.io"},"state_class":"measurement","state_topic":"zigbee2mqtt/0x8c8b48fffe1bd61d","unique_id":"0x8c8b48fffe1bd61d_voltage_zigbee2mqtt","unit_of_measurement":"mV","value_template":"{{ value_json.voltage }}"}'
[5-9-2025, 06:52:57] z2m:mqtt: MQTT publish: topic 'homeassistant/sensor/0x8c8b48fffe1bd61d/action/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state","value_template":"{{ value_json.state }}"},{"topic":"zigbee2mqtt/0x8c8b48fffe1bd61d/availability","value_template":"{{ value_json.state }}"}],"availability_mode":"all","device":{"hw_version":1,"identifiers":["zigbee2mqtt_0x8c8b48fffe1bd61d"],"manufacturer":"IKEA","model":"SOMRIG shortcut button","model_id":"E2213","name":"0x8c8b48fffe1bd61d","sw_version":"1.0.20","via_device":"zigbee2mqtt_bridge_0x00124b00258d3d34"},"entity_category":"diagnostic","icon":"mdi:gesture-double-tap","name":"Action","object_id":"0x8c8b48fffe1bd61d_action","origin":{"name":"Zigbee2MQTT","sw":"2.6.1","url":"https://www.zigbee2mqtt.io"},"state_topic":"zigbee2mqtt/0x8c8b48fffe1bd61d","unique_id":"0x8c8b48fffe1bd61d_action_zigbee2mqtt","value_template":"{{ value_json.action }}"}'
[5-9-2025, 06:52:57] z2m:mqtt: MQTT publish: topic 'homeassistant/sensor/0x8c8b48fffe1bd61d/linkquality/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state","value_template":"{{ value_json.state }}"},{"topic":"zigbee2mqtt/0x8c8b48fffe1bd61d/availability","value_template":"{{ value_json.state }}"}],"availability_mode":"all","device":{"hw_version":1,"identifiers":["zigbee2mqtt_0x8c8b48fffe1bd61d"],"manufacturer":"IKEA","model":"SOMRIG shortcut button","model_id":"E2213","name":"0x8c8b48fffe1bd61d","sw_version":"1.0.20","via_device":"zigbee2mqtt_bridge_0x00124b00258d3d34"},"enabled_by_default":false,"entity_category":"diagnostic","icon":"mdi:signal","name":"Li [... cut ...]
[5-9-2025, 06:53:22] z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/LICHTEN_HUE_WALL_8', payload '{"brightness":254,"color":{"h":23,"hue":23,"s":70,"saturation":70,"x":0.439,"y":0.374},"color_mode":"xy","color_temp":370,"effect":null,"last_seen":"2025-09-05T04:53:22.765Z","linkquality":69,"power_on_behavior":null,"state":"OFF","update":{"installed_version":16786944,"latest_version":16786944,"state":"idle"},"update_available":false}'
[5-9-2025, 06:53:22] z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/PLUG_TOUCHSCREEN_KEUKEN_DISPLAY', payload '{"last_seen":"2025-

davidsmoker, 2025-09-05:
there is now way to wake up the device

UsefulVid, 2025-09-05:
While connection you could try to power off all other zigbee router devices. My problem seemed to be another device in the zigbee mesh network while pairing.

Koenkk, 2025-09-05:
> there is now way to wake up the device

Try waking it up by pressing a button on it right before clicking the configure button.

davidsmoker, 2025-09-06:
@Koenkk I’ve already done that several times, but just to be sure I tried it again with two of these buttons. The result is always the same: I can pair them, but I can’t control or wake them up.

With around 180 devices in my Zigbee network, this is actually the first time I’ve had this much trouble with a product.

BenjaminHae, 2025-09-06:
@davidsmoker my somrig behaves exactly like yours. I bought it in October 2024 and never got it to work. Is yours a newer model?

closed 2025-12-19 by github-actions[bot]

arnoldxt, 2026-03-29:
> I bought my IKEA E2213 (SOMRIG) this week and immediately ran into the same problem: The switch paired, but no actions were registered, and I was getting the Failed to configure timeout errors.
> 
> This happens because the battery-powered switch falls asleep before Z2M can finish the initial binding and reporting setup.
> 
> **The Fix: Keep the Device Awake During Reconfiguration** The solution is to force the device to stay awake while Zigbee2MQTT (Z2M) completes the configuration.
> 
> 1. Remove & Reset: Completely remove the E2213 from Z2M, then factory reset the switch (press the internal pairing button 4 times quickly).
> 2. Re-Pair: Pair the switch again.
> 3. The Critical Step: As soon as the device appears in Z2M, start pressing both outer buttons repeatedly and alternatingly (every 1-2 seconds).
> 4. Reconfigure: While continuously pressing the buttons, click the "Reconfigure" button on the device's page in the Z2M web interface.
> 5. Hold It: Do not stop pressing the buttons until the configuration is confirmed as successful (or the loading icon disappears).
> 
> This process forces Z2M to successfully set up all necessary bindings and reporting, and your actions (1_single, 2_single, etc.) should now appear correctly!

that helped me. Thanks a lot
but it still took me 10 ten times to get it working