Koenkk/zigbee2mqtt #25786: set state almost always fails with BUFFER_FULL error

dijk039, 2025-01-13:
### What happened?

During the past weeks I noticed that some automations didn't work correctly intermittently. After some troubleshooting I noticed regular BUFFER_FULL error messages in the logs. Since these automation failures were internittend (days without issues) I didn't pursue errors it activerly.

After upgrading today to version 2.0.0-2 today, all my Zigbee devices were unusable. Every time I attempted to manually switch on 
a lidl light bulb or ikea switch for example, the BUFFER_FULL error message appears. My Zigbee network is unusable now.

I do still get regular updates from my Aqara and Sonoff climate sensors and also powermetering data from for Ikea INSPELNING switches

Reverting back to a (VMware ESXi) snapshot, which I made prior to upgrading from z2m 1.42.0-2 to 2.0.0-2 resulted in the same error messages and unusable lights and switches.

In the z2m logs I see the following log lines appear when attempting to switch on an Ikea light bulb:

```
[2025-01-13 12:56:57] debug: 	z2m:mqtt: Received MQTT message on 'zigbee2mqtt/Zolderkamer lampje/set' with data '{"state":"ON"}'
[2025-01-13 12:56:57] debug: 	z2m: Publishing 'set' 'state' to 'Zolderkamer lampje'
[2025-01-13 12:56:57] debug: 	zh:controller:endpoint: ZCL command 0x000b57fffeeb0505/1 genOnOff.on({}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"reservedBits":0,"writeUndiv":false})
[2025-01-13 12:56:57] debug: 	zh:zstack: sendZclFrameToEndpointInternal 0x000b57fffeeb0505:15678/1 (0,0,1)
[2025-01-13 12:56:57] debug: 	zh:zstack:znp: --> SREQ: AF - dataRequest - {"dstaddr":15678,"destendpoint":1,"srcendpoint":1,"clusterid":6,"transid":44,"options":0,"radius":30,"len":3,"data":{"type":"Buffer","data":[1,10,1]}}
[2025-01-13 12:56:57] debug: 	zh:zstack:unpi:writer: --> frame [254,13,36,1,62,61,1,1,6,0,44,0,30,3,1,10,1,22]
[2025-01-13 12:56:57] debug: 	zh:zstack:unpi:parser: <-- [254,1,100,1,17,117]
[2025-01-13 12:56:57] debug: 	zh:zstack:unpi:parser: --- parseNext [254,1,100,1,17,117]
[2025-01-13 12:56:57] debug: 	zh:zstack:unpi:parser: --> parsed 1 - 3 - 4 - 1 - [17] - 117
[2025-01-13 12:56:57] debug: 	zh:zstack:znp: <-- SRSP: AF - dataRequest - {"status":17}
[2025-01-13 12:56:57] debug: 	zh:zstack:unpi:parser: --- parseNext []
[2025-01-13 12:56:57] debug: 	zh:controller:endpoint: Error: ZCL command 0x000b57fffeeb0505/1 genOnOff.on({}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"reservedBits":0,"writeUndiv":false}) failed (--> 'SREQ: AF - dataRequest - {"dstaddr":15678,"destendpoint":1,"srcendpoint":1,"clusterid":6,"transid":44,"options":0,"radius":30,"len":3,"data":{"type":"Buffer","data":[1,10,1]}}' failed with status '(0x11: BUFFER_FULL)' (expected '(0x00: SUCCESS)'))
    at /app/node_modules/zigbee-herdsman/src/adapter/z-stack/znp/znp.ts:291:27
    at Queue.execute (/app/node_modules/zigbee-herdsman/src/utils/queue.ts:36:20)
    at ZStackAdapter.dataRequest (/app/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:1034:9)
    at ZStackAdapter.sendZclFrameToEndpointInternal (/app/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:488:35)
    at /app/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:422:20
    at Queue.execute (/app/node_modules/zigbee-herdsman/src/utils/queue.ts:36:20)
    at ZStackAdapter.sendZclFrameToEndpoint (/app/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:420:16)
    at Request.send (/app/node_modules/zigbee-herdsman/src/controller/helpers/request.ts:82:20)
    at Endpoint.sendRequest (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:325:20)
    at Endpoint.zclCommand (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:951:28)
[2025-01-13 12:56:57] error: 	z2m: Publish 'set' 'state' to 'Zolderkamer lampje' failed: 'Error: ZCL command 0x000b57fffeeb0505/1 genOnOff.on({}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"reservedBits":0,"writeUndiv":false}) failed (--> 'SREQ: AF - dataRequest - {"dstaddr":15678,"destendpoint":1,"srcendpoint":1,"clusterid":6,"transid":44,"options":0,"radius":30,"len":3,"data":{"type":"Buffer","data":[1,10,1]}}' failed with status '(0x11: BUFFER_FULL)' (expected '(0x00: SUCCESS)'))'
[2025-01-13 12:56:57] debug: 	z2m: Error: ZCL command 0x000b57fffeeb0505/1 genOnOff.on({}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"reservedBits":0,"writeUndiv":false}) failed (--> 'SREQ: AF - dataRequest - {"dstaddr":15678,"destendpoint":1,"srcendpoint":1,"clusterid":6,"transid":44,"options":0,"radius":30,"len":3,"data":{"type":"Buffer","data":[1,10,1]}}' failed with status '(0x11: BUFFER_FULL)' (expected '(0x00: SUCCESS)'))
    at /app/node_modules/zigbee-herdsman/src/adapter/z-stack/znp/znp.ts:291:27
    at Queue.execute (/app/node_modules/zigbee-herdsman/src/utils/queue.ts:36:20)
    at ZStackAdapter.dataRequest (/app/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:1034:9)
    at ZStackAdapter.sendZclFrameToEndpointInternal (/app/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:488:35)
    at /app/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:422:20
    at Queue.execute (/app/node_modules/zigbee-herdsman/src/utils/queue.ts:36:20)
    at ZStackAdapter.sendZclFrameToEndpoint (/app/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:420:16)
    at Request.send (/app/node_modules/zigbee-herdsman/src/controller/helpers/request.ts:82:20)
    at Endpoint.sendRequest (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:325:20)
    at Endpoint.zclCommand (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:951:28)
```

### What did you expect to happen?

nothing realy; Hoping the BUFFER_FULL mesages would disapear at some point in time.

### How to reproduce it (minimal and precise)

Everytime I start z2m results in the same end-result.
Before the upgrade 1.42.0-2 to 2.0.0-2 BUFFER_FULL messages appeared sometimes; now they are all the time (3 hours after the upgrade / downgrade)

### Zigbee2MQTT version

2.0.0-2

### Adapter firmware version

20240710

### Adapter

SONOFF Zigbee 3.0 USB Dongle Plus by ITead | Coördinator type: zStack3x0

### Setup

Add-on on Supervised Home Assistant 2025-01-0 on NUC (ESXi)

### Debug log

[45df7312_zigbee2mqtt_2025-01-13T12-05-31.741Z.log](https://github.com/user-attachments/files/18396490/45df7312_zigbee2mqtt_2025-01-13T12-05-31.741Z.log)

dijk039, 2025-01-13:
**Update:**

Since my issue does not seem to be related to the 2.0.0-2 upgrade, I again upgraded tot 2.0.0-2. After the upgrade to 2.0.0-2 and restarting HASS/Z2M, I noticed that the problem was still not resolved and decided to shutdown the HASS VM and re-seat the Sonoff USB Zigbee USB coordinator. This solved my problem!

Some context: I use VMware ESXi (version 7) with an HASS VM running on HassOS 14.1 and Supervisor 2024.12.3. The Sonoff USB coordinator is connected to the VM using an USB device mapping. Since restarting the VM doesn't power cycle the USB device, I figured to shutdown the VM and manually re-seat (effectively power-cycle) the Sonoff USB coordinator. This seams to have solved my problem (for now). After powering on the VM and waiting for all HASS + Z2M services to be fully started, all devices in my network were usable again (pheww).

I find it strange though that this happend. I've run HASS with the Z2M  Add-on for over 3 years now. I did encounter some issues over time, but never did I have to re-seat the USB coordinator.

Any ideas why this happend?

jdbrookes, 2025-01-14:
I've had this BUFFER_FULL issue once or twice with my own Z2M setup and Sonoff stick (original P model). It was resolved the first time by plugging and unplugging the device and restarting Z2M, but in one spectacular failure last year I needed to reflash the firmware to the device. In that case, BUFFER_FULL appeared in the logs after I attempted to join a new device, and devices began to drop off the network. Even after unplug and restart, Z2M couldn't toggle device joining (it seemed to be stuck in ON) and the BUFFER_FULL errors were consistent. So i had to flash the same firmware to the stick on my laptop, and only then once reconnected did the network stabilise. Luckily I didn't have to rejoin all of my devices that time.

Koenkk, 2025-01-14:
Could you try with this firmware: https://github.com/Koenkk/Z-Stack-firmware/releases/tag/Z-Stack_3.x.0_coordinator_20221226 ?

Koenkk, 2025-01-15:
@dijk039 if you cannot reproduce this issue then it doesn't make sense to try the other fw. Likely there was some connection issue.

closed 2025-01-15 by Koenkk