Koenkk/zigbee2mqtt #25789: [New device support]: TS000F _TZ3000_hdc8bbha

agnus777, 2025-01-13:
### Link

https://it.aliexpress.com/item/1005006993707388.html?spm=a2g0o.order_list.order_list_main.23.717d36962vgy7M&gatewayAdapt=glo2ita#nav-specification

### Database entry

{"id":32,"type":"Router","ieeeAddr":"0xa4c138641257fe7f","nwkAddr":39647,"manufId":4417,"manufName":"_TZ3000_hdc8bbha","powerSource":"Mains (single phase)","modelId":"TS000F","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":256,"inClusterList":[4,5,6,57344,57345,0],"outClusterList":[25,10],"clusters":{"57344":{"attributes":{"53251":"AAAA"}},"genBasic":{"attributes":{"stackVersion":0,"dateCode":"","zclVersion":3,"modelId":"TS000F","manufacturerName":"_TZ3000_hdc8bbha","powerSource":1,"appVersion":112,"hwVersion":1}},"genOnOff":{"attributes":{"onOff":0,"onTime":0,"offWaitTime":0,"moesStartUpOnOff":0,"tuyaBacklightMode":1}},"manuSpecificTuya_3":{"attributes":{"powerOnBehavior":0}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0xe0798dfffe77c68c","endpointID":1}],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":112,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":332242049},"lastSeen":1736774303949}

### Zigbee2MQTT version

2.0.0

### Comments

I did the pairing on the fly and z2m recognized this device incorrectly (as https://www.zigbee2mqtt.io/devices/QS-Zigbee-SEC01-U.html#tuya-qs-zigbee-sec01-u )
My device has a dry contact that I wanted to use for a garage door, but the momentary function in the switch type does not react to changes of state in drop menu in any case.
Is there a way to update the definition for this device to a compatible mode that works with the momentary button correctly?

### External definition

```shell
const {onOff} = require('zigbee-herdsman-converters/lib/modernExtend');

const definition = {
    zigbeeModel: ['TS000F'],
    model: 'TS000F',
    vendor: '_TZ3000_hdc8bbha',
    description: 'Automatically generated definition',
    extend: [onOff({"powerOnBehavior":false})],
    meta: {},
};

module.exports = definition;
```

### What does/doesn't work with the external definition?

turn on-off works
choosing of Switch type doesn't work

agnus777, 2025-01-23:
I buyed 2 of this devices, the second one, with updated firmware from tuya, it's no more recognized by Z2M... what's wrong?

[image]

[image]

[image]

Moreover the model recognized in the first one is QS-Zigbee-SEC01-U but should be QS-Zigbee-SEC01-DC

Any chance to have a correct converter for these?

agnus777, 2025-01-23:
**Link**
https://it.aliexpress.com/item/1005006993707388.html?spm=a2g0o.order_list.order_list_main.23.717d36962vgy7M&gatewayAdapt=glo2ita#nav-specification

[image]

**Database entry**
{"id":34,"type":"Router","ieeeAddr":"0xa4c1381782017f20","nwkAddr":10836,"manufId":4417,"manufName":"_TZ3218_hdc8bbha","powerSource":"Mains (single phase)","modelId":"TS000F","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":256,"inClusterList":[0,3,4,5,6,57344,61184],"outClusterList":[10,25],"clusters":{"genBasic":{"attributes":{"dateCode":"","zclVersion":3}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":260,"epId":242,"devId":256,"inClusterList":[0,3,4,5,6,57344,61184],"outClusterList":[10,25],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":128,"stackVersion":1,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1737638289655}

Zigbee2MQTT version
2.0.0

**Comments**
device has a dry contact

**External definition**

const {onOff} = require('zigbee-herdsman-converters/lib/modernExtend');

const definition = {
    zigbeeModel: ['TS000F'],
    model: 'TS000F',
    vendor: '_TZ3218_hdc8bbha',
    description: 'Automatically generated definition',
    extend: [onOff({"powerOnBehavior":false,"endpointNames":["1","242"]})],
    meta: {},
};

module.exports = definition;

**What does/doesn't work with the external definition?**

not recognized

metyrlost, 2025-02-06:
I  have the same problem. Device is not reckognized in Zigbee2mqtt.

ahmetem, 2025-02-19:
I installed the dev(edge) version for this fix. But there was no change. It still remains with its old definition.  
I uninstalled and reinstalled but it still shows as QS-Zigbee-SEC01-U. what should I do ?

chrishoage, 2025-04-24:
Still an issue for me

chrishoage, 2025-06-24:
Not stale bump