Koenkk/zigbee2mqtt #25791: Philips Hue Smart Button, toggle light on off, wrong action

feluxe, 2025-01-13:
### What happened?

I have two "Philips Hue Smart Buttons" bound to a group "Kitchen Light Group".

When the lights are off and I press the the button `Kitchen Switch Button All` the lights toggle on as expected.

Now when I press the other button `Kitchen Switch Button All 2` the lights won't toggle off. I have to press it a second time in order to turn off the lights.

### What did you expect to happen?

I expected the lights to turn off when I press the second button once.

### How to reproduce it (minimal and precise)

1. Create a group in zigbee2mqtt.
2. Connect lights and add them to the group.
3. Make sure reporting for on/off is set for each light.
4. Connect two Philips Hue Smart Buttons.
5. Bind the on/off cluster of each Smart Button to the group.
6. Add both Smart Buttons to the group
7. Press the first button to turn on the lights
8. Press the second button to turn off the lights. (They won't turn off)

This is what I have currently set:

Group:

[image]

Button 1 Bindings:

[image]

Button 2 Bindings:

[image]

Light Reporting (the others are the same):

[image]

### Zigbee2MQTT version

 1.41.0

### Adapter firmware version

7.4.1 [GA]

### Adapter

EmberZNet

### Setup

Docker on Rocky Linux

### Debug log

Log after button 1 pressed:

```
debug 2025-01-13 15:21:12z2m: Received Zigbee message from 'Kitchen Switch Button All', type 'commandHueNotification', cluster 'manuSpecificPhilips', data '{"button":1,"time":0,"type":0,"unknown1":3145728,"unknown2":0}' from endpoint 1 with groupID 39665
info 2025-01-13 15:21:12z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Kitchen Switch Button All', payload '{"action":"press","battery":100,"linkquality":200,"update":{"installed_version":-1,"latest_version":-1,"state":null},"update_available":null}'
info 2025-01-13 15:21:12z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Kitchen Switch Button All', payload '{"action":"","battery":100,"linkquality":200,"update":{"installed_version":-1,"latest_version":-1,"state":null},"update_available":null}'
info 2025-01-13 15:21:12z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Kitchen Switch Button All/action', payload 'press'
debug 2025-01-13 15:21:12z2m: Received Zigbee message from 'Kitchen Switch Button All', type 'commandHueNotification', cluster 'manuSpecificPhilips', data '{"button":1,"time":1,"type":2,"unknown1":3145728,"unknown2":0}' from endpoint 1 with groupID 39665
info 2025-01-13 15:21:12z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Kitchen Switch Button All', payload '{"action":"release","battery":100,"linkquality":172,"update":{"installed_version":-1,"latest_version":-1,"state":null},"update_available":null}'
info 2025-01-13 15:21:12z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Kitchen Switch Button All', payload '{"action":"","battery":100,"linkquality":172,"update":{"installed_version":-1,"latest_version":-1,"state":null},"update_available":null}'
info 2025-01-13 15:21:12z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Kitchen Switch Button All/action', payload 'release'
[... 13 lines cut ...]
```
Log after button 2 pressed:
```
debug 2025-01-13 15:21:26z2m: Received Zigbee message from 'Kitchen Switch Button All 2', type 'commandHueNotification', cluster 'manuSpecificPhilips', data '{"button":1,"time":0,"type":0,"unknown1":3145728,"unknown2":0}' from endpoint 1 with groupID 0
info 2025-01-13 15:21:26z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Kitchen Switch Button All 2', payload '{"action":"press","battery":100,"linkquality":140,"update":{"installed_version":-1,"latest_version":-1,"state":null},"update_available":null}'
info 2025-01-13 15:21:26z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Kitchen Switch Button All 2', payload '{"action":"","battery":100,"linkquality":140,"update":{"installed_version":-1,"latest_version":-1,"state":null},"update_available":null}'
info 2025-01-13 15:21:26z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Kitchen Switch Button All 2/action', payload 'press'
debug 2025-01-13 15:21:26z2m: Received Zigbee message from 'Kitchen Switch Button All 2', type 'commandHueNotification', cluster 'manuSpecificPhilips', data '{"button":1,"time":0,"type":0,"unknown1":3145728,"unknown2":0}' from endpoint 1 with groupID 39665
info 2025-01-13 15:21:26z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Kitchen Switch Button All 2', payload '{"action":"press","battery":100,"linkquality":140,"update":{"installed_version":-1,"latest_version":-1,"state":null},"update_available":null}'
info 2025-01-13 15:21:26z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Kitchen Switch Button All 2', payload '{"action":"","battery":100,"linkquality":140,"update":{"installed_version":-1,"latest_version":-1,"state":null},"update_available":null}'
info 2025-01-13 15:21:26z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Kitchen Switch Button All 2/action', payload 'press'
debug 2025-01-13 15:21:26z2m: Received Zigbee message from 'Kitchen Switch Button All 2', type 'commandHueNotification', cluster 'manuSpecificPhilips', data '{"button":1,"time":1,"type":2,"unknown1":3145728,"unknown2":0}' from endpoint 1 with groupID 0
info 2025-01-13 15:21:26z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Kitchen Switch Button All 2', payload '{"action":"release","battery":100,"linkquality":140,"update":{"installed_version":-1,"latest_version":-1,"state":null},"update_available":null}'
info 2025-01-13 15:21:26z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Kitchen Switch Button All 2', payload '{"action":"","battery":100,"linkquality":140,"update":{"installed_version":-1,"latest_version":-1,"state":null},"update_available":null}'
info 2025-01-13 15:21:26z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Kitchen Switch Button All 2/action', payload 'release'
debug 2025-01-13 15:21:26z2m: Received Zigbee message from 'Kitchen Switch Button All 2', type 'commandHueNotification', cluster 'manuSpecificPhilips', data '{"button":1,"time":1,"type":2,"unknown1":3145728,"unknown2":0}' from endpoint 1 with groupID 39665
info 2025-01-13 15:21:26z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Kitchen Switch Button All 2', payload '{"action":"release","battery":100,"linkquality":140,"update":{"installed_version":-1,"latest_version":-1,"state":null},"update_available":null}'
info 2025-01-13 15:21:26z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Kitchen Switch Button All 2', payload '{"action":"","battery":100,"linkquality":140,"update":{"installed_version":-1,"latest_version":-1,"state":null},"update_available":null}'
info 2025-01-13 15:21:26z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Kitchen Switch Button All 2/action', payload 'release'
debug 2025-01-13 15:21:26z2m: Received Zigbee message from 'Kitchen Switch Button All 2', type 'commandOn', cluster 'genOnOff', data '{}' from endpoint 1 with groupID 0
info 2025-01-13 15:21:26z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Kitchen Switch Button All 2', payload '{"action":"on","battery":100,"linkquality":144,"update":{"installed_version":-1,"latest_version":-1,"state":null},"update_available":null}'
info 2025-01-13 15:21:26z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Kitchen Switch Button All 2', payload '{"action":"","battery":100,"linkquality":144,"update":{"installed_version":-1,"latest_version":-1,"state":null},"update_available":null}'
info 2025-01-13 15:21:26z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Kitchen Switch Button All 2/action', payload 'on'
debug 2025-01-13 15:21:26z2m: Received Zigbee message from 'Kitchen Switch Button All 2', type 'commandOn', cluster 'genOnOff', data '{}' from endpoint 1 with groupID 39665
```

feluxe, 2025-03-06:
You are right.

I was using a Hue Bridge before, where they worked as a toggle switch, which was much better, in my opinion. 

Does anyone know why they use on/off instead of toggle? It's a single button, so wouldn't a toggle make more sense? 

Would it be possible to implement an option for the Smart Buttons to make them act as a toggle button?

f-mosquera, 2026-09-20:
Hi, I have been experiencing the same situation and thought it was a problem with zigbee2mqtt as I had been using previously deconz and it worked as expected as a toggle switch (I have three buttons controlling one light). As it's a button and not a switch, I would have expected a toggle command instead of the on-off sequential commands currently implemented. Is it possible to implement this fix at some point? Thanks in advance.