Koenkk/zigbee2mqtt #25796: From Z2M 2.0 EKO09716 periodically offline

devildant, 2025-01-13:
### What happened?

since I updated to Z2M 2.0 all my EKO09716 sockets disconnect in cascade once a day, I have to either disconnect them electrically or re-pair them...

### Zigbee2MQTT version

2.0.0-2

### Adapter firmware version

20210708

### Adapter

sonoff P

### Setup

HAOS rpi 4

Enf0, 2025-01-13:
I'm seeing the same with all 5 of my [316GLEDRF](https://www.zigbee2mqtt.io/devices/316GLEDRF.html#elko-316gledrf) wall dimmers (same vendor). They were working for a few days after upgrading to 2.0, now they suddenly all went offline.

devildant, 2025-01-13:
I'm not the only one, that reassures me, everything worked for 2 years without any problems, but since the upgrade to 2.0 it's been a pain. The only plug that doesn't disconnect is the one connected to my fridge, because the current is constantly drawn... the others as soon as they are stopped without power they systematically disconnect (few hours), I've had to re appear them every day for a week... I tried with new ones, same result

devildant, 2025-01-13:
But for me the vendor is schneider electric

devildant, 2025-01-13:
observation: the devices do not respond, they do not come back available with a manual action, the on/off link buttons in zigbee connection no longer work. only the re paire, or a power outage. the z2m must send or no longer send something… the sockets crash completely, it worked well before the switch to 2.0, @Koenkk there were changes in the Schneider Electric implementation for these sockets?

Enf0, 2025-01-14:
> But for me the vendor is schneider electric

Schneider Electric and Elko seems to be the same thing. Searching for your model gives a bunch of Elko results for me, so it's probably just dependent on the country.

devildant, 2025-01-15:
I tried to do a regular ping genOnOff with an automation, but to no avail my sockets end up offline :(, I have no other idea

devildant, 2025-01-15:
I checked the logs but all that appears is that the socket has stopped responding with a timeout, they have completely crashed, only the physical button works, the zigbee bind made with other buttons no longer works, that whether the new ones or the old ones I have the same behavior

devildant, 2025-01-17:
I tried to create several reports to try to maintain but nothing to do, I don't know what to do

devildant, 2025-01-25:
observation if I put a zigbee socket (a1z us) right next to the EKO09716, they no longer disconnect, but as soon as they are alone they end up offline < 24h it's as if they weren't routers... I don't understand, they worked for 2 years without problem, it's following the z2m 2.0 that I have this problem...

@Koenkk do you have any idea? I'm starting to despair :(

devildant, 2025-02-03:
Someone?

devildant, 2025-02-13:
ping, someone ?

devildant, 2025-04-15:
problem still present

closed 2025-06-23 by github-actions[bot]