Koenkk/zigbee2mqtt #25800: [New device support]: Tuya 4 gang smart switch  (model: 'TS0601',  vendor: '_TZE204_unsxl4ir' )

gersonfer, 2025-01-13:
### Link

https://www.novadigitalsmart.com.br/produtos/interruptor-inteligente-tecla-fisica-4x4-fzb-4-6---w-b

### Database entry

{"id":11,"type":"Router","ieeeAddr":"0xa4c138a67e247cf0","nwkAddr":60421,"manufId":4417,"manufName":"_TZE204_unsxl4ir","powerSource":"Mains (single phase)","modelId":"TS0601","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[4,5,61184,0],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"65503":"\f\u0000\u0000\u0000i\f\u0000\u0000\u0000i\r\u0000\u0000\u0000i\r\u0000\u0000\u0000i\u000e\u0000\u0000\u0000i\u000e\u0000\u0000\u0000i\u000f\u0000\u0000\u0000i\u000f\u0000\u0000\u0000i","65506":56,"65508":0,"65534":0,"modelId":"TS0601","manufacturerName":"_TZE204_unsxl4ir","powerSource":1,"zclVersion":3,"appVersion":74,"stackVersion":0,"hwVersion":1,"dateCode":""}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":74,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1736797817199}

### Zigbee2MQTT version

1.41.0 

### Comments

Hello, Everyone !
I tried to follow the instructions on how to add support but failed. 

At interview the messages were:

[2025-01-11 11:42:07] info: 	zh:controller: Interview for '0xa4c138a67e247cf0' started
[2025-01-11 11:42:07] info: 	z2m: Device '0xa4c138a67e247cf0' joined
[2025-01-11 11:42:08] info: 	z2m: Starting interview of '0xa4c138a67e247cf0'
[2025-01-11 11:42:08] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0xa4c138a67e247cf0","ieee_address":"0xa4c138a67e247cf0"},"type":"device_joined"}'
[2025-01-11 11:42:08] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0xa4c138a67e247cf0","ieee_address":"0xa4c138a67e247cf0","status":"started"},"type":"device_interview"}'
[2025-01-11 11:42:08] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0xa4c138a67e247cf0","ieee_address":"0xa4c138a67e247cf0"},"type":"device_announce"}'
[2025-01-11 11:42:08] info: 	zh:controller:device: Device '0xa4c138a67e247cf0' is only compliant to revision '21' of the ZigBee specification (current revision: 23).
[2025-01-11 11:42:08] info: 	zh:controller: Succesfully interviewed '0xa4c138a67e247cf0'
[2025-01-11 11:42:08] info: 	z2m: Successfully interviewed '0xa4c138a67e247cf0', device has successfully been paired
[2025-01-11 11:42:08] warning: 	z2m: Device '0xa4c138a67e247cf0' with Zigbee model 'TS0601' and manufacturer name '_TZE204_unsxl4ir' is NOT supported, please follow https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html
[2025-01-11 11:42:09] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0xa4c138a67e247cf0","ieee_address":"0xa4c138a67e247cf0","status":"successful","supported":false},"type":"device_interview"}'

### External definition

```shell
const definition = {
    zigbeeModel: ['TS0601'],
    model: 'TS0601',
    vendor: '_TZE204_unsxl4ir',
    description: 'Automatically generated definition',
    extend: [],
    meta: {},
};

module.exports = definition;
```

### What does/doesn't work with the external definition?

It doesn't show the device on Z2M dashboard

felipemfz, 2025-01-18:
It's the same device as _TZE204_aagrxlbd, which is also the same as TB26-4. 

`const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const modernExtend = require('zigbee-herdsman-converters/lib/modernExtend');
const e = exposes.presets;
const ea = exposes.access;
const tuya = require('zigbee-herdsman-converters/lib/tuya');

const definition = {
    zigbeeModel: ['TS0601'],
    model: 'TS0601',
    vendor: '_TZE204_unsxl4ir',
    description: 'Light Switch',
    fromZigbee: [tuya.fz.datapoints],
    toZigbee: [tuya.tz.datapoints],
    //onEvent: tuya.onEventSetTime, // Add this if you are getting no converter for 'commandMcuSyncTime'
    configure: tuya.configureMagicPacket,
    exposes: [
        // Here you should put all functionality that your device exposes
        tuya.exposes.switch().withEndpoint('l1'),
        tuya.exposes.switch().withEndpoint('l2'),
        tuya.exposes.switch().withEndpoint('l3'),
        tuya.exposes.switch().withEndpoint('l4'),
    ],
    endpoint: (device) => {
            return {'l1': 1, 'l2': 1, 'l3': 1, 'l4': 1};
        },
    meta: {
        // All datapoints go in here
        multiEndpoint: true,
        tuyaDatapoints: [
            [1, 'state_l1', tuya.valueConverter.onOff],
            [2, 'state_l2', tuya.valueConverter.onOff],
            [3, 'state_l3', tuya.valueConverter.onOff],
            [4, 'state_l4', tuya.valueConverter.onOff],
        ],
    },
};

module.exports = definition;`

The same code works, just different model. 

I pasted it as a external converter, and it's now working normally.

felipemfz, 2025-01-18:
It's the same device as _TZE204_aagrxlbd, which is also the same as TB26-4.

```js
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const modernExtend = require('zigbee-herdsman-converters/lib/modernExtend');
const e = exposes.presets;
const ea = exposes.access;
const tuya = require('zigbee-herdsman-converters/lib/tuya');

const definition = {
zigbeeModel: ['TS0601'],
model: 'TS0601',
vendor: '_TZE204_unsxl4ir',
description: 'Light Switch',
fromZigbee: [tuya.fz.datapoints],
toZigbee: [tuya.tz.datapoints],
//onEvent: tuya.onEventSetTime, // Add this if you are getting no converter for 'commandMcuSyncTime'
configure: tuya.configureMagicPacket,
exposes: [
// Here you should put all functionality that your device exposes
tuya.exposes.switch().withEndpoint('l1'),
tuya.exposes.switch().withEndpoint('l2'),
tuya.exposes.switch().withEndpoint('l3'),
tuya.exposes.switch().withEndpoint('l4'),
],
endpoint: (device) => {
return {'l1': 1, 'l2': 1, 'l3': 1, 'l4': 1};
},
meta: {
// All datapoints go in here
multiEndpoint: true,
tuyaDatapoints: [
[1, 'state_l1', tuya.valueConverter.onOff],
[2, 'state_l2', tuya.valueConverter.onOff],
[3, 'state_l3', tuya.valueConverter.onOff],
[4, 'state_l4', tuya.valueConverter.onOff],
],
},
};

module.exports = definition;
```

The same code works, just different model.

I pasted it as a external converter, and it's now working normally.

gersonfer, 2025-01-18:
> It's the same device as _TZE204_aagrxlbd, which is also the same as TB26-4.
> 
> The same code works, just different vendor.
> 
> This is what I used.
> 
> const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
> const tz = require('zigbee-herdsman-converters/converters/toZigbee');
> const exposes = require('zigbee-herdsman-converters/lib/exposes');
> const reporting = require('zigbee-herdsman-converters/lib/reporting');
> const modernExtend = require('zigbee-herdsman-converters/lib/modernExtend');
> const e = exposes.presets;
> const ea = exposes.access;
> const tuya = require('zigbee-herdsman-converters/lib/tuya');
> 
> const definition = {
> zigbeeModel: ['TS0601'],
> model: 'TS0601',
> vendor: '_TZE204_unsxl4ir',
> description: 'Light Switch',
> fromZigbee: [tuya.fz.datapoints],
> toZigbee: [tuya.tz.datapoints],
> //onEvent: tuya.onEventSetTime, // Add this if you are getting no converter for 'commandMcuSyncTime'
> configure: tuya.configureMagicPacket,
> exposes: [
> // Here you should put all functionality that your device exposes
> tuya.exposes.switch().withEndpoint('l1'),
> tuya.exposes.switch().withEndpoint('l2'),
> tuya.exposes.switch().withEndpoint('l3'),
> tuya.exposes.switch().withEndpoint('l4'),
> ],
> endpoint: (device) => {
> return {'l1': 1, 'l2': 1, 'l3': 1, 'l4': 1};
> },
> meta: {
> // All datapoints go in here
> multiEndpoint: true,
> tuyaDatapoints: [
> [1, 'state_l1', tuya.valueConverter.onOff],
> [2, 'state_l2', tuya.valueConverter.onOff],
> [3, 'state_l3', tuya.valueConverter.onOff],
> [4, 'state_l4', tuya.valueConverter.onOff],
> ],
> },
> };
> 
> module.exports = definition;
> I pasted it as a external converter, and it's now working normally.

gersonfer, 2025-01-18:
I will use it.
Thanks in advance

closed 2025-01-18 by gersonfer