Koenkk/zigbee2mqtt #25803: zigbee2mqtt does not start: Could not resolve host: supervisor

tardich, 2025-01-14:
### What happened?

Deploying zigbee2mqtt outside of home-assistant from your helm chart, I was not able to make zigbee2mqtt start. Failing with:

> [02:44:28] INFO: Preparing to start...
> curl: (6) Could not resolve host: supervisor
> [02:44:29] ERROR: Something went wrong contacting the API
> [02:44:29] FATAL:
> [02:44:29] FATAL: A required add-on configuration option is missing!
> [02:44:29] FATAL:
> [02:44:29] FATAL: Please set a value for the 'data_path' option.
> [02:44:29] FATAL:
> [02:44:29] FATAL: If unsure, check the add-on manual for more information.
> [02:44:29] FATAL:

### What did you expect to happen?

...  it starts

### How to reproduce it (minimal and precise)

I have a rather minimal configuration.yaml (mostly the same used with version 1.x, but with the required additions for upgrade)

### Zigbee2MQTT version

2.0.0-2

### Adapter firmware version

20220219 (irrelevant at this point)

### Adapter

tubezb-cc2652p2-ethusb-2022 (irrelevant as well I guess)

### Setup

Running in plain docker

alejandro5x, 2025-01-14:
Hi, try this https://www.youtube.com/watch?v=PJth_a85wUs
hope it helps

tardich, 2025-01-14:
Unfortunately not helping, as the issue is with the docker image that just doesn't start at all as, as written above, it complains about not being able to resolve "supervisor".  I don't even understand what it's trying to do. Unable to find any information yet.

tardich, 2025-01-14:
Changed the docker image I'm using. Used to use zigbee2mqtt/zigbee2mqtt, which is causing issues for me in favor of kkoenkk/zigbee2mqtt:2.0.0, and it's no longer talking about the supervisor issue.  Now just facing an issue with the filesystem for configuration.yaml, which I normally mount from a configMap. But that's an easy one to look at.  :-)

closed 2025-01-14 by tardich