Koenkk/zigbee2mqtt #25805: [New device support]: Philips MasterConnect LEDtube EM T8 V3

raphpa, 2025-01-14:
### Link

https://www.lighting.philips.com/prof/led-lamps-and-tubes/led-tubes/masterconnect-ledtube-em-mains-t8/LP_CF_7904008_EU/family

### Database entry

{"id":23,"type":"Router","ieeeAddr":"0x00178801019afe0a","nwkAddr":48209,"manufId":4098,"manufName":"Philips","powerSource":"Mains (single phase)","modelId":"LEDtube_EMA_T8_2FT_V3","epList":[64,65,66,80,144,242],"endpoints":{"64":{"profId":260,"epId":64,"devId":257,"inClusterList":[0,3,4,5,6,8,21,769,1794,64771,64772,64773,64774,64775],"outClusterList":[25,1024,1030,1793,64771,64772],"clusters":{"genOnOff":{"attributes":{"onOff":1,"startUpOnOff":1,"offWaitTime":0}},"genBasic":{"attributes":{"appVersion":0,"stackVersion":0,"zclVersion":3,"modelId":"LEDtube_EMA_T8_2FT_V3","swBuildId":"6.6.16","dateCode":"20210622","hwVersion":16}},"genLevelCtrl":{"attributes":{"currentLevel":254,"onLevel":254}},"seMetering":{"attributes":{"meteringDeviceType":0,"currentSummDelivered":23,"status":0,"unitOfMeasure":0,"divisor":1000,"multiplier":1}},"lightingBallastCfg":{"attributes":{"lampBurnHours":3,"ballastStatus":0,"maxLevel":254,"minLevel":170,"powerOnLevel":254,"physicalMinLevel":170,"physicalMaxLevel":254}}},"binds":[],"configuredReportings":[],"meta":{}},"65":{"profId":260,"epId":65,"devId":262,"inClusterList":[0,3,4,1024],"outClusterList":[],"clusters":{"genBasic":{"attributes":{"appVersion":0,"stackVersion":0}},"msIlluminanceMeasurement":{"attributes":{"maxMeasuredValue":65534,"measuredValue":0,"lightSensorType":255}}},"binds":[],"configuredReportings":[],"meta":{}},"66":{"profId":260,"epId":66,"devId":263,"inClusterList":[0,3,4,1030],"outClusterList":[],"clusters":{"genBasic":{"attributes":{"appVersion":0,"stackVersion":0}},"msOccupancySensing":{"attributes":{"occupancy":0,"occupancySensorType":0}}},"binds":[],"configuredReportings":[],"meta":{}},"80":{"profId":260,"epId":80,"devId":7,"inClusterList":[0,3,4,10,64769],"outClusterList":[],"clusters":{"genBasic":{"attributes":{"appVersion":0,"stackVersion":0}},"genTime":{"attributes":{"timeStatus":0,"time":4294967295}},"genIdentify":{"attributes":{"identifyTime":0}}},"binds":[],"configuredReportings":[],"meta":{}},"144":{"profId":260,"epId":144,"devId":8,"inClusterList":[3,25,33,1793],"outClusterList":[0,3,4,5,6,8,10,21,33,768,769,1024,1030,1794,64769,64771,64772,64773,64774],"clusters":{"genOta":{"attributes":{}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":102,"inClusterList":[4,33],"outClusterList":[4,33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":0,"stackVersion":0,"hwVersion":16,"dateCode":"20210622","swBuildId":"6.6.16","zclVersion":3,"interviewCompleted":true,"meta":{"configured":332242049},"lastSeen":1736829788564}

### Zigbee2MQTT version

2.0.0

### Comments

Hello.

I created an external definition for this device which seems to work, but I am not sure how to proceed from there. 

For example how is the documentation generated? 
This device uses quite a different reset procedure than how it is described for the [MasterConnect LEDtube V2](https://www.zigbee2mqtt.io/devices/LP_CF_7904008_EU.html) (if that is even correct). See [https://www.docs.lighting.philips.com/en_gb/oem/download/connected-lighting-solutions/user-manual-easyair-sns210-mc.pdf](https://www.docs.lighting.philips.com/en_gb/oem/download/connected-lighting-solutions/user-manual-easyair-sns210-mc.pdf) in the maintenance section of the document.

### External definition

```shell
const { philipsLight } = require("zigbee-herdsman-converters/lib/philips");
const { electricityMeter, numeric } = require('zigbee-herdsman-converters/lib/modernExtend');

const definition = {
    zigbeeModel: ['LEDtube_EMA_T8_2FT_V3'],
    model: 'LEDtube T8 2ft V3.0',
    vendor: 'Philips',
    description: 'MasterConnect LEDtube EM/mains T8',
    extend: [
                philipsLight(), 
                electricityMeter({current: false, voltage: false, power: false}), 
                numeric({"name":"Lamp burn hours", "cluster":"lightingBallastCfg", "attribute":"lampBurnHours", 
                    "reporting":{"attribute":"lampBurnHours", "min":"MIN", "max":"MAX","change":1}, 
                    "description":"Lamp burn hours", "access":"STATE_GET", "endpointName":"64", "unit": "h"
                })
            ]
};

module.exports = definition;
```

### What does/doesn't work with the external definition?

-

Koenkk, 2025-01-14:
> For example how is the documentation generated?

It is automatically generated when the device is included in a release.

For the device; 
Could you make a PR by clicking [here](https://github.com/Koenkk/zigbee-herdsman-converters/edit/master/src/devices/philips.ts)?

raphpa, 2025-01-19:
I don't get the CI to build it without errors, so I'll leave this here for someone who is more familiar with the expected code structure.

Koenkk, 2025-01-19:
No problem, just make a PR and I can help fixing it.

closed 2025-03-28 by github-actions[bot]