Koenkk/zigbee2mqtt #25809: energy meter PJ-1203A vendor  _TZE284_81yrt3lo is pairing as not supported device

karen-gith, 2025-01-14:
### What happened?

Hello everyone
I bought the following module [https://www.zigbee2mqtt.io/devices/PJ-1203A.html](url) because it was in the list of equipment supported by z2m.
I manage to pair the module in z2m but it is paired as unsupported device TS0601.
I tried to follow the howto [Support new devices](https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html#support-new-devices)
Arriving in 2.1 I followed the link to “similar device” and there I think I found my module in the tuya.js def with the same vendor “_TZE284_81yrt3lo” on line 11347
https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/src/devices/tuya.ts
So I don't understand why my module isn't recognized, any ideas?

### Zigbee2MQTT version

1.42

### Adapter firmware version

7.3.1.0 build 176

### Adapter

EZSP v12

### Setup

jeedom on PI3B+

### Debug log

info 2025-01-14 11:27:00z2m: Successfully interviewed '0xa4c138246020d7fd', device has successfully been paired
warning 2025-01-14 11:27:00z2m: Device '0xa4c138246020d7fd' with Zigbee model 'TS0601' and manufacturer name '_TZE284_81yrt3lo' is NOT supported, please follow https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html

Koenkk, 2025-01-14:
`_TZE284_81yrt3lo` will be supported in the next z2m release (1 feb) and is supported in the  [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) already.

closed 2025-01-14 by Koenkk

sasojerman, 2025-01-14:
https://github.com/zigpy/zha-device-handlers/issues/3658

ThatMishakov, 2025-09-05:
Hello, I'm having an issue with this device on the latest version of Z2M, where after some time it starts to fill the logs with messages like this every 10 seconds:
```
[2025-09-05 07:15:51] error:    zhc:tuya: Failed respond to version response '0xa4c138c2f89ca0a0' (Error: ZCL command 0xa4c138c2f89ca0a0/1 manuSpecificTuya.mcuVersionRequest({"seq":2}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"reservedBits":0,"writeUndiv":false}) failed ({"target":30656,"apsFrame":{"profileId":260,"clusterId":61184,"sourceEndpoint":1,"destinationEndpoint":1,"options":4416,"groupId":0,"sequence":251},"zclSequence":27,"commandIdentifier":11} timed out after 10000ms))
```