Koenkk/zigbee2mqtt #25810: Incorrect current value for _TZ3000_kqvb5akv

gio82c, 2025-01-14:
### What happened?

When adding the device I get an error, even though the device is correctly configured the current value reported is not correct, it occurs after updating to version 2.0.0, the devices configured before are working correctly.
[image]
[image]

### Zigbee2MQTT version

2.0.0

### Adapter firmware version

20221226

### Adapter

SLZB-06

### Setup

docker in raspian OS

gio82c, 2025-01-14:
Also Energy value is not correct.

rouloy, 2025-01-15:
Same goes for the _TZ3000_cicwjqth, I changed the "current calibration" in the "settings (specific)" and it works for some time, but not consistently

rouloy, 2025-03-18:
Still present