Koenkk/zigbee2mqtt #25813: Motion sensor works in Z2M, but not shown in HA/MQTT

gyenesadam, 2025-01-14:
### What happened?

Randomly one of my Aqara moition sensors ([RTCGQ11LM]) dissappeared from HA/MQTT.
After reconnect in Z2M, the sensor still not exist in HA/MQTT.
Z2M works perfectly, post to te MQTT topic the sensor datas.

What can I do in this case?
All other devices works perfectly (include other same type sensors).

### Zigbee2MQTT version

 2.0.0

### Adapter firmware version

7.3.1.0 build 0

### Adapter

Sonoff Zb-Dongle E

### Setup

Add-on, Ha, x86-x64 config

gyenesadam, 2025-01-14:
`info 2025-01-14 13:05:04z2m:mqtt:
MQTT publish: topic 'zigbee2mqtt/Bxjsdjj', payload '{"battery":100,"device_temperature":26,"illuminance":42,"linkquality":255,"occupancy":false,"power_outage_count":182,"voltage":3015}'`

lux73, 2025-01-15:
if the Sensor works without Problem in z2m - don't you think your Question is related to HA?

Their Forums would be far better to get help 😉

closed 2025-01-15 by gyenesadam