Koenkk/zigbee2mqtt #25814: Aqara Z1 Pro ZNQBKG43LM (double rocker) keeps getting interviewed as Z1 Quad rocker ZNQBKG41LM

epicurean70, 2025-01-14:
### What happened?

every interview with the Z1 Pro double switch, z2m sees it as a Z1 Quad switch

### What did you expect to happen?

expected to interview correctly as ZNQBKG43LM

I also have the ZNQBKG44LM and the ZNQBKG45LM , both of which interviewed correctly

### How to reproduce it (minimal and precise)

force remove the device, repair, and same results . repeatedly

### Zigbee2MQTT version

2.0 Dev

### Adapter firmware version

20221226

### Adapter

slzb-06

### Setup

z2m docker install in ubuntu 22

### Debug log

 2025-01-14 20:44:33zh:controller: Interview for '0x54ef441000baf275' started
info 2025-01-14 20:44:33z2m: Device '0x54ef441000baf275' joined
info 2025-01-14 20:44:33z2m: Starting interview of '0x54ef441000baf275'
info 2025-01-14 20:44:33z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x54ef441000baf275","ieee_address":"0x54ef441000baf275"},"type":"device_joined"}'
info 2025-01-14 20:44:33z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x54ef441000baf275","ieee_address":"0x54ef441000baf275","status":"started"},"type":"device_interview"}'
info 2025-01-14 20:44:33z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x54ef441000baf275","ieee_address":"0x54ef441000baf275"},"type":"device_announce"}'
info 2025-01-14 20:44:33z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"2ndfloorlanding_Dimmer_HZC","ieee_address":"0xdc8e95fffe2562e6"},"type":"device_announce"}'
info 2025-01-14 20:44:34z2m: Accepting joining not in blocklist device '0xdc8e95fffe2562e6'
info 2025-01-14 20:44:35z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/2ndfloorlanding_Dimmer_HZC', payload '{"brightness":124,"effect":null,"linkquality":72,"power_on_behavior":null,"state":"OFF"}'
info 2025-01-14 20:44:49z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/DiningR_wallrouter_cc2530', payload '{"description":"33797/0x00158D000464A625","led":false,"linkquality":25,"rssi":48,"type":"CHILD_RFD"}'
info 2025-01-14 20:44:49z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/DiningR_wallrouter_cc2530', payload '{"description":"46497/0x00124B001F2B2630","led":false,"linkquality":32,"rssi":87,"type":"PARENT"}'
info 2025-01-14 20:44:49z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/DiningR_wallrouter_cc2530', payload '{"description":"13117/0x00158D00042D8384","led":false,"linkquality":29,"rssi":123,"type":"CHILD_RFD"}'
info 2025-01-14 20:44:49z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/DiningR_wallrouter_cc2530', payload '{"description":"5829/0xDC8E95FFFE2562E6","led":false,"linkquality":29,"rssi":0,"type":"CHILD_FFD_RXI"}'
info 2025-01-14 20:44:49z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/DiningR_wallrouter_cc2530', payload '{"description":"33797/0x00158D000464A625","led":false,"linkquality":32,"rssi":48,"type":"CHILD_RFD"}'
info 2025-01-14 20:44:52zh:controller: Succesfully interviewed '0x54ef441000baf275'
info 2025-01-14 20:44:52z2m: Successfully interviewed '0x54ef441000baf275', device has successfully been paired
info 2025-01-14 20:44:52z2m: Device '0x54ef441000baf275' is supported, identified as: Aqara Smart wall switch Z1 (quadruple rocker) (ZNQBKG41LM)
info 2025-01-14 20:44:52z2m: Configuring '0x54ef441000baf275'
info 2025-01-14 20:44:52z2m:mqtt: MQTT publish: topic 'homeassistant/switch/0x54ef441000baf275/switch_top/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state","value_template":"{{ value_json.state }}"},{"topic":"zigbee2mqtt/0x54ef441000baf275/availability","value_template":"{{ value_json.state }}"}],"availability_mode":"all","command_topic":"zigbee2mqtt/0x54ef441000baf275/top/set","device":{"hw_version":1,"identifiers":["zigbee2mqtt_0x54ef441000baf275"],"manufacturer":"Aqara","model":"Smart wall switch Z1 (quadruple rocker)","model_id":"ZNQBKG41LM","name":"0x54ef441000baf275","via_device":"zigbee2mqtt_bridge_0x00124b00258dac1b"},"name":"Top","object_id":"0x54ef441000baf275_top","origin":{"name":"Zigbee2MQTT","sw":"2.0.0-dev","url":"https://www.zigbee2mqtt.io"},"payload_off":"OFF","payload_on":"ON","state_topic":"zigbee2mqtt/0x54ef441000baf275","unique_id":"0x54ef441000baf275_switch_top_zigbee2mqtt","value_template":"{{ value_json.state_top }}"}'

epicurean70, 2025-01-20:
Anything I can do to amend/correct to recognise the switch correctly?

epicurean70, 2025-01-27:
I got another brand new ZNQBKG43LM, and pairing it with zigbee2mqtt went correctly ( as in it interviewed correctly). I can only assume the previous one is somehow warped . Is there a way to "factory reset" the switch, before trying the pairing process again?

epicurean70, 2025-02-19:
Maybe I am approaching this the wrong way. 
Is there a way to completely reset the swtich, so that it has no memory of its previous pair settings?

ShadowFistOkami, 2025-03-16:
> Is there a way to completely reset the swtich

In Z2M dashboard, click the Delete (bin) icon, then make sure you select the "force remove" option in the popup before clicking Delete.

closed 2025-05-23 by github-actions[bot]