Koenkk/zigbee2mqtt #25815: [Wrong device]: LIVARNO home LED-plafondlamp - Zigbee Smart Home is recognized as TS0502A Tuya LED driver

MathijsG, 2025-01-14:
### Link

https://www.lidl.nl/p/livarno-home-led-plafondlamp-zigbee-smart-home/p100339622

### Model

dontknowalreadyinstalled

### Description

Livarno lamp with warm white and brightness toggle

### Vendor

LIVARNO / Lidl

### Picture (link)

https://www.lidl.nl/assets/962cec90625450dd7bf3a6650e2f100a.jpeg

### Database entry

{}

### Notes

Somehow I couldn't find a file called "database.db" in data nor in the place of Zigbee2mqtt.

My (apparantly) Tuya based Lidl wall light (https://www.lidl.nl/p/livarno-home-led-plafondlamp-zigbee-smart-home/p100339622)

Reports itself as:

[image]

MathijsG, 2025-03-16:
Still the case.

MathijsG, 2025-05-17:
Still active

MathijsG, 2025-09-17:
Still active

MathijsG, 2025-11-17:
Still active

Ih0rd, 2026-01-17:
@MathijsG Your chandelier has a similar LED driver installed. You can create a separate device page with a corresponding image via whitelabel, but the new functionality will not appear just like that if you would like to have it.

closed 2026-03-26 by github-actions[bot]