Koenkk/zigbee2mqtt #25823: Touchlink doesnt work for Silicon Labs controllers (EFR32MG21 like Zigbee-E dongle)

awawa-dev, 2025-01-14:
### What happened?

As we already talked on Discord @Nerivec, I'm reporting a problem: touchlink feature does not find any device when the controller is a ZBDongle-E or other device based on Silicon Labs controllers (tested on 3 different units). For comparison, I am attaching also the log from cc2651p (zstack) which finds the lamp and the positioning is almost identical as in the case of the ZBDongle-E.

### What did you expect to happen?

Without Touchlink feature we are unable to reset Philips Hue lamps and join them to Z2M network. The lamp reset feature by turning it on and off seems to work only for bulbs

### How to reproduce it (minimal and precise)

Open Z2M frontend and start touchlink-ing discovery.

### Zigbee2MQTT version

2.0.0

### Adapter firmware version

8.0.2

### Adapter

ZBDongle-E

### Setup

RPi4

### Debug log

Not working touchlink ZBDongle-E 8.0.2 logs: https://pastebin.com/Xg0LVXkQ
Working touchlink cc2651p logs: https://pastebin.com/TkWfeSuL

Nerivec, 2025-01-15:
Could you get debug logs with a 7.4.x firmware? Silabs changed the handler in 8.0.x, just want to confirm what you're seeing with previous version.

awawa-dev, 2025-01-15:
Sure, here they are: https://pastebin.com/8TxQ2jQk

awawa-dev, 2025-01-15:
Hmm...`zh:ember:ezsp: Received network/route error ROUTE_ERROR_ADDRESS_CONFLICT for "0".` Something is trying to talk?

Nerivec, 2025-01-15:
Looks like something is reporting itself as a "coordinator" (network address 0) on channel 15 when the scan triggers, which conflicts with the coordinator that does the scan. Does that make any sense? At least it means something on 15 is being detected, which makes sense since zstack found a match on 15.
As far as I can tell, the payloads look fine though.
Any chance you can [sniff the process](https://www.zigbee2mqtt.io/advanced/zigbee/04_sniff_zigbee_traffic.html#with-emberznet-and-husbzb-1-adapters) with both zstack and ember so we can see the other side of the conversation?

awawa-dev, 2025-01-15:
Is sniffing for default Zigbee key (ZigbeeAlliance09): 5A:69:67:42:65:65:41:6C:6C:69:61:6E:63:65:30:39 will be enough during Touchlink-ing and can I skip my network key?

awawa-dev, 2025-01-15:
@Nerivec this time I get this error only once or twice just after resetting Philips Hue lamp.
But I managed to succesfully sniff transmission both for ember(not working) and zstack(working) and there are some differences
cc2651p https://pastebin.com/51XBHtTP
ZBDongle-E https://pastebin.com/KP65cfEn

awawa-dev, 2025-01-16:
New logs from ember: https://pastebin.com/WfcpRSFw and corresponding pcap is here: [sniff.zip](https://github.com/user-attachments/files/18432100/sniff.zip)

Nerivec, 2025-01-16:
Well, indeed, seems the device sends the response, but the firmware is not passing it over to Z2M, so it times out.
I find it strange that I have not seen reports of this before though, since 7.4.x has been around for almost a year now.
Any chance you have another device model (another brand?) to test it with? I'm wondering if this device is getting filtered out for some reason.

awawa-dev, 2025-01-16:
I tested it on 3 different silicone controllers, the effect is the same. Link: https://github.com/awawa-dev/HyperHDR/pull/1024#issuecomment-2583747779 it doesn't work also for our users. One Texas controllers and zstack have working touchlink.

awawa-dev, 2025-01-16:
Does https://github.com/Koenkk/zigbee-herdsman/pull/1300 should fix it? How can I switch z2m to this PR plugin and tested it?

awawa-dev, 2025-01-16:
Ezsp 7.3.3 seems something is happening on channel 15 but it timeouts out anyway: https://pastebin.com/nYnW5Khq

Nerivec, 2025-01-17:
> should fix it

No it's just general maintenance.

> Ezsp 7.3.3 seems something is happening

Nothing Touchlink related.

> I tested it on 3 different silicone controllers

I meant another Touchlink device model/brand, not another controller. Just trying to eliminate possibilities, since I can't see anything wrong with the raw frames sent (identical to TI), nor with the firmware (Touchlink is enabled by default).

@kirovilya I think you did some experimentation with Touchlink in the past, do you still have a device to test this with (7.4.x or 8.0.x firmware)?

kirovilya, 2025-01-17:
Hi. My testing was a simple case:
- a scan was done on all channels and the response of the devices was listened to in order to detect them.
- then, a reset was sent to the device so that it could pair with the network.
- sometimes can send an identification to indicate the found device.

I checked mainly on the light bulbs.

I'll check the operation of the touchlink, but later, maybe on the weekend.

kirovilya, 2025-01-19:
@Nerivec 
I looked through my old research... I found a mention that there were many efr32-firmwares in which the touchlink was not enabled... One of the working firmwares [7.1.1 was taken here](https://github.com/xsp1989/zigbeeFirmware/blob/edf2a0d3a717373764ec0673d4fecb0b882a636a/firmware/Zigbee3.0_Dongle-NoSigned/EZSP/ncp-uart-sw_7.1.1.0_115200.gbl) - I loaded it into the E stick - the touchlink works with the ezsp adapter (z2m 2.0.0) - found an Ikea light bulb on scan.
then I download a more modern [firmware from here 7.4.3](https://github.com/xsp1989/zigbeeFirmware/blob/master/firmware/Zigbee3.0_Dongle-NoSigned/EZSP/ncp-uart-sw_7.4.3.0_115200.gbl) - doesn't find the bulb. [firmware 8.0.2 ](https://github.com/Nerivec/silabs-firmware-builder/releases/download/v2024.6.2-update5/sonoff_zbdonglee_zigbee_ncp_8.0.2.0_115200_sw_flow.gbl) with ember adapter - doesn't find the bulb

awawa-dev, 2025-01-19:
I can confirm: touchlink works with the mentioned firmware version 7.1.1 and finds my lamp.

Nerivec, 2025-01-19:
@kirovilya is it the same problem of not seeing the response message come in at all in Z2M (but visible when sniffing) in all non-working cases?
Touchlink is enabled by default in firmware projects, so I don't think that's the problem (hasn't changed since 4.0 from what I can tell with the [old gecko_sdk](https://github.com/SiliconLabs/gecko_sdk/blob/gsdk_4.0/protocol/zigbee/app/ncp/sample-app/ncp-uart-hw/ncp-uart-hw.slcp#L36)).
We're sending raw MAC messages for InterPAN in Z2M (not using firmware-specific ZLL functions), so, my guess is, it's hitting a filter somewhere that may not have been configured the same in the past versions, and messages don't get outputted to the EZSP layer. All that stuff is closed source though 😢 I'll see if we can get something from Silabs...

@xsp1989 any chance you still have the old config you used for that 7.1.1 firmware that works (compared to the 7.4.3 that doesn't)?

awawa-dev, 2025-01-19:
@Nerivec logs from firmware 7.1.1 (touchlink is working) https://pastebin.com/kKz6BZD6

kirovilya, 2025-01-19:
> is it the same problem of not seeing the response message come in at all in Z2M (but visible when sniffing) in all non-working cases?

yes. sniff shows the correct answer, but stick does not give it.
[sniff.zip](https://github.com/user-attachments/files/18469792/sniff.zip)

xsp1989, 2025-01-20:
> [@kirovilya](https://github.com/kirovilya) is it the same problem of not seeing the response message come in at all in Z2M (but visible when sniffing) in all non-working cases? Touchlink is enabled [...]

I have only built 7.1.4 for ZB-GW04, not 7.1.1. You can download it from my firmware repository. You need to check the historical version. I deleted this firmware a few days ago, and I did not keep the previous build configuration.

awawa-dev, 2025-01-20:
Touchlink doesnt work for me in 7.1.4 :-(

awawa-dev, 2025-01-20:
False alarm, I had to reset the lamps and now touchlink detects the lamp 7.1.4

xsp1989, 2025-01-20:
> False alarm, I had to reset the lamps and now touchlink detects the lamp 7.1.4

I just checked the chat record with the Sonoff engineer, the 7.1 firmware does support touchlink, and I have successfully tested adding a device via touchlink, maybe it is the 7.1.4 I released. But I can't be 100% sure, maybe the Sonoff firmware can also do it.

awawa-dev, 2025-01-20:
On the https://github.com/itead/Sonoff_Zigbee_Dongle_Firmware/tree/master/Dongle-E/NCP_7.3.3 I see they do not support touchlink in the release note. Also 7.4.4 does not support it.

closed 2025-03-29 by github-actions[bot]

infabo, 2025-10-31:
Home Assistant Connect ZBT-1 is also efr32mg21 based. I am running on 7.4.4.3. Touchlink not working.