Koenkk/zigbee2mqtt #25824: [New device support]: Support request for TS0601 (_TZE200_s7rsrtbg): Tuya Zigbee Wall Light Switch Distance Detection human

Zazuza777, 2025-01-15:
### Link

https://aliexpress.ru/item/1005008159784117.html?ysclid=m5wv0b4qfx630259148&sku_id=12000044035103304

### Database entry

{   "id": 53,   "type": "Router",   "ieeeAddr": "0xa4c138aafe26d06a",   "nwkAddr": 51548,   "modelId": "TS0601",   "manufacturerName": "_TZE200_s7rsrtbg",   "powerSource": "Mains (single phase)",   "status": "online",   "lastSeen": 1673747000 }

### Zigbee2MQTT version

 2.0.0

### Comments

Hello Zigbee2MQTT team,

I am trying to integrate a Zigbee device with the following details:
- Model: TS0601
- Manufacturer: _TZE200_s7rsrtbg
- Description: Switch with two buttons and presence sensor.

[custom_ts0601_generated.zip](https://github.com/user-attachments/files/18417500/custom_ts0601_generated.zip)
[45df7312_zigbee2mqtt_2025-01-14T23-11-49.376Z.zip](https://github.com/user-attachments/files/18417502/45df7312_zigbee2mqtt_2025-01-14T23-11-49.376Z.zip)
[image]

The device connects to Zigbee2MQTT, but its functionality is not fully supported. I do not have much experience in setting up Zigbee devices, but with the help of ChatGPT, I have attempted the following steps to integrate the device.

### What we have tried:
1. Generated an external definition via the Zigbee2MQTT Dev Console.
2. Created and tested several custom converters based on the documentation.
3. Collected logs to analyze datapoints (`dp`) observed in the messages:
   - `dp: 1` seems to control the left switch.
   - `dp: 2` seems to control the right switch.
   - `dp: 109` seems related to the presence sensor.
   - Additional `dp` values appear but are not handled properly.

### Current issues:
- Some datapoints are not handled correctly.
- The presence sensor functionality is not working.
- Other unhandled datapoints are logged as warnings.

### Logs:
I have attached detailed logs and the custom converters we created during testing.

### Steps we have taken:
1. Added detailed debug logging to track all incoming messages.
2. Attempted to use `modernExtend` and `tuya.fz.datapoints` for handling the device.

### Expected behavior:
- Both switches should work independently.
- The presence sensor should report its state.

### Request:
Could you please help with adding full support for this device? If you need additional information or specific tests, I am happy to provide them.

Thank you for your assistance!

---

**Logs and details attached below.**

### External definition

```shell
const {deviceEndpoints, onOff, iasZoneAlarm} = require('zigbee-herdsman-converters/lib/modernExtend');

const definition = {
    zigbeeModel: ['TS0601'],
    model: 'TS0601',
    vendor: '_TZE200_s7rsrtbg',
    description: 'Automatically generated definition',
    extend: [deviceEndpoints({"endpoints":{"1":1,"2":2}}), onOff({"powerOnBehavior":false}), iasZoneAlarm({"zoneType":"generic","zoneAttributes":["alarm_1","alarm_2","tamper","battery_low"]})],
    meta: {"multiEndpoint":true},
};

module.exports = definition;
```

### What does/doesn't work with the external definition?

Unfortunately, nothing works. The device does not perform any functions with the generated external definition.
What doesn't work with the external definition:

Switches:

The two switches (dp: 1 and dp: 2) do not function.
The states of the switches (ON/OFF) are not displayed or controlled via Zigbee 2 MQTT.

Presence sensor:

The presence sensor (dp: 109) does not report any state.
Any tests with the presence sensor yield no results.

General integration:

The device is recognized as TS0601 by _TZE200_s7rsrtbg, but it is marked as unsupported in the Zigbee 2 MQTT logs.
No additional features of the device are accessible or functional.

Zazuza777, 2025-01-15:
Additional information:

This device successfully connects directly to Yandex Station 2 via Zigbee (without using the Tuya Smart Home integration). It is recognized as two light switches (two bulbs) and works as such, but no additional sensors or features (e.g., presence detection) are available.

Zazuza777, 2025-01-16:
Additional Information:

Hello Zigbee2MQTT Community,

I have conducted additional testing for the TS0601 (_TZE200_s7rsrtbg) device. Below are the details:

Actions Performed:
Connected the device to ZHA in Home Assistant.
Tested both switches (turned them on and off).
Triggered the presence sensor by moving in front of it.
Collected detailed logs during the testing.
Observations:
Switches:
Work via the On/Off (0x0006) cluster.
States are reported correctly.
Presence sensor:
Sends data through the IAS Zone (0x0500) cluster.
Reports zonestatus: 1 (motion detected) and zonestatus: 0 (no motion).
Tuya-specific cluster (0xEF00):
Sends dpValues but requires additional decoding for full functionality.
Attached Log File: The attached log contains all actions and interactions recorded during testing.

[home-assistant_zha_2025-01-15T23-58-17.415Z.log](https://github.com/user-attachments/files/18431657/home-assistant_zha_2025-01-15T23-58-17.415Z.log)

kwatt67, 2025-01-19:
I ordered one of the single gang versions to try it, not thinking to check for support, so it'd be nice to have these added.

lpmorin, 2025-01-22:
Also working on this one now. I'll report in a few days

closed 2025-03-31 by github-actions[bot]