Koenkk/zigbee2mqtt #25827: SLZB-06P7 connects but is not showing up in UI, 'Permit join (all)' button not available.

BHam86, 2025-01-15:
### What happened?

Set up a fresh system and coordinator. Logging reports everything working and connecting but the 'Permit join (All)' button is not doing anything (also no dropdown) and the settings/about page lists all coordinator fields as 'unknown' while that data is clearly visible in the Z2M log.

### What did you expect to happen?

I expected the coordinator to show up in the UI and being able to add devices :)

### How to reproduce it (minimal and precise)

Connected a new SLZB-06P7 via ethernet with static IP, powerd by usb. The SLZB UI works and I updated the firmware to the latest versions; Core v2.6.8.dev26, coordinator 20240716.

Running zigbee2mqtt and mosquitto containers in docker compose, both latest. Docker runs on an Alpine VM in Proxmox.

Z2M logging shows it connects to both the SLZB and MQTT and is running without any reported issues.

**Docker compose**
```
name: mqtt-z2m
services:
  mosquitto:
    container_name: mosquitto
    image: eclipse-mosquitto
    volumes:
      - /home/admin/mosquitto:/mosquitto
      - /home/admin/mosquitto/log:/mosquitto/log
      - /home/admin/mosquitto/mosquitto.conf:/mosquitto/config/mosquitto.conf
    ports:
      - 1883:1883
      - 9001:9001
  zigbee2mqtt:
    container_name: zigbee2mqtt
    image: koenkk/zigbee2mqtt
    restart: unless-stopped
    volumes:
        - /home/admin/zigbee2mqtt:/app/data
    ports:
        - 8080:8080
    environment:
        - TZ=Europe/Amsterdam
```

**SLZB config**
```
Radiomodule mode: Zigbee coordinator
Serial speed: 115200
Socket port: 6638
```

**Zigbee2MQTT config**
```
version: 4
homeassistant:
  enabled: true
frontend:
  enabled: true
mqtt:
  base_topic: zigbee2mqtt
  server: mqtt://192.168.50.162
  user: ********
  password: ********
advanced:
  transmit_power: 20
serial:
  port: tcp://192.168.50.204:6638
  baudrate: 115200
  adapter: zstack
  disable_led: false
```

Tried it with and without the additional transmit_power and disable_led settings without change.

**MQTT config**
```
allow_anonymous false
password_file /mosquitto/passwd
listener 1883
```

### Zigbee2MQTT version

2.0.0 (koenkk/zigbee2mqtt)

### Adapter firmware version

20240716

### Adapter

SLZB-06P7

### Setup

Docker container

### Debug log

**Zigbee2MQTT log**
[2025-01-14 23:09:00] info: 	z2m: Logging to console, file (filename: log.log)
[2025-01-14 23:09:00] info: 	z2m: Starting Zigbee2MQTT version 2.0.0 (commit #060ae99cff715cc9ef6a0cfa7cae0cdcb17f0c8a)
[2025-01-14 23:09:00] info: 	z2m: Starting zigbee-herdsman (3.2.1)
[2025-01-14 23:09:00] info: 	zh:zstack:znp: Opening TCP socket with 192.168.50.204:6638
[2025-01-14 23:09:00] info: 	zh:zstack:znp: Socket connected
[2025-01-14 23:09:00] info: 	zh:zstack:znp: Socket ready
[2025-01-14 23:09:00] info: 	zh:zstack:znp: Writing CC2530/CC2531 skip bootloader payload
[2025-01-14 23:09:01] info: 	zh:zstack:znp: Skip bootloader for CC2652/CC1352
[2025-01-14 23:09:38] info: 	zh:controller: Wrote coordinator backup to '/app/data/coordinator_backup.json'
[2025-01-14 23:09:38] info: 	z2m: zigbee-herdsman started (resumed)
[2025-01-14 23:09:38] info: 	z2m: Coordinator firmware version: '{"meta":{"maintrel":1,"majorrel":2,"minorrel":7,"product":1,"revision":20240716,"transportrev":2},"type":"zStack3x0"}'
[2025-01-14 23:09:38] info: 	z2m: Currently 0 devices are joined.
[2025-01-14 23:09:38] info: 	z2m: Connecting to MQTT server at mqtt://192.168.50.162
[2025-01-14 23:09:38] info: 	z2m: Connected to MQTT server
[2025-01-14 23:09:38] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/state', payload '{"state":"online"}'
[2025-01-14 23:09:38] info: 	z2m: Started frontend on port 8080
[2025-01-14 23:09:38] info: 	z2m: Zigbee2MQTT started!
[2025-01-14 23:09:42] info: 	z2m:mqtt: MQTT publish: topic 'homeassistant/...

**SLZB-06P7 log**
[23:09:02] taskZB | New client: 192.168.50.162 id: 0

Koenkk, 2025-01-15:
Could you see if this is fixed in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)?

BHam86, 2025-01-15:
> Could you see if this is fixed in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)?

Running latest-dev, no changes.

BHam86, 2025-01-16:
Out of curiosity I just installed z2m via the community-scripts LXC on Proxmox and that build does work. Their script also installs 2.0.0 and I've copied the config verbatim. Logging between the two variants shows no discernable differences.

Koenkk, 2025-01-17:
Z2M seems to start fine, are you running the frontend behind a reverse proxy or something? It seems that your browser cannot connect to the websocket.

BHam86, 2025-01-18:
I read 'reverse proxy' and smacked myself in the head. I immediately knew I didn't enable websocket support in NGINX... :|
Thanks for the hint in the right direction!

closed 2025-01-18 by BHam86