Koenkk/zigbee2mqtt #25841: YOKIS MTV300E-UP cannot set brightness

musapinar, 2025-01-15:
### What happened?

Can't set brightness via mqtt. Values set above 200 and below 60 seem ok. But anything inbetween saves wrong values.

### What did you expect to happen?

When i set brightness to 90 via mqtt, I expect "light.televariateur_salle_a_manger" to be 90, not 230.
When i set brightness to 110, i expect it to be 110, not 240.

### How to reproduce it (minimal and precise)

- Add device YOKIS MTV300E-UP
- Rename device to "televariateur_salle_a_manger"
- Verify you have entity "light.televariateur_salle_a_manger"
- Send this via mqtt to topic ```zigbee2mqtt/televariateur_salle_a_manger/set```

```{
  "brightness": 110
}```

- Send again changing brightness to 90
- Send again some brightness values around 100, say between 70 and 130, a few time

At some point, it will not save the desired value into "light.televariateur_salle_a_manger", but erratic values, like 220.
It is not a matter of spamming/dos'ing the bus. Take your time to send the commands. Anything around brightness 100 produces wrong values to be saved, like one time out of three.

### Zigbee2MQTT version

2.0.0

### Adapter firmware version

20240710

### Adapter

zStack3x0

### Setup

haos addon, running on oracle vm

### Debug log

none

musapinar, 2025-01-15:
Also the set values do "drift". When I set brightness to 150, it save 151. 
The key to reproduce the issue is to cross the brightness 100 threshold I think. Do 130, then 80, back and forth. etc.

LaurentChardin, 2025-01-18:
Dear @musapinar 

I will do some cleaning in the device declaration : the Yokis device does not fully support the LevelCtrl cluster and therefore some of the controls are not working, mostly because it provides its own dimming implementation on a custom cluster.

I will update the documentation but also try to figure out how to remove the controls from the generic cluster exposed by Z2M, and detail which ones should be used instead.

LaurentChardin, 2025-01-19:
@musapinar  
I tried to reproduce the issue without success at the moment : the values do not drift. but i will try maybe to script it more aggressively. 

But just out of curiosity : can you please check the firmware version of your MTV ?

musapinar, 2025-01-19:
Bonsoir @LaurentChardin 

Firmware build date : 20240415
Firmware version : 2.4.0

How to reproduce : https://youtu.be/orkp9MDK1AE

When I send "brightness: 50", if the previous value was above 100, it sets the entity at 128. If the previous value was below 100, it does set the brightness at 50. I have a feeling that the device somehow has to decide whether it should interpret the value I am sending it as a percentage (0-100) or as a byte (0-255). Where 50% corresponds to 128.
Could it be a lack of consistency (percentages versus bits) somewhere ? Eventhough I can see a mapping at line 1068. I'm not yet familiar with Zigbee and converters. I can only help manually testing for now.

- Sending "brightness: 50" sometimes sets the "light.televariateur" brightness at 50
- Sending "brightness: 50" sometimes sets the "light.televariateur" brightness at 128 (which is 50%)
- Sending "brightness: 75" sometimes sets the "light.televariateur" brightness at 75% (while I would expect about 30%)
- Sending "brightness: 50" sometimes sets the "light.televariateur" brightness at 50 but then it jumps to 128 (almost immediately but slow enough that I can see it rapidly change in developer-tools/state), or after a few seconds (about 1s to 3s).

Value is not drifting or "sliding" per se, there is an offset by one to be exact indeed.
The reason it drifts is more relevant considering my use case : I "paired" the Yokis MTV device with a rotary knob (Aqara H1) to control the brightness. Since I both read and set the value of "light.televariateur", each knob's step introduces an increasing offset (that I could solve via software I guess).

Thank you for your time investigating this issue.

LaurentChardin, 2025-01-19:
@musapinar my firmware for the MTV is `3.3.3`, i highly recommend to update it as many things were fixed in the previous releases.

Yokis does not support OTA through Z2M yet but this is in discussion.

LaurentChardin, 2025-01-19:
@musapinar 

Thank you for the video, very informative of your test setup. I have few remarks to try to understand what is going here, because there are quite a few things happening in the backgroup between:
 device <-> zigbee network <-> z2m <-> z2m HA extension <-> mqtt <-> home assistant listener <-> home assistant UI

Reminder:
- Z2M communicates with HA through a dedicated topic, setting up specific messages than the regular ones, but they should the same values

so here my questions :
- Do you have access to z2m UI in order to be able to see the same values there (aka before HomeAssistant) ? i think it is possble from HA addon but i dont remember
- Z2M UI would also give you access to logs (warning those can be quite verboses in debug mode)
- Did you created a zigbee binding between the aqara H1 or are you going through HA to connect them via MQTT ? (suspecting additional values for brightness coming from something else in the background)

Best,

musapinar, 2025-01-19:
@LaurentChardin is there a documented procedure somewhere on where to get the latest bin and how to wire/flash the mcu ? I could not find anything on the internet except the vendor's "up2pro" app that seems to require a €150 device ("KEY-UP Bluetooth/radio interface").

- I do have access to the z2m ui. Here is a new video : https://youtu.be/B016JrekgPQ
- I do have access to debug logs. See video. It seems like the erratic values are due to the command being sent twice back to back ?
`
info 2025-01-19 13:21:21z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/televariateur_salle_a_manger', payload '{"blink_amount":null,"blink_off_time":null,"blink_on_time":null,"brightness":90,"contact_mode":null,"current_position":null,"deaf_blink_amount":null,"deaf_blink_time":null,"dimmer_Up_down_command":null,"dimmer_move_to_favorite":null,"dimmer_start_nightlight_command":null,"effect":null,"enable_R12M_long_press":"OFF","enable_blink":"OFF","enable_deaf":"OFF","enable_local_config_lock":"OFF","enable_long_press":"OFF","enable_nc_command":"OFF","enable_pre_off_delay":"OFF","enable_pre_on_delay":"OFF","enable_ramp_down":"OFF","enable_ramp_up":"OFF","enable_short_press":"OFF","enable_stop_announce":"OFF","favorite_position_1":null,"favorite_position_2":null,"favorite_position_3":null,"high_dim_limit":null,"identify":null,"input_mode":null,"last_bp_connect_state":"RELEASED","last_local_command_state":"RELEASED","linkquality":29,"long_on_command":null,"long_on_duration":null,"long_press_duration":null,"low_dim_limit":null,"memory_position":null,"memory_position_mode":"OFF","nightlight_ending_brightness":null,"nightlight_on_time":null,"nightlight_ramp_time":null,"nightlight_starting_brightness":null,"nightlight_starting_delay":null,"on_timer":null,"operating_mode":null,"power_failure_mode":null,"pre_off_delay":null,"pre_on_delay":null,"prev_state":"OFF","pulse_duration":null,"ramp_continuous_time":null,"ramp_down":null,"ramp_up":null,"relaunch_ble_advert":null,"reset_to_factory_settings":null,"select_input_mode":null,"send_press":null,"send_release":null,"state":"ON","state_after_blink":null,"step_continuous":null,"step_controller_mode":"OFF","step_down":null,"step_nightlight":null,"step_up":null,"stop_announce_time":null,"time_between_press":null,"time_type":null}'

info 2025-01-19 13:21:21z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/televariateur_salle_a_manger', payload '{"blink_amount":null,"blink_off_time":null,"blink_on_time":null,"brightness":229,"contact_mode":null,"current_position":null,"deaf_blink_amount":null,"deaf_blink_time":null,"dimmer_Up_down_command":null,"dimmer_move_to_favorite":null,"dimmer_start_nightlight_command":null,"effect":null,"enable_R12M_long_press":"OFF","enable_blink":"OFF","enable_deaf":"OFF","enable_local_config_lock":"OFF","enable_long_press":"OFF","enable_nc_command":"OFF","enable_pre_off_delay":"OFF","enable_pre_on_delay":"OFF","enable_ramp_down":"OFF","enable_ramp_up":"OFF","enable_short_press":"OFF","enable_stop_announce":"OFF","favorite_position_1":null,"favorite_position_2":null,"favorite_position_3":null,"high_dim_limit":null,"identify":null,"input_mode":null,"last_bp_connect_state":"RELEASED","last_local_command_state":"RELEASED","linkquality":32,"long_on_command":null,"long_on_duration":null,"long_press_duration":null,"low_dim_limit":null,"memory_position":null,"memory_position_mode":"OFF","nightlight_ending_brightness":null,"nightlight_on_time":null,"nightlight_ramp_time":null,"nightlight_starting_brightness":null,"nightlight_starting_delay":null,"on_timer":null,"operating_mode":null,"power_failure_mode":null,"pre_off_delay":null,"pre_on_delay":null,"prev_state":"OFF","pulse_duration":null,"ramp_continuous_time":null,"ramp_down":null,"ramp_up":null,"relaunch_ble_advert":null,"reset_to_factory_settings":null,"select_input_mode":null,"send_press":null,"send_release":null,"state":"ON","state_after_blink":null,"step_continuous":null,"step_controller_mode":"OFF","step_down":null,"step_nightlight":null,"step_up":null,"stop_announce_time":null,"time_between_press":null,"time_type":null}'
`

- Regarding the Aqara H1 : it is just an automation via mqtt. Please ignore its existance. Videos where recorded without interacting with the rotary knob, while the automation was disabled (in order to narrow down the investigation perimeter with less variables/middlewares). To me, the Aqara H1 has nothing to do with the raised issue.

LaurentChardin, 2025-01-19:
@musapinar 

For OTA : currently you don't need the KEY-UP devices. I actually update the firmware from the U2PRO IOS app and it updates the firmware through BLE. You need to create an account in order to access it i believe. I was in touch with Yokis team so i had access to it, but we are discussing OTA actually, but it is not in place yet. In the meantime, you can probably download the U2PRO mobile app and request an account. (the update process is quite easy when you connect to the device through BLE).

What i am trying to understand is what causes the different changes of value, if it actually comes from the device itself or is something else interfering (in z2m for example). 

Thanks for the video : however you are not really in debug mode as you dont have the Zigbee messages from the adapter in the logs. So we are missing the details of the zigbee network. It's either because the debug mode didn't really kicked in when you selected it, or because there are default filtering for the logs. Please check in `Settings > Advanced > Log debug namespace ignore`. What you currently see in the logs are only the MQTT messages.

If the bug is coming from the device, nothing we can do in z2m: we would need to report the issue to Yokis as it would be firmware related. Thus the need to update it first to see if the bug is still there.

If the bug is not coming from the device, meaning something else is interacting, and chaining a command that triggers the second report: this is what i am looking for.
Also i am puzzled about the relation between the value and the percent : would one of a system (z2m or ha) actually messing up confusing percentage and value ? Intriguing ... I dont think the issue is related to the device definition in z2m, and i might try to setup a HA as well to try to be in the same conditions than you. 

I was not able to reproduce the issue on my MTV. 
So i am more inclined to point to the firmware issue at this stage.

musapinar, 2025-01-19:
@LaurentChardin I can confirm this issue is now resolved after updating the firmware from 2.4.0 to 3.3.3. Thank you for your time and effort investigating it, Laurent.
I am closing this ticket.

FYI, the minor concern regarding the offset by 1 still exists but I do not think it is within your converters scope.
- When I set brightness greater than or equal to 127 via mqtt, entity "light.televariateur" in HAOS has attribute "brightness" value off set by +1
- When I set brightness lower than 127 via mqtt, entity "light.televariateur" in HAOS has correct attribute "brightness" value
- On the Z2M UI, the brightness value is always correct

Test cases : 
- Setting brightness to 210 via mqtt, light.televariateur attribute "brightness" shows 211. Z2M UI shows 210
- Setting brightness to 127 via mqtt, light.televariateur attribute "brightness" shows 128. Z2M UI shows 127
- Setting brightness to 126 via mqtt, light.televariateur attribute "brightness" shows 126. Z2M UI shows 126
- Setting brightness to 70 via mqtt, light.televariateur attribute "brightness" shows 70. Z2M UI shows 70

I will circumvent this with a workaround in my rotary knob's automation

closed 2025-01-19 by musapinar

LaurentChardin, 2025-01-19:
Glad to hear @musapinar 

Don't hesitate to share your experience with Yokis devices : i will continue to polish the device definition in Z2M and also probably will do more checks with HA in the future.

The +1 offset rings a lot some form of approximation being done (HA expect % maybe ?). I will also check if we are doing anything in the HA extension.