Koenkk/zigbee2mqtt #25842: [New device support]:

didierma34, 2025-01-15:
### Link

https://fr.aliexpress.com/item/1005005881727281.html?gatewayAdapt=glo2fra

### Database entry

{     "last_seen": "2025-01-15T21:37:00.655Z",     "linkquality": 92,     "device": {         "applicationVersion": 78,         "dateCode": "",         "friendlyName": "0xa4c138c05b1e123f",         "hardwareVersion": 1,         "ieeeAddr": "0xa4c138c05b1e123f",         "manufacturerID": 4417,         "manufacturerName": "_TZE284_81yrt3lo",         "networkAddress": 32925,         "powerSource": "Mains (single phase)",         "stackVersion": 0,         "type": "Router",         "zclVersion": 3     } }

### Zigbee2MQTT version

2.00.0

### Comments

After pairing, no commande. Non Supported

Nom familler : 0xa4c138c05b1e123f
Last seen : right now
Type de périphérique : Router
Modèle Zigbee : TS0601
Fabricant Zigbee : _TZE284_81yrt3lo
Statut du support : Non supporté

Adresse IEEE : 0xa4c138c05b1e123f
Adresse réseau : 0x809D / 32925

### External definition

```shell
const definition = {
    zigbeeModel: ['TS0601'],
    model: 'TS0601',
    vendor: '_TZE284_81yrt3lo',
    description: 'Automatically generated definition',
    extend: [],
    meta: {},
};

module.exports = definition;
```

### What does/doesn't work with the external definition?

Nothing work... cause no commande.

simonedimarzio, 2025-01-17:
Hello, please check https://github.com/Koenkk/zigbee2mqtt/issues/25809#issuecomment-2590949216

`_TZE284_81yrt3lo will be supported in the next z2m release (1 feb) and is supported in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) already.`

simonedimarzio, 2025-01-17:
> ### Link
> https://fr.aliexpress.com/item/1005005881727281.html?gatewayAdapt=glo2fra
> 
> ### Database entry
> { "last_seen": "2025-01-15T21:37:00.655Z", "linkquality": 92, "device": { "applicationVersion": 78, "dateCode": "", "friendlyName": "0xa4c138c05b1e123f", "hardwareVersion": 1, "ieeeAddr": "0xa4c138c05b1e123f", "manufacturerID": 4417, "manufacturerName": "_TZE284_81yrt3lo", "networkAddress": 32925, "powerSource": "Mains (single phase)", "stackVersion": 0, "type": "Router", "zclVersion": 3 } }
> 
> ### Zigbee2MQTT version
> 2.00.0
> 
> ### Comments
> After pairing, no commande. Non Supported
> 
> Nom familler : 0xa4c138c05b1e123f Last seen : right now Type de périphérique : Router Modèle Zigbee : TS0601 Fabricant Zigbee : _TZE284_81yrt3lo Statut du support : Non supporté
> 
> Adresse IEEE : 0xa4c138c05b1e123f Adresse réseau : 0x809D / 32925
> 
> ### External definition
> const definition = {
>     zigbeeModel: ['TS0601'],
>     model: 'TS0601',
>     vendor: '_TZE284_81yrt3lo',
>     description: 'Automatically generated definition',
>     extend: [],
>     meta: {},
> };
> 
> module.exports = definition;
> ### What does/doesn't work with the external definition?
> Nothing work... cause no commande.

simonedimarzio, 2025-01-17:
@didierma34, please check https://github.com/Koenkk/zigbee2mqtt/issues/25809#issuecomment-2590949216

`_TZE284_81yrt3lo will be supported in the next z2m release (1 feb) and is supported in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) already.`

Koenkk, 2025-01-17:
`_TZE284_81yrt3lo` will be supported in the 1 February release indeed!

closed 2025-01-17 by Koenkk

NedalLive, 2025-11-15:
Hello,
I've seen several comments saying it would be supported in the February version, but it's now November 15th and the device isn't recognized on my system.

Any ideas?

[image]