Koenkk/zigbee2mqtt #25843: Zigbee2mqtt 2.0 no external converters

NightFlexer, 2025-01-16:
### What happened?

In docker version Zigbee2mqtt 2.0 there are no external converters

### What did you expect to happen?

I expected that I would be able to add external converters

### Zigbee2MQTT version

oenkk/zigbee2mqtt:latest-dev (2.0)

### Adapter firmware version

6.10.3

### Adapter

EZSP and ember

### Setup

Rasppbary Pi 5 / docker-compose

chris-1243, 2025-01-16:
From #24198

> External converters and extensions

The external_converters setting is no longer used. Instead all external converters inside data/external_converters directory are now automatically loaded. Make sure to move all your external converters to this directory.

NightFlexer, 2025-01-16:
> From [#24198](https://github.com/Koenkk/zigbee2mqtt/discussions/24198)

Thanks for the tip

closed 2025-01-16 by NightFlexer