Koenkk/zigbee2mqtt #25846: [New device support]: eWeLink MG1-5RZ Human Presence Radar

87insane, 2025-01-16:
### Link

http://product.maifar.com/wl/?id=Qeu5zz09VMS3tBxE34g7DMUcHlt7B71o&path=Human%20Presence%20Radar%2FMG1-5RZ%20%20Human%20Presence%20Radar%20%28Zigbee%29%20Manual%20V1.0.pdf&mode=list

### Database entry

{"id":50,"type":"Router","ieeeAddr":"0x7cb94c78021d0000","nwkAddr":33998,"manufId":4742,"manufName":"eWeLink","powerSource":"Unknown","modelId":"CK-BL702-MWS-01(7016)","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":263,"inClusterList":[0,3,4,5,1030,4096,64512,64599],"outClusterList":[25],"clusters":{"genBasic":{"attributes":{"stackVersion":0,"hwVersion":0}},"msOccupancySensing":{"attributes":{"occupancy":1}}},"binds":[],"configuredReportings":[{"cluster":1030,"attrId":0,"minRepIntval":10,"maxRepIntval":60,"repChange":0}],"meta":{}}},"appVersion":1,"stackVersion":0,"hwVersion":0,"dateCode":"20210610-BL","swBuildId":"1.0.2","zclVersion":8,"interviewCompleted":true,"meta":{"configured":332242049},"lastSeen":1736953234205}

### Zigbee2MQTT version

2.0.0 commit: 060ae99

### Comments

Unfortunately I don't know how to set the exposes correctly. I've tried copying from other devices but can't get it to work.

### External definition

```shell
const {occupancy} = require('zigbee-herdsman-converters/lib/modernExtend');

const definition = {
    zigbeeModel: ['CK-BL702-MWS-01(7016)'],
    model: 'CK-BL702-MWS-01(7016)',
    vendor: 'eWeLink',
    description: 'Automatically generated definition',
    extend: [occupancy()],
    meta: {},
};

module.exports = definition;
```

### What does/doesn't work with the external definition?

Unfortunately, you can't set anything with the Automatically generated definition or you can get reasonable values.

87insane, 2025-02-13:
Nobody here for help?

kachalex, 2025-03-03:
const m = require('zigbee-herdsman-converters/lib/modernExtend');

const definition = {
    zigbeeModel: ['CK-BL702-MWS-01(7016)'],
    model: 'CK-BL702-MWS-01(7016)',
    vendor: 'eWeLink',
    description: 'Automatically generated definition',
    extend: [m.occupancy()],
    meta: {},
};

module.exports = definition;

i have same problem pls fix for us

moshiko2312, 2025-03-03:
hi mi to 
i have this product waiting for add to database

moshiko2312, 2025-05-03:
> Still require this please

Me too

cornflache, 2025-07-05:
Still required please