Koenkk/zigbee2mqtt #25849: Moes ZSS-S01-TH-C-MS incorrect match + no battery readings

epd5, 2025-01-16:
### What happened?

I have 2 Moes ZSS-S01-TH-C-MS Temperature & humidity sensors. They were both recognized by Zigbee2mqtt as 
Zigbee Model
    TS0201
Zigbee Manufacturer
    _TZ3000_f2bw0b6k
However they work as expected apart from never getting a battery reading. When I used ZHA it was able to read the battery %.

### What did you expect to happen?

Incorrectly recognized device + no battery reading

### How to reproduce it (minimal and precise)

Install Moes ZSS-S01-TH-C-MS device

### Zigbee2MQTT version

2.0.0 commit: 060ae99cff715cc9ef6a0cfa7cae0cdcb17f0c8a

### Adapter firmware version

20240710

### Adapter

zStack3x0. Sonoff Dongle P

### Setup

Docker container on Debian x86-64

Synochy, 2025-01-19:
Same problem with the IH-K009, #25086

seizmos-sudo, 2025-01-25:
I confirm, problem with battery report with IH-K009. I use latest version zigbee2mqtt.

liamstears, 2025-03-25:
I have the same issue, same device, identified the same I have multiple of these yet they all report voltage as 2600 but somehow z2m also gives them a percentage which differs slightly between 25% and 31% even though they all say 2600

`z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Living Room Thermometer', payload '{"battery":28,"humidity":59.01,"linkquality":232,"temperature":18.69,"update":{"installed_version":-1,"latest_version":-1,"state":null},"voltage":2600}'`

I did previously open a thread here about these being misidentified: https://github.com/Koenkk/zigbee2mqtt/issues/22802

Unfortunately no reply

@Koenkk please can you help us with these...

Koenkk, 2025-03-25:
Could you provide a screenshot of the device about page from the z2m frontend?

liamstears, 2025-03-25:
> Could you provide a screenshot of the device about page from the z2m frontend?

Of course, let me know if you need anything else 

[image]

[image]

Koenkk, 2025-03-26:
The reported 28% comes from the device, can you try to read it directly from the device like this:

[image]

Make sure to wakeup the device right before sending it, the reported value has to be divided by 2.

liamstears, 2025-03-26:
> The reported 28% comes from the device, can you try to read it directly from the device like this:
> 
> <img alt="Image" width="1125" src="https://private-user-images.githubusercontent.com/2892853/427226329-02fdac54-8564-47d4-a053-7c2226da01aa.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3NDMwMjQ1OTMsIm5iZiI6MTc0MzAyNDI5MywicGF0aCI6Ii8yODkyODUzLzQyNzIyNjMyOS0wMmZkYWM1NC04NTY0LTQ3ZDQtYTA1My03YzIyMjZkYTAxYWEucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI1MDMyNiUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNTAzMjZUMjEyNDUzWiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9ZjUwM2U3ZWI0Y2I2NzkzM2ViNDAwY2E2YjU2ZDc3ZjEyMzkxOWNmNTM0ZWRmNjk4NmJmMTJmMjY4ZTA2YzU3MiZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QifQ.dFW4BAiqVGbtb5Qu_hrVDK8CiqZEtELWGY87fltACH0">
> Make sure to wakeup the device right before sending it, the reported value has to be divided by 2.

[image]

Ok I've don't that and attached and to double check I put in a brand new battery afterwards and repeated the steps and it gave me 30/200 so maybe it is reading the correct battery and mine all weirdly having the same battery voltage level across all the sensors I have (6 or so) is just coincidence that they are all roughly the same and not an issue like the op is having. So I believe mine are reading correctly now. Still odd and doesn't explain why the op isn't getting a reading and I am though with the same sensors. Is there hardware differences maybe? Any way we can pull more info from the devices to test between mine and ops to maybe find ops issue?

Mine are still identified incorrectly in z2m though...

Koenkk, 2025-03-27:
There could be hardware diffs indeed.

> Mine are still identified incorrectly in z2m though...

How should it be detected? Could you provide the model number and description.

Koenkk, 2025-03-28:
Added it.

Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now and in the next release which is every 1st of the month.

Could you also submit a picture for the docs? This can be done by:
1. Create a fork by clicking [here](https://github.com/Koenkk/zigbee2mqtt.io/fork)
2. Go to the `public/images/devices` directory, *Add file* -> *Upload files*
3. Upload the files and press *Commit changes*
4. Press *Contribute* -> *Open pull request* -> update title/description -> *Create pull request*

Make sure that:
- The filename is `ZSS-S01-TH.png`
- The size is 512x512
- The background is transparent (use e.g. [Adobe remove background](https://new.express.adobe.com/tools/remove-background))

liamstears, 2025-03-28:
> Added it.
>
> Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now and in the next release which is every 1st of the [...]

Done

Thank you

closed 2025-06-05 by github-actions[bot]