Koenkk/zigbee2mqtt #25859: [New device support]: 10 Key CCT Zigbee Remote Control R2(WZS)

darknessblade, 2025-01-16:
### Link

https://aliexpress.com/item/1005005182031336.html

### Database entry

{"id":2,"type":"EndDevice","ieeeAddr":"0xe0798dfffecf814b","nwkAddr":48212,"manufId":4098,"manufName":"_TZ3000_xa9g7rxs","powerSource":"Battery","modelId":"TS1002","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":261,"inClusterList":[0,1,3,4,4096,57345],"outClusterList":[25,10,3,4,5,6,8,768,4096],"clusters":{"genBasic":{"attributes":{"modelId":"TS1002","manufacturerName":"_TZ3000_xa9g7rxs","powerSource":3,"zclVersion":3,"appVersion":67,"stackVersion":0,"hwVersion":1,"dateCode":""}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":67,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1737050789436}

### Zigbee2MQTT version

2.0.0

### Comments

Model/manufacturer TS1002
Model/manufacturer  _TZ3000_xa9g7rxs

Item is a similar product as a previous request:
https://github.com/Koenkk/zigbee2mqtt/issues/20645

### External definition

```shell
const {battery, commandsOnOff, commandsLevelCtrl, commandsColorCtrl} = require('zigbee-herdsman-converters/lib/modernExtend');

const definition = {
    zigbeeModel: ['TS1002'],
    model: 'TS1002',
    vendor: '_TZ3000_xa9g7rxs',
    description: 'Automatically generated definition',
    extend: [battery(), commandsOnOff(), commandsLevelCtrl(), commandsColorCtrl()],
    meta: {},
};

module.exports = definition;
```

### What does/doesn't work with the external definition?

Currently no features work under Z2M

Under ZHA the 
- on/off
- cct +/-
- brightness +/-
buttons are recognized but not functional without blueprint.

darknessblade, 2025-01-16:
Note: I recently started out with Z2M, so I haven't fully used it yet

Sergei7885, 2025-01-23:
Dear friend, did you manage to set up the remote control in z2m?

darknessblade, 2025-01-23:
> Dear friend, did you manage to set up the remote control in z2m?

I haven't really been able to set it up properly, 
I am quite new to Z2M. so not really familiar with it.
I mainly use ZHA. but have setup Z2M mainly for MMwave motion-sensors. and a few more zigbee devices [I have yet to order] that have more features under Z2M than ZHA.

Sergei7885, 2025-01-24:
I need an external converter, but I can't make it, I need experience, I need a new remote control, I need to wait until they add it to support

closed 2025-04-03 by github-actions[bot]