Koenkk/zigbee2mqtt #25861: Aqara switches QBKG11LM and QBKG12LM lost action hold

midlan, 2025-01-16:
### What happened?

With some update of zigbee2mqtt my switches QBKG11LM and QBKG12LM lost `hold` action. Now when I hold the switch, it reports it as `single` (the same as short press). I am 100% sure the `hold` action worked in some previous version of z2m.

Looking into zigbee-herdsman-converters it should support theese actions:
https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/src/devices/lumi.ts#L1032
https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/src/devices/lumi.ts#L1109-L1120

right?

my z2m versions:
```
Zigbee2MQTT version
2.0.0 commit: unknown
Coordinator type
zStack3x0
Coordinator revision
20240315
Coordinator IEEE Address
0x00124b00257c8f10
Frontend version
0.9.4
zigbee-herdsman-converters version
21.12.0
zigbee-herdsman version
3.2.1
```

### Zigbee2MQTT version

2.0.0

### Adapter firmware version

20240315

### Adapter

SLZB-06

### Setup

Add-on on haos, proxmox VM

midlan, 2025-01-17:
Before updating to 2.0.0 I was running 1.35.x and the problem was present there as well.

midlan, 2025-03-19:
physical hold is reported as single:

```
info 2025-03-19 19:43:19z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/VYP ložnice', payload '{"action":"single","consumption":24.21842384338379,"device_temperature":26,"energy":24.22,"last_seen":"2025-03-19T19:43:18+01:00","linkquality":18,"operation_mode":"decoupled","power":21.04,"power_outage_count":24038,"state":"ON","update":{"installed_version":95,"latest_version":95,"state":"idle"}}'
info 2025-03-19 19:43:19z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/VYP ložnice/action', payload 'single'
```

@Koenkk do you need other logs to debug the issue?

Koenkk, 2025-03-20:
Could  you provide the debug log when triggering a hold action?

See [this](https://www.zigbee2mqtt.io/guide/configuration/logging.html) on how to enable debug logging.

midlan, 2025-03-21:
QBKG11LM hold (reported wrong as single):
```
[2025-03-21 08:11:13] debug: 	zh:zstack:unpi:parser: <-- [254,28,68,129,0,0,18,0,126,135,5,1,0,105,0,109,167,194,0,0,8,24,72,10,85,0,33,1,0,126,135,29,148]
[2025-03-21 08:11:13] debug: 	zh:zstack:unpi:parser: --- parseNext [254,28,68,129,0,0,18,0,126,135,5,1,0,105,0,109,167,194,0,0,8,24,72,10,85,0,33,1,0,126,135,29,148]
[2025-03-21 08:11:13] debug: 	zh:zstack:unpi:parser: --> parsed 28 - 2 - 4 - 129 - [0,0,18,0,126,135,5,1,0,105,0,109,167,194,0,0,8,24,72,10,85,0,33,1,0,126,135,29] - 148
[2025-03-21 08:11:13] debug: 	zh:zstack:znp: <-- AREQ: AF - incomingMsg - {"groupid":0,"clusterid":18,"srcaddr":34686,"srcendpoint":5,"dstendpoint":1,"wasbroadcast":0,"linkquality":105,"securityuse":0,"timestamp":12756845,"transseqnumber":0,"len":8,"data":{"type":"Buffer","data":[24,72,10,85,0,33,1,0]}}
[2025-03-21 08:11:13] debug: 	zh:controller: Received payload: clusterID=18, address=34686, groupID=0, endpoint=5, destinationEndpoint=1, wasBroadcast=false, linkQuality=105, frame={"header":{"frameControl":{"frameType":0,"manufacturerSpecific":false,"direction":1,"disableDefaultResponse":true,"reservedBits":0},"transactionSequenceNumber":72,"commandIdentifier":10},"payload":[{"attrId":85,"dataType":33,"attrData":1}],"command":{"ID":10,"name":"report","parameters":[{"name":"attrId","type":33},{"name":"dataType","type":32},{"name":"attrData","type":1000}]}}
[2025-03-21 08:11:13] debug: 	zh:zstack:unpi:parser: --- parseNext []
[2025-03-21 08:11:13] debug: 	z2m: Received Zigbee message from 'VYP ložnice', type 'attributeReport', cluster 'genMultistateInput', data '{"presentValue":1}' from endpoint 5 with groupID 0
[2025-03-21 08:11:13] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/VYP ložnice', payload '{"action":"single","consumption":24.24286460876465,"device_temperature":26,"energy":24.24,"last_seen":"2025-03-21T08:11:13+01:00","linkquality":105,"operation_mode":"decoupled","power":22.1,"power_outage_count":24038,"state":"ON","update":{"installed_version":95,"latest_version":95,"state":"idle"}}'
[2025-03-21 08:11:13] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/VYP ložnice/action', payload 'single'
```

QBKG11LM single:
```
[2025-03-21 08:15:46] debug: 	zh:zstack:unpi:parser: <-- [254,3,69,196,126,135,0,123]
[2025-03-21 08:15:46] debug: 	zh:zstack:unpi:parser: --- parseNext [254,3,69,196,126,135,0,123]
[2025-03-21 08:15:46] debug: 	zh:zstack:unpi:parser: --> parsed 3 - 2 - 5 - 196 - [126,135,0] - 123
[2025-03-21 08:15:46] debug: 	zh:zstack:znp: <-- AREQ: ZDO - srcRtgInd
[2025-03-21 08:15:46] debug: 	zh:zstack:unpi:parser: --- parseNext []
[2025-03-21 08:15:46] debug: 	zh:zstack:unpi:parser: <-- [254,28,68,129,0,0,18,0,126,135,5,1,0,138,0,218,249,214,0,0,8,24,77,10,85,0,33,1,0,126,135,29,143]
[2025-03-21 08:15:46] debug: 	zh:zstack:unpi:parser: --- parseNext [254,28,68,129,0,0,18,0,126,135,5,1,0,138,0,218,249,214,0,0,8,24,77,10,85,0,33,1,0,126,135,29,143]
[2025-03-21 08:15:46] debug: 	zh:zstack:unpi:parser: --> parsed 28 - 2 - 4 - 129 - [0,0,18,0,126,135,5,1,0,138,0,218,249,214,0,0,8,24,77,10,85,0,33,1,0,126,135,29] - 143
[2025-03-21 08:15:46] debug: 	zh:zstack:znp: <-- AREQ: AF - incomingMsg - {"groupid":0,"clusterid":18,"srcaddr":34686,"srcendpoint":5,"dstendpoint":1,"wasbroadcast":0,"linkquality":138,"securityuse":0,"timestamp":14088666,"transseqnumber":0,"len":8,"data":{"type":"Buffer","data":[24,77,10,85,0,33,1,0]}}
[2025-03-21 08:15:46] debug: 	zh:controller: Received payload: clusterID=18, address=34686, groupID=0, endpoint=5, destinationEndpoint=1, wasBroadcast=false, linkQuality=138, frame={"header":{"frameControl":{"frameType":0,"manufacturerSpecific":false,"direction":1,"disableDefaultResponse":true,"reservedBits":0},"transactionSequenceNumber":77,"commandIdentifier":10},"payload":[{"attrId":85,"dataType":33,"attrData":1}],"command":{"ID":10,"name":"report","parameters":[{"name":"attrId","type":33},{"name":"dataType","type":32},{"name":"attrData","type":1000}]}}
[2025-03-21 08:15:46] debug: 	zh:zstack:unpi:parser: --- parseNext []
[2025-03-21 08:15:46] debug: 	z2m: Received Zigbee message from 'VYP ložnice', type 'attributeReport', cluster 'genMultistateInput', data '{"presentValue":1}' from endpoint 5 with groupID 0
[2025-03-21 08:15:46] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/VYP ložnice', payload '{"action":"single","consumption":24.24407196044922,"device_temperature":26,"energy":24.24,"last_seen":"2025-03-21T08:15:46+01:00","linkquality":138,"operation_mode":"decoupled","power":22.1,"power_outage_count":24038,"state":"ON","update":{"installed_version":95,"latest_version":95,"state":"idle"}}'
[2025-03-21 08:15:46] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/VYP ložnice/action', payload 'single'

```

midlan, 2025-03-21:
QBKG12LM hold left (reported wrong as single left)
```
[2025-03-21 08:19:58] debug: 	zh:zstack:unpi:parser: <-- [254,28,68,129,0,0,18,0,4,199,5,1,0,116,0,183,80,215,0,0,8,24,24,10,85,0,33,1,0,4,199,29,225]
[2025-03-21 08:19:58] debug: 	zh:zstack:unpi:parser: --- parseNext [254,28,68,129,0,0,18,0,4,199,5,1,0,116,0,183,80,215,0,0,8,24,24,10,85,0,33,1,0,4,199,29,225]
[2025-03-21 08:19:58] debug: 	zh:zstack:unpi:parser: --> parsed 28 - 2 - 4 - 129 - [0,0,18,0,4,199,5,1,0,116,0,183,80,215,0,0,8,24,24,10,85,0,33,1,0,4,199,29] - 225
[2025-03-21 08:19:58] debug: 	zh:zstack:znp: <-- AREQ: AF - incomingMsg - {"groupid":0,"clusterid":18,"srcaddr":50948,"srcendpoint":5,"dstendpoint":1,"wasbroadcast":0,"linkquality":116,"securityuse":0,"timestamp":14110903,"transseqnumber":0,"len":8,"data":{"type":"Buffer","data":[24,24,10,85,0,33,1,0]}}
[2025-03-21 08:19:58] debug: 	zh:controller: Received payload: clusterID=18, address=50948, groupID=0, endpoint=5, destinationEndpoint=1, wasBroadcast=false, linkQuality=116, frame={"header":{"frameControl":{"frameType":0,"manufacturerSpecific":false,"direction":1,"disableDefaultResponse":true,"reservedBits":0},"transactionSequenceNumber":24,"commandIdentifier":10},"payload":[{"attrId":85,"dataType":33,"attrData":1}],"command":{"ID":10,"name":"report","parameters":[{"name":"attrId","type":33},{"name":"dataType","type":32},{"name":"attrData","type":1000}]}}
[2025-03-21 08:19:58] debug: 	zh:zstack:unpi:parser: --- parseNext []
[2025-03-21 08:19:58] debug: 	z2m: Received Zigbee message from 'VYP obývák vlevo (řídící)', type 'attributeReport', cluster 'genMultistateInput', data '{"presentValue":1}' from endpoint 5 with groupID 0
[2025-03-21 08:19:58] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/VYP obývák vlevo (řídící)', payload '{"action":"single_left","consumption":179.6724395751953,"device_temperature":27,"energy":179.67,"last_seen":"2025-03-21T08:19:58+01:00","linkquality":116,"operation_mode_left":"decoupled","operation_mode_right":"decoupled","power":1.56,"power_outage_count":53756,"state_left":"ON","state_right":"ON","update":{"installed_version":95,"latest_version":95,"state":"idle"}}'
[2025-03-21 08:19:58] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/VYP obývák vlevo (řídící)/action', payload 'single_left'
[2025-03-21 08:19:58] debug: 	zh:zstack:unpi:parser: <-- [254,30,68,129,0,0,12,0,4,199,3,1,0,105,0,81,159,215,0,0,10,24,25,10,85,0,57,20,174,99,65,4,199,29,77]
[2025-03-21 08:19:58] debug: 	zh:zstack:unpi:parser: --- parseNext [254,30,68,129,0,0,12,0,4,199,3,1,0,105,0,81,159,215,0,0,10,24,25,10,85,0,57,20,174,99,65,4,199,29,77]
[2025-03-21 08:19:58] debug: 	zh:zstack:unpi:parser: --> parsed 30 - 2 - 4 - 129 - [0,0,12,0,4,199,3,1,0,105,0,81,159,215,0,0,10,24,25,10,85,0,57,20,174,99,65,4,199,29] - 77
[2025-03-21 08:19:58] debug: 	zh:zstack:znp: <-- AREQ: AF - incomingMsg - {"groupid":0,"clusterid":12,"srcaddr":50948,"srcendpoint":3,"dstendpoint":1,"wasbroadcast":0,"linkquality":105,"securityuse":0,"timestamp":14131025,"transseqnumber":0,"len":10,"data":{"type":"Buffer","data":[24,25,10,85,0,57,20,174,99,65]}}
[2025-03-21 08:19:58] debug: 	zh:controller: Received payload: clusterID=12, address=50948, groupID=0, endpoint=3, destinationEndpoint=1, wasBroadcast=false, linkQuality=105, frame={"header":{"frameControl":{"frameType":0,"manufacturerSpecific":false,"direction":1,"disableDefaultResponse":true,"reservedBits":0},"transactionSequenceNumber":25,"commandIdentifier":10},"payload":[{"attrId":85,"dataType":57,"attrData":14.229999542236328}],"command":{"ID":10,"name":"report","parameters":[{"name":"attrId","type":33},{"name":"dataType","type":32},{"name":"attrData","type":1000}]}}
[2025-03-21 08:19:58] debug: 	zh:zstack:unpi:parser: --- parseNext []
[2025-03-21 08:19:58] debug: 	z2m: Received Zigbee message from 'VYP obývák vlevo (řídící)', type 'attributeReport', cluster 'genAnalogInput', data '{"presentValue":14.229999542236328}' from endpoint 3 with groupID 0
[2025-03-21 08:19:58] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/VYP obývák vlevo (řídící)', payload '{"consumption":179.6724395751953,"device_temperature":27,"energy":179.67,"last_seen":"2025-03-21T08:19:58+01:00","linkquality":105,"operation_mode_left":"decoupled","operation_mode_right":"decoupled","power":14.23,"power_outage_count":53756,"state_left":"ON","state_right":"ON","update":{"installed_version":95,"latest_version":95,"state":"idle"}}'

```

QBKG12LM hold right (reported wrong as single right)
```
[2025-03-21 08:22:03] debug: 	zh:zstack:unpi:parser: <-- [254,30,68,129,0,0,12,0,4,199,3,1,0,153,0,150,84,94,0,0,10,24,28,10,85,0,57,184,30,197,63,4,199,29,249]
[2025-03-21 08:22:03] debug: 	zh:zstack:unpi:parser: --- parseNext [254,30,68,129,0,0,12,0,4,199,3,1,0,153,0,150,84,94,0,0,10,24,28,10,85,0,57,184,30,197,63,4,199,29,249]
[2025-03-21 08:22:03] debug: 	zh:zstack:unpi:parser: --> parsed 30 - 2 - 4 - 129 - [0,0,12,0,4,199,3,1,0,153,0,150,84,94,0,0,10,24,28,10,85,0,57,184,30,197,63,4,199,29] - 249
[2025-03-21 08:22:03] debug: 	zh:zstack:znp: <-- AREQ: AF - incomingMsg - {"groupid":0,"clusterid":12,"srcaddr":50948,"srcendpoint":3,"dstendpoint":1,"wasbroadcast":0,"linkquality":153,"securityuse":0,"timestamp":6182038,"transseqnumber":0,"len":10,"data":{"type":"Buffer","data":[24,28,10,85,0,57,184,30,197,63]}}
[... cut ...]
[2025-03-21 08:23:11] debug: 	z2m: Received Zigbee message from 'VYP obývák vlevo (řídící)', type 'attributeReport', cluster 'genAnalogInput', data '{"presentValue":40.33000183105469}' from endpoint 3 with groupID 0
[2025-03-21 08:23:11] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/VYP obývák vlevo (řídící)', payload '{"consumption":179.67320251464844,"device_temperature":26,"energy":179.67,"last_seen":"2025-03-21T08:23:11+01:00","linkquality":127,"operation_mode_left":"decoupled","operation_mode_right":"decoupled","power":40.33,"power_outage_count":53756,"state_left":"ON","state_right":"ON","update":{"installed_version":95,"latest_version":95,"state":"idle"}}'
```

midlan, 2025-03-21:
when hold much longer seems to report unknown key 65520

```
[2025-03-21 08:31:01] debug: 	zh:zstack:unpi:parser: <-- [254,3,69,196,126,135,0,123]
[2025-03-21 08:31:01] debug: 	zh:zstack:unpi:parser: --- parseNext [254,3,69,196,126,135,0,123]
[2025-03-21 08:31:01] debug: 	zh:zstack:unpi:parser: --> parsed 3 - 2 - 5 - 196 - [126,135,0] - 123
[2025-03-21 08:31:01] debug: 	zh:zstack:znp: <-- AREQ: ZDO - srcRtgInd
[2025-03-21 08:31:01] debug: 	zh:zstack:unpi:parser: --- parseNext []
[2025-03-21 08:31:01] debug: 	zh:zstack:unpi:parser: <-- [254,28,68,129,0,0,18,0,126,135,5,1,0,102,0,40,33,128,0,0,8,24,84,10,85,0,33,1,0,126,135,29,6]
[2025-03-21 08:31:01] debug: 	zh:zstack:unpi:parser: --- parseNext [254,28,68,129,0,0,18,0,126,135,5,1,0,102,0,40,33,128,0,0,8,24,84,10,85,0,33,1,0,126,135,29,6]
[2025-03-21 08:31:01] debug: 	zh:zstack:unpi:parser: --> parsed 28 - 2 - 4 - 129 - [0,0,18,0,126,135,5,1,0,102,0,40,33,128,0,0,8,24,84,10,85,0,33,1,0,126,135,29] - 6
[2025-03-21 08:31:01] debug: 	zh:zstack:znp: <-- AREQ: AF - incomingMsg - {"groupid":0,"clusterid":18,"srcaddr":34686,"srcendpoint":5,"dstendpoint":1,"wasbroadcast":0,"linkquality":102,"securityuse":0,"timestamp":8397096,"transseqnumber":0,"len":8,"data":{"type":"Buffer","data":[24,84,10,85,0,33,1,0]}}
[2025-03-21 08:31:01] debug: 	zh:controller: Received payload: clusterID=18, address=34686, groupID=0, endpoint=5, destinationEndpoint=1, wasBroadcast=false, linkQuality=102, frame={"header":{"frameControl":{"frameType":0,"manufacturerSpecific":false,"direction":1,"disableDefaultResponse":true,"reservedBits":0},"transactionSequenceNumber":84,"commandIdentifier":10},"payload":[{"attrId":85,"dataType":33,"attrData":1}],"command":{"ID":10,"name":"report","parameters":[{"name":"attrId","type":33},{"name":"dataType","type":32},{"name":"attrData","type":1000}]}}
[2025-03-21 08:31:01] debug: 	zh:zstack:unpi:parser: --- parseNext []
[2025-03-21 08:31:01] debug: 	z2m: Received Zigbee message from 'VYP ložnice', type 'attributeReport', cluster 'genMultistateInput', data '{"presentValue":1}' from endpoint 5 with groupID 0
[2025-03-21 08:31:01] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/VYP ložnice', payload '{"action":"single","consumption":24.249073028564453,"device_temperature":26,"energy":24.25,"last_seen":"2025-03-21T08:31:01+01:00","linkquality":102,"operation_mode":"decoupled","power":22.1,"power_outage_count":24038,"state":"ON","update":{"installed_version":95,"latest_version":95,"state":"idle"}}'
[2025-03-21 08:31:01] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/VYP ložnice/action', payload 'single'
[2025-03-21 08:31:04] debug: 	zh:zstack:unpi:parser: <-- [254,38,68,129,0,0,0,0,126,135,1,1,0,83,0,25,88,130,0,0,18,28,95,17,85,10,240,255,65,9,170,16,5,65,135,1,1,16,0,126,135,29,95]
[2025-03-21 08:31:04] debug: 	zh:zstack:unpi:parser: --- parseNext [254,38,68,129,0,0,0,0,126,135,1,1,0,83,0,25,88,130,0,0,18,28,95,17,85,10,240,255,65,9,170,16,5,65,135,1,1,16,0,126,135,29,95]
[2025-03-21 08:31:04] debug: 	zh:zstack:unpi:parser: --> parsed 38 - 2 - 4 - 129 - [0,0,0,0,126,135,1,1,0,83,0,25,88,130,0,0,18,28,95,17,85,10,240,255,65,9,170,16,5,65,135,1,1,16,0,126,135,29] - 95
[2025-03-21 08:31:04] debug: 	zh:zstack:znp: <-- AREQ: AF - incomingMsg - {"groupid":0,"clusterid":0,"srcaddr":34686,"srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"linkquality":83,"securityuse":0,"timestamp":8542233,"transseqnumber":0,"len":18,"data":{"type":"Buffer","data":[28,95,17,85,10,240,255,65,9,170,16,5,65,135,1,1,16,0]}}
[2025-03-21 08:31:04] debug: 	zh:controller: Received payload: clusterID=0, address=34686, groupID=0, endpoint=1, destinationEndpoint=1, wasBroadcast=false, linkQuality=83, frame={"header":{"frameControl":{"frameType":0,"manufacturerSpecific":true,"direction":1,"disableDefaultResponse":true,"reservedBits":0},"manufacturerCode":4447,"transactionSequenceNumber":85,"commandIdentifier":10},"payload":[{"attrId":65520,"dataType":65,"attrData":{"type":"Buffer","data":[170,16,5,65,135,1,1,16,0]}}],"command":{"ID":10,"name":"report","parameters":[{"name":"attrId","type":33},{"name":"dataType","type":32},{"name":"attrData","type":1000}]}}
[2025-03-21 08:31:04] debug: 	zh:zstack:unpi:parser: --- parseNext []
[2025-03-21 08:31:04] debug: 	z2m: Received Zigbee message from 'VYP ložnice', type 'attributeReport', cluster 'genBasic', data '{"65520":{"data":[170,16,5,65,135,1,1,16,0],"type":"Buffer"}}' from endpoint 1 with groupID 0
[2025-03-21 08:31:04] debug: 	zhc:lumi: QBKG11LM: unknown key 65520 with value �A�
```

Koenkk, 2025-03-21:
Thanks for the logs! The Zigbee message send by the device seems to be exactly the same for hold and single, so we cannot distinguish it.

>QBKG11LM hold (reported wrong as single):
>```
>[2025-03-21 08:11:13] debug: 	z2m: Received Zigbee message from 'VYP ložnice', type 'attributeReport', cluster 'genMultistateInput', data '{"presentValue":1}' from endpoint 5 with groupID 0
>```
>
>QBKG11LM single (reported properly, just for reference):
>```
>[2025-03-21 08:15:46] debug: 	z2m: Received Zigbee message from 'VYP ložnice', type 'attributeReport', cluster 'genMultistateInput', data '{"presentValue":1}' from endpoint 5 with groupID 0
>
>```
>
>

> I am 100% sure the hold action worked in some previous version of z2m.

Could you figure out which version and provide the debug logging for hold and click?

midlan, 2025-03-24:
> Thanks for the logs! The Zigbee message send by the device seems to be exactly the same for hold and single, so we cannot distinguish it.

So then it must have been changed by firmware update rather than z2m update. I installed at least one or two firmwares updates during the time I am using the switches.

What means that "unknown key 65520" message? May that be the `release` action? [I saw in source code](https://github.com/Koenkk/zigbee-herdsman-converters/blob/67cc0cf78954ace63fc7ab19a08995fb15306962/src/devices/lumi.ts#L1015) but never seen it actully be reported by the switch.

closed 2025-05-31 by github-actions[bot]