Koenkk/zigbee2mqtt #25864: Maginon WT-1 new Version?

dermigoe, 2025-01-17:
### What happened?

Hello,
I ordered a Maginon WT-1 thermostat and apparently something has been changed in the hardware.
It looks a little different than in the manufacturer's pictures and is recognized in zigbee2mqtt like this:
```
Device type
EndDevice
Zigbee Model
TS0601
Zigbee Manufacturer
_TZE204_p1qrtljn
Description
Smart radiator valve
Support status
Supported
Manufacturer
TECH
Model
VNTH-T2
```
The picture on the zigbee2mqtt [device page](https://www.zigbee2mqtt.io/devices/VNTH-T2.html#tech-vnth-t2) matches the thermostat I have.

BUT:
Unfortunately there are 2 problems with it.
1. I can't switch the system_mode to "auto". There is only "heat" and "off".
2. When I use the preset "schedule" the target temperature jumps to 153.6°C. The problem has already been described here for another thermostat: https://github.com/Koenkk/zigbee2mqtt/issues/19787

I (a layman) then tried to create an external converter and change it. To do this, I copied the contents of zigbee-herdsman-converters/src/devices/tech.ts into the file and then only changed the description to see if it worked. I then saved this file under tech2.js in the path data/external_converters and restarted zigbee2mqtt.
Unfortunately I was unsuccessful.
Is this because the thermostat is already supported?
Or do you have an idea how I can get the external converter to work?

Greetings and thanks

### Zigbee2MQTT version

2.0.0 commit: 060ae99

### Adapter firmware version

20240710

### Adapter

SONOFF Zigbee 3.0 USB Dongle Plus ZBDongle-P

### Setup

plain on a Raspberry Pi 3 Model B

dermigoe, 2025-01-21:
Unfortunately, I couldn't get the external converter to work, so I edited the tech.js in the zigbee-herdsman-converters folder and restarted zigbee2mqtt over and over again. Now I have a solution that works for me. I created a separate definition for each hardware version of the thermostat and only adapted the version I have (_TZE204_p1qrtljn). It works for me like this, and I hope the other version works well with this definition too. I can't test the version _TZE204_r7brscr6 because I don't have this thermostat.

### **Can an expert please take another look to see if this is all correct before I make a pull request?**

**Thanks!**

```
"use strict";
var __createBinding = (this && this.__createBinding) || (Object.create ? (function(o, m, k, k2) {
    if (k2 === undefined) k2 = k;
    var desc = Object.getOwnPropertyDescriptor(m, k);
    if (!desc || ("get" in desc ? !m.__esModule : desc.writable || desc.configurable)) {
      desc = { enumerable: true, get: function() { return m[k]; } };
    }
    Object.defineProperty(o, k2, desc);
}) : (function(o, m, k, k2) {
    if (k2 === undefined) k2 = k;
    o[k2] = m[k];
}));
var __setModuleDefault = (this && this.__setModuleDefault) || (Object.create ? (function(o, v) {
    Object.defineProperty(o, "default", { enumerable: true, value: v });
}) : function(o, v) {
    o["default"] = v;
});
var __importStar = (this && this.__importStar) || (function () {
    var ownKeys = function(o) {
        ownKeys = Object.getOwnPropertyNames || function (o) {
            var ar = [];
            for (var k in o) if (Object.prototype.hasOwnProperty.call(o, k)) ar[ar.length] = k;
            return ar;
        };
        return ownKeys(o);
    };
    return function (mod) {
        if (mod && mod.__esModule) return mod;
        var result = {};
        if (mod != null) for (var k = ownKeys(mod), i = 0; i < k.length; i++) if (k[i] !== "default") __createBinding(result, mod, k[i]);
        __setModuleDefault(result, mod);
        return result;
    };
})();
Object.defineProperty(exports, "__esModule", { value: true });
const exposes = __importStar(require("../lib/exposes"));
const tuya = __importStar(require("../lib/tuya"));
const e = exposes.presets;
const ea = exposes.access;
const definitions = [
    {
        fingerprint: [
            { modelID: 'TS0601', manufacturerName: '_TZE204_r7brscr6' },
        ],
        model: 'VNTH-T2',
        vendor: 'TECH',
        description: 'Smart radiator valve',
        ota: true,
        onEvent: tuya.onEventSetLocalTime,
        fromZigbee: [tuya.fz.datapoints],
        toZigbee: [tuya.tz.datapoints],
        configure: tuya.configureMagicPacket,
        exposes: [
            e.child_lock(),
            e.window_detection(),
            e.window_open(),
            e
                .climate()
                .withSetpoint('current_heating_setpoint', 5, 35, 0.5, ea.STATE_SET)
                .withLocalTemperature(ea.STATE)
                .withSystemMode(['heat', 'auto', 'off'], ea.STATE_SET, 'Mode of this device, in the `heat` mode the TS0601 will remain continuously heating, i.e. it does not regulate ' +
                'to the desired temperature. If you want TRV to properly regulate the temperature you need to use mode `auto` ' +
                'instead setting the desired temperature.')
[... 127 lines cut ...]
                    'preset',
                    tuya.valueConverterBasic.lookup({
                        manual: tuya.enum(0), // Hand / "normal"
                        schedule: tuya.enum(1), // Clock / "Auto"
                        eco: tuya.enum(2), // "E" / Moon
                        comfort: tuya.enum(3), // Leaf / Sun
                        antifrost: tuya.enum(4), // Flake
                        holiday: tuya.enum(5), // Palm tree
                    }),
                ],
                [28, 'schedule_monday', tuya.valueConverter.thermostatScheduleDayMultiDPWithDayNumber(1, 6)],
                [29, 'schedule_tuesday', tuya.valueConverter.thermostatScheduleDayMultiDPWithDayNumber(2, 6)],
                [30, 'schedule_wednesday', tuya.valueConverter.thermostatScheduleDayMultiDPWithDayNumber(3, 6)],
                [31, 'schedule_thursday', tuya.valueConverter.thermostatScheduleDayMultiDPWithDayNumber(4, 6)],
                [32, 'schedule_friday', tuya.valueConverter.thermostatScheduleDayMultiDPWithDayNumber(5, 6)],
                [33, 'schedule_saturday', tuya.valueConverter.thermostatScheduleDayMultiDPWithDayNumber(6, 6)],
                [34, 'schedule_sunday', tuya.valueConverter.thermostatScheduleDayMultiDPWithDayNumber(7, 6)],
            ],
        },
    },
];
exports.default = definitions;
module.exports = definitions;
//# sourceMappingURL=tech.js.map
```

gaertnert, 2025-02-13:
Is this now fixed? Does the latest version work with the described Thermostat Maginon WT-1?

dermigoe, 2025-02-13:
I created a [PR](https://github.com/Koenkk/zigbee-herdsman-converters/pull/8750) and it works for me so far. Unfortunately, I'm not a professional programmer. I could only test the new version because that's the only one I have available.

gaertnert, 2025-02-14:
It works! 👍 
Thank you!