Koenkk/zigbee2mqtt #25871: [New device support]: Temperature & Humidity Sensor - Tuya based from Nova Digital

gui28347, 2025-01-17:
### Link

https://www.lojageeksmart.com.br/sensor-inteligenter-de-temperatura-e-umidade-zigbee-com-display-nova-digital-tuya

### Database entry

 Device '0xa4c1382c83a7c150' with Zigbee model 'TS0601' and manufacturer name '_TZE284_hdyjyqjm' is NOT supported, please follow https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html

### Zigbee2MQTT version

2.0.0

### Comments

Device from "NovaDigital" is temperature & humidity sensor (Tuya based) , currently not supported.

I found in https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/src/devices/tuya.ts a similar one which I believe is the same hardware. TS0201.

1) Is there any way how I can do a local test with my device (_TZE284_hdyjyqjm) forcing it to be a TS0201 to see if it can read temperature & humidity?

2) After I test it, what would be the process to make it official support for it?

Debug Logs:

2025-01-17 15:52:15z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0xa4c1382c83a7c150","ieee_address":"0xa4c1382c83a7c150"},"type":"device_announce"}'
info 2025-01-17 15:52:17zh:controller:device: Device '0xa4c1382c83a7c150' is only compliant to revision '21' of the ZigBee specification (current revision: 23).
info 2025-01-17 15:52:21zh:controller: Succesfully interviewed '0xa4c1382c83a7c150'
info 2025-01-17 15:52:21z2m: Successfully interviewed '0xa4c1382c83a7c150', device has successfully been paired
warning 2025-01-17 15:52:21z2m: Device '0xa4c1382c83a7c150' with Zigbee model 'TS0601' and manufacturer name '_TZE284_hdyjyqjm' is NOT supported, please follow https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html
info 2025-01-17 15:52:21z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0xa4c1382c83a7c150","ieee_address":"0xa4c1382c83a7c150","status":"successful","supported":false},"type":"device_interview"}'

[2025-01-17 15:59:34] debug: zh:controller:endpoint: ZCL command 0xa4c1382c83a7c150/1 genBasic.defaultRsp({"cmdId":10,"statusCode":0}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":1,"reservedBits":0,"transactionSequenceNumber":10,"writeUndiv":false})
[2025-01-17 15:59:34] debug: zh:zstack: sendZclFrameToEndpointInternal 0xa4c1382c83a7c150:23365/1 (0,0,1)
[2025-01-17 15:59:34] debug: zh:zstack:znp: --> SREQ: AF - dataRequest - {"dstaddr":23365,"destendpoint":1,"srcendpoint":1,"clusterid":0,"transid":4,"options":0,"radius":30,"len":5,"data":{"type":"Buffer","data":[24,10,11,10,0]}}
[2025-01-17 15:59:34] debug: zh:zstack:unpi:writer: --> frame [254,15,36,1,69,91,1,1,0,0,4,0,30,5,24,10,11,10,0,56]
[2025-01-17 15:59:34] debug: zh:zstack:unpi:parser: --- parseNext []
[2025-01-17 15:59:34] debug: z2m: Received Zigbee message from '0xa4c1382c83a7c150', type 'attributeReport', cluster 'genBasic', data '{"65506":56,"65508":1,"appVersion":77}' from endpoint 1 with groupID 0
[2025-01-17 15:59:34] debug: z2m: Skipping message, still interviewing
[2025-01-17 15:59:34] debug: zh:zstack:unpi:parser: <-- [254,1,100,1,0,100]
[2025-01-17 15:59:34] debug: zh:zstack:unpi:parser: --- parseNext [254,1,100,1,0,100]
[2025-01-17 15:59:34] debug: zh:zstack:unpi:parser: --> parsed 1 - 3 - 4 - 1 - [0] - 100
[2025-01-17 15:59:34] debug: zh:zstack:znp: <-- SRSP: AF - dataRequest - {"status":0}
[2025-01-17 15:59:34] debug: zh:zstack:unpi:parser: --- parseNext []
[2025-01-17 15:59:34] debug: zh:zstack:unpi:parser: <-- [254,3,68,128,0,1,4,194]
[2025-01-17 15:59:34] debug: zh:zstack:unpi:parser: --- parseNext [254,3,68,128,0,1,4,194]
[2025-01-17 15:59:34] debug: zh:zstack:unpi:parser: --> parsed 3 - 2 - 4 - 128 - [0,1,4] - 194
[2025-01-17 15:59:34] debug: zh:zstack:znp: <-- AREQ: AF - dataConfirm - {"status":0,"endpoint":1,"transid":4}
[2025-01-17 15:59:34] debug: zh:zstack:unpi:parser: --- parseNext []
[2025-01-17 16:00:53] debug: zh:zstack:unpi:parser: <-- [254,42,68,129,0,0,0,0,69,91,1,1,0,98,0,143,227,115,0,0,22,8,11,10,223,255,66,15,218,58,29,47,25,218,58,29,47,7,1,0,0,0,17,69,91,29,243]
[2025-01-17 16:00:53] debug: zh:zstack:unpi:parser: --- parseNext [254,42,68,129,0,0,0,0,69,91,1,1,0,98,0,143,227,115,0,0,22,8,11,10,223,255,66,15,218,58,29,47,25,218,58,29,47,7,1,0,0,0,17,69,91,29,243]
[2025-01-17 16:00:53] debug: zh:zstack:unpi:parser: --> parsed 42 - 2 - 4 - 129 - [0,0,0,0,69,91,1,1,0,98,0,143,227,115,0,0,22,8,11,10,223,255,66,15,218,58,29,47,25,218,58,29,47,7,1,0,0,0,17,69,91,29] - 243
[2025-01-17 16:00:53] debug: zh:zstack:znp: <-- AREQ: AF - incomingMsg - {"groupid":0,"clusterid":0,"srcaddr":23365,"srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"linkquality":98,"securityuse":0,"timestamp":7594895,"transseqnumber":0,"len":22,"data":{"type":"Buffer","data":[8,11,10,223,255,66,15,218,58,29,47,25,218,58,29,47,7,1,0,0,0,17]}}
[2025-01-17 16:00:53] debug: zh:controller: Received payload: clusterID=0, address=23365, groupID=0, endpoint=1, destinationEndpoint=1, wasBroadcast=false, linkQuality=98, frame={"header":{"frameControl":{"frameType":0,"manufacturerSpecific":false,"direction":1,"disableDefaultResponse":false,"reservedBits":0},"transactionSequenceNumber":11,"commandIdentifier":10},"payload":[{"attrId":65503,"dataType":66,"attrData":" :\u001d/\u0019 :\u001d/\u0007\u0001\u0000\u0000\u0000\u0011"}],"command":{"ID":10,"name":"report","parameters":[{"name":"attrId","type":33},{"name":"dataType","type":32},{"name":"attrData","type":1000}]}}
[2025-01-17 16:00:53] debug: zh:controller:endpoint: ZCL command 0xa4c1382c83a7c150/1 genBasic.defaultRsp({"cmdId":10,"statusCode":0}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":1,"reservedBits":0,"transactionSequenceNumber":11,"writeUndiv":false})d
[2025-01-17 16:00:53] debug: zh:zstack: sendZclFrameToEndpointInternal 0xa4c1382c83a7c150:23365/1 (0,0,1)
[2025-01-17 16:00:53] debug: zh:zstack:znp: --> SREQ: AF - dataRequest - {"dstaddr":23365,"destendpoint":1,"srcendpoint":1,"clusterid":0,"transid":5,"options":0,"radius":30,"len":5,"data":{"type":"Buffer","data":[24,11,11,10,0]}}
[2025-01-17 16:00:53] debug: zh:zstack:unpi:writer: --> frame [254,15,36,1,69,91,1,1,0,0,5,0,30,5,24,11,11,10,0,56]
[2025-01-17 16:00:53] debug: zh:zstack:unpi:parser: --- parseNext []
[2025-01-17 16:00:53] debug: z2m: Received Zigbee message from '0xa4c1382c83a7c150', type 'attributeReport', cluster 'genBasic', data '{"65503":" :\u001d/\u0019 :\u001d/\u0007\u0001\u0000\u0000\u0000\u0011"}' from endpoint 1 with groupID 0
[2025-01-17 16:00:53] debug: z2m: Skipping message, still interviewing
[2025-01-17 16:00:53] debug: zh:zstack:unpi:parser: <-- [254,1,100,1,0,100]
[2025-01-17 16:00:53] debug: zh:zstack:unpi:parser: --- parseNext [254,1,100,1,0,100]
[2025-01-17 16:00:53] debug: zh:zstack:unpi:parser: --> parsed 1 - 3 - 4 - 1 - [0] - 100
[2025-01-17 16:00:53] debug: zh:zstack:znp: <-- SRSP: AF - dataRequest - {"status":0}
[2025-01-17 16:00:53] debug: zh:zstack:unpi:parser: --- parseNext []
[2025-01-17 16:00:53] debug: zh:zstack:unpi:parser: <-- [254,3,68,128,0,1,5,195]
[2025-01-17 16:00:53] debug: zh:zstack:unpi:parser: --- parseNext [254,3,68,128,0,1,5,195]
[2025-01-17 16:00:53] debug: zh:zstack:unpi:parser: --> parsed 3 - 2 - 4 - 128 - [0,1,5] - 195
[2025-01-17 16:00:53] debug: zh:zstack:znp: <-- AREQ: AF - dataConfirm - {"status":0,"endpoint":1,"transid":5}
[2025-01-17 16:00:53] debug: zh:zstack:unpi:parser: --- parseNext []
[... 37 lines cut ...]
[2025-01-17 16:01:32] debug: zh:zstack:unpi:parser: --> parsed 3 - 2 - 4 - 128 - [0,1,7] - 193
[2025-01-17 16:01:32] debug: zh:zstack:znp: <-- AREQ: AF - dataConfirm - {"status":0,"endpoint":1,"transid":7}
[2025-01-17 16:01:32] debug: zh:zstack:unpi:parser: --- parseNext []
[2025-01-17 16:01:32] debug: zh:zstack:unpi:parser: <-- [254,35,68,129,0,0,0,0,69,91,1,1,0,98,0,147,28,153,0,0,15,8,202,10,1,0,32,77,226,255,32,56,228,255,32,1,69,91,29,27]
[2025-01-17 16:01:32] debug: zh:zstack:unpi:parser: --- parseNext [254,35,68,129,0,0,0,0,69,91,1,1,0,98,0,147,28,153,0,0,15,8,202,10,1,0,32,77,226,255,32,56,228,255,32,1,69,91,29,27]
[2025-01-17 16:01:32] debug: zh:zstack:unpi:parser: --> parsed 35 - 2 - 4 - 129 - [0,0,0,0,69,91,1,1,0,98,0,147,28,153,0,0,15,8,202,10,1,0,32,77,226,255,32,56,228,255,32,1,69,91,29] - 27
[... cut ...] urityuse":0,"timestamp":10034323,"transseqnumber":0,"len":15,"data":{"type":"Buffer","data":[8,202,10,1,0,32,77,226,255,32,56,228,255,32,1]}}
[2025-01-17 16:01:32] debug: zh:controller: Receiv

### External definition

```shell
const definition = {
    zigbeeModel: ['TS0601'],
    model: 'TS0601',
    vendor: '_TZE284_hdyjyqjm',
    description: 'Automatically generated definition',
    extend: [],
    meta: {},
};
```

### What does/doesn't work with the external definition?

Tuya based device?

aJesus37, 2025-02-02:
+1, I also have this device and am interested in the support

barrymac, 2025-04-09:
I managed to get these working eventually with the following external converter: 

```
import { temperature, humidity, battery } from 'zigbee-herdsman-converters/lib/modernExtend';

export default {
    // Use fingerprint to identify the device
    fingerprint: [
        { modelID: 'TS0601', manufacturerName: '_TZE200_bq5c8xfe' },
        { modelID: 'TS0601', manufacturerName: '_TZE200_bjawzodf' },
        { modelID: 'TS0601', manufacturerName: '_TZE200_qyflbnbj' },
        { modelID: 'TS0601', manufacturerName: '_TZE200_9yapgbuv' },
        { modelID: 'TS0601', manufacturerName: '_TZE200_vs0skpuc' },
        { modelID: 'TS0601', manufacturerName: '_TZE284_hdyjyqjm' },
        { modelID: 'TS0601', manufacturerName: '_TZE200_zl1kmjqx' }
    ],
    model: 'TS0601_temperature_humidity_sensor',
    vendor: 'TuYa',
    description: 'Temperature & humidity sensor',
    // Use modernExtend to add common functionality
    extend: [temperature(), humidity(), battery()],
}
```

saved as /opt/zigbee2mqtt/data/external_converters/tuya-temp-sensor.mjs

engel102030, 2025-04-13:
here is the working external converter for novadigital temperature and humidity sensor:

`const exposes = require('zigbee-herdsman-converters/lib/exposes');
const e = exposes.presets;
const ea = exposes.access;
const tuya = require('zigbee-herdsman-converters/lib/tuya');

const ts0601_temperature_humidity_custom = {
    fingerprint: [{modelID: 'TS0601', manufacturerName: '_TZE284_hdyjyqjm'}],
    model: 'TS0601_TEMPHUMID',
    vendor: 'Tuya',
    description: 'Temperature and humidity sensor (custom)',
    fromZigbee: [tuya.fz.datapoints],
    toZigbee: [tuya.tz.datapoints],
    onEvent: tuya.onEvent({queryOnDeviceAnnounce: true}),
    configure: async (device, coordinatorEndpoint) => {
        await tuya.configureMagicPacket(device, coordinatorEndpoint);
        await device.getEndpoint(1).command("manuSpecificTuya", "dataQuery", {});
    },
    exposes: [
        e.temperature(), 
        e.humidity(), 
        e.enum("temperature_unit", ea.STATE, ["celsius", "fahrenheit"]).withDescription("Temperature unit"),
        e.enum("battery_level", ea.STATE, ["low", "medium", "high"]).withDescription("Battery level")
    ],
    meta: {
        tuyaDatapoints: [
            [1, "temperature", tuya.valueConverter.divideBy10],
            [2, "humidity", tuya.valueConverter.raw],
            [3, "battery_level", tuya.valueConverterBasic.lookup({low: tuya.enum(0), medium: tuya.enum(1), high: tuya.enum(2)})],
            [9, "temperature_unit", tuya.valueConverter.temperatureUnitEnum]
        ],
    },
};

module.exports = ts0601_temperature_humidity_custom;`

barrymac, 2025-04-15:
```
info 2025-04-15 12:59:31z2m: Configuring 'NovaDigiTemp1'
error 2025-04-15 12:59:32z2m: Failed to configure 'NovaDigiTemp1', attempt 2 (Error: Device 0xa4c138c28c29deec has no input cluster msTemperatureMeasurement at getEndpointsWithCluster (/app/node_modules/.pnpm/zigbee-herdsman-converters@23.27.0/node_modules/zigbee-herdsman-converters/src/lib/modernExtend.ts:61:15) at setupAttributes (/app/node_modules/.pnpm/zigbee-herdsman-converters@23.27.0/node_modules/zigbee-herdsman-converters/src/lib/modernExtend.ts:146:55) at configure (/app/node_modules/.pnpm/zigbee-herdsman-converters@23.27.0/node_modules/zigbee-herdsman-converters/src/lib/modernExtend.ts:209:23) at Object.configure (/app/node_modules/.pnpm/zigbee-herdsman-converters@23.27.0/node_modules/zigbee-herdsman-converters/src/index.ts:363:27) at Configure.configure (/app/lib/extension/configure.ts:124:37) at /app/lib/extension/configure.ts:82:62 at EventEmitter.wrappedCallback (/app/lib/eventBus.ts:204:23) at EventEmitter.emit (node:events:536:35) at EventBus.emitLastSeenChanged (/app/lib/eventBus.ts:74:22) at Controller.<anonymous> (/app/lib/zigbee.ts:80:27))
```

unfortunately no matter what I try, I still get this error. I'm giving up on these devices, and will be the last novadigital I will buy.

rc125, 2025-04-20:
Working converter

`import * as tuya from 'zigbee-herdsman-converters/lib/tuya';
import {presets} from 'zigbee-herdsman-converters/lib/exposes';

export default {
    fingerprint: tuya.fingerprint('TS0601', ['_TZE284_hdyjyqjm']),
    model: 'STU‑ZBD',
    vendor: 'NovaDigital',
    description: 'Temperature & humidity sensor with display',
    fromZigbee: [tuya.fz.datapoints],
    toZigbee:   [tuya.tz.datapoints],
    onEvent:    tuya.onEvent({queryOnDeviceAnnounce: true}),
    async configure(device, coordinatorEndpoint) {
        await tuya.configureMagicPacket(device, coordinatorEndpoint);
        await device.getEndpoint(1).command('manuSpecificTuya', 'dataQuery', {});
    },
    exposes: [presets.temperature(), presets.humidity(), tuya.exposes.batteryState(), tuya.exposes.temperatureUnit()],
    meta: {
        tuyaDatapoints: [
            [1, "temperature", tuya.valueConverter.divideBy10],
            [2, "humidity", tuya.valueConverter.raw],
            [3, "battery_state", tuya.valueConverter.batteryState],
            [9, "temperature_unit", tuya.valueConverter.temperatureUnitEnum],
        ]
    }
};`