Koenkk/zigbee2mqtt #25873: Failed to call Bridge Start & unable to pair (no devices discovered)

dafocusz, 2025-01-17:
### What happened?

Recently I upgraded to a more recent build.
- Zigbee2MQTT version 2.0.0 (commit #060ae99cff715cc9ef6a0cfa7cae0cdcb17f0c8a)
- zigbee-herdsman (3.2.1) started
- adapter doscovery matched

Im running Z2M in a docker container. Using a Conbee II stick with latest firmware. Before setting up the new container everything was working well.

I now have two suspicious observations:
- in the configuration I kept all my zigbee devices with their IDs and friendly names. They are not visible in the front end.
- I tried to pair devices again. But it seems I cannot activate "permit join (all)" on front end (at least no visible reaction after clicking)
 
Checking the log file I spotted two error messages:

> [2025-01-17 22:40:12] error: 	z2m: Failed to call 'Bridge' 'start' (TypeError: Cannot read properties of undefined (reading 'getDevice')
    at Bridge.publishInfo (/app/lib/extension/bridge.ts:692:69)
    at Bridge.start (/app/lib/extension/bridge.ts:179:9)
    at Controller.callExtensions (/app/lib/controller.ts:368:17)
    at Controller.start (/app/lib/controller.ts:186:9)
    at start (/app/index.js:161:5))
[2025-01-17 22:40:12] info: 	z2m: Started frontend on port 8080
[2025-01-17 22:40:12] error: 	z2m: Failed to call 'HomeAssistant' 'start' (TypeError: Cannot read properties of undefined (reading 'deviceIeeeAddress')
    at HomeAssistant.getBridgeEntity (/app/lib/extension/homeassistant.ts:1918:78)
    at HomeAssistant.start (/app/lib/extension/homeassistant.ts:439:28)
    at Controller.callExtensions (/app/lib/controller.ts:368:17)
    at Controller.start (/app/lib/controller.ts:186:9)
    at start (/app/index.js:161:5))

### What did you expect to happen?

Launch container, known zigbee devices from config file loaded and everything is working.

### How to reproduce it (minimal and precise)

Whenever I launch my container...

[log.log](https://github.com/user-attachments/files/18460169/log.log)

### Zigbee2MQTT version

0.9.4 (front end; no other versions mentioned in front end)

### Adapter firmware version

deCONZ_ConBeeII_0x26780700.bin.GCF 

### Adapter

usb-dresden_elektronik_ingenieurtechnik_GmbH_ConBee_II_DE2668745-if00

### Setup

plain on docker container, raspberry pi

Koenkk, 2025-01-19:
Could you provide the debug log of stopping z2m, removing the `data/database.db` file, and starting z2m?

giovannigenna, 2025-01-26:
I’m experiencing the same issue. I’m using **HAOS 2025.1.4** and **Zigbee2MQTT version 2.0.0-2**. My setup runs on **Proxmox**, and I’ve attempted to expose the USB device using both the **Vendor/Device ID** and the **USB port**. 

I only modified the serial field as follows:
```
adapter: deconz
port: /dev/ttyACM0
```
I also tried using the device ID, but the same issue persists.
If I delete the ‘zigbee2mqtt’ folder and start fresh, I get the following log:
```
[12:42:43] INFO: Preparing to start...
[12:42:43] INFO: Socat not enabled
[12:42:43] INFO: Starting Zigbee2MQTT...
Starting Zigbee2MQTT without watchdog.
[2025-01-26 12:42:45] info: 	z2m: Logging to console, file (filename: log.log)
[2025-01-26 12:42:45] info: 	z2m: Starting Zigbee2MQTT version 2.0.0 (commit #unknown)
[2025-01-26 12:42:45] info: 	z2m: Starting zigbee-herdsman (3.2.1)
[2025-01-26 12:42:45] info: 	zh:adapter:discovery: Matched adapter: {"path":"/dev/ttyACM0","manufacturer":"dresden elektronik ingenieurtechnik GmbH","serialNumber":"DE2470390","pnpId":"usb-dresden_elektronik_ingenieurtechnik_GmbH_ConBee_II_DE2470390-if00","vendorId":"1cf1","productId":"0030"} => deconz: 4
[2025-01-26 12:43:51] error: 	z2m: Error while starting zigbee-herdsman
[2025-01-26 12:43:51] error: 	z2m: Failed to start zigbee-herdsman
[2025-01-26 12:43:51] error: 	z2m: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start_crashes-runtime.html for possible solutions
[2025-01-26 12:43:51] error: 	z2m: Exiting...
[2025-01-26 12:43:51] error: 	z2m: Error: waiting for response TIMEOUT
    at DeconzAdapter.checkReceivedDataPayload (/app/node_modules/.pnpm/zigbee-herdsman@3.2.1/node_modules/zigbee-herdsman/src/adapter/deconz/adapter/deconzAdapter.ts:703:28)
    at Timeout._onTimeout (/app/node_modules/.pnpm/zigbee-herdsman@3.2.1/node_modules/zigbee-herdsman/src/adapter/deconz/adapter/deconzAdapter.ts:75:18)
    at listOnTimeout (node:internal/timers:594:17)
    at processTimers (node:internal/timers:529:7)
```

If I stop the add-ons and restart them, I get the following log:
```
[12:45:15] INFO: Preparing to start...
[12:45:15] INFO: Socat not enabled
[12:45:16] INFO: Starting Zigbee2MQTT...
Starting Zigbee2MQTT without watchdog.
[2025-01-26 12:45:17] info: 	z2m: Logging to console, file (filename: log.log)
[2025-01-26 12:45:17] info: 	z2m: Starting Zigbee2MQTT version 2.0.0 (commit #unknown)
[2025-01-26 12:45:17] info: 	z2m: Starting zigbee-herdsman (3.2.1)
[2025-01-26 12:45:17] info: 	zh:adapter:discovery: Matched adapter: {"path":"/dev/ttyACM0","manufacturer":"dresden elektronik ingenieurtechnik GmbH","serialNumber":"DE2470390","pnpId":"usb-dresden_elektronik_ingenieurtechnik_GmbH_ConBee_II_DE2470390-if00","vendorId":"1cf1","productId":"0030"} => deconz: 4
[2025-01-26 12:45:18] info: 	z2m: zigbee-herdsman started (resumed)
[2025-01-26 12:45:18] info: 	z2m: Coordinator firmware version: '{"meta":{"maintrel":0,"majorrel":38,"minorrel":114,"product":0,"revision":"0x26720700","transportrev":0},"type":"ConBee2/RaspBee2"}'
[2025-01-26 12:45:18] info: 	z2m: Currently 0 devices are joined.
[2025-01-26 12:45:18] info: 	z2m: Connecting to MQTT server at mqtt://core-mosquitto:1883
[2025-01-26 12:45:18] info: 	z2m: Connected to MQTT server
[2025-01-26 12:45:18] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/state', payload '{"state":"online"}'
[2025-01-26 12:45:18] error: 	z2m: Failed to call 'Bridge' 'start' (TypeError: Cannot read properties of undefined (reading 'getDevice')
    at Bridge.publishInfo (/app/lib/extension/bridge.ts:692:69)
    at Bridge.start (/app/lib/extension/bridge.ts:179:9)
    at Controller.callExtensions (/app/lib/controller.ts:368:17)
    at Controller.start (/app/lib/controller.ts:186:9)
    at start (/app/index.js:161:5))
[2025-01-26 12:45:18] info: 	z2m: Started frontend on port 8099
[2025-01-26 12:45:18] error: 	z2m: Failed to call 'HomeAssistant' 'start' (TypeError: Cannot read properties of undefined (reading 'deviceIeeeAddress')
    at HomeAssistant.getBridgeEntity (/app/lib/extension/homeassistant.ts:1918:78)
    at HomeAssistant.start (/app/lib/extension/homeassistant.ts:439:28)
    at Controller.callExtensions (/app/lib/controller.ts:368:17)
    at Controller.start (/app/lib/controller.ts:186:9)
    at start (/app/index.js:161:5))
[2025-01-26 12:45:18] info: 	z2m: Zigbee2MQTT started!
[2025-01-26 12:45:20] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/state', payload '{"state":"online"}'
```

Here my files:
**zigbee2mqtt/configuration.yaml**
```
version: 4
homeassistant:
  enabled: true
advanced:
  network_key:
    - 114
    - 245
    - 157
    - 245
    - 15
    - 192
    - 153
    - 71
    - 123
    - 53
    - 29
    - 104
    - 179
    - 62
    - 29
    - 43
  pan_id: 12501
  ext_pan_id:
    - 210
    - 217
    - 77
    - 119
    - 205
    - 169
    - 186
    - 182
mqtt:
  server: mqtt://core-mosquitto:1883
  user: addons
  password: {PASSWORD_HERE}
serial:
  port: /dev/ttyACM0
  adapter: deconz
frontend:
  enabled: true
  port: 8099
```

**zigbee2mqtt/database.db**
`{"id":1,"type":"Coordinator","ieeeAddr":"0x00212effff07ce25","nwkAddr":0,"manufId":4405,"epList":[],"endpoints":{},"interviewCompleted":true,"meta":{}}`

Koenkk, 2025-01-27:
@giovannigenna please provide the debug log of this

See [this](https://www.zigbee2mqtt.io/guide/configuration/logging.html) on how to enable debug logging.

giovannigenna, 2025-01-27:
@Koenkk 
Thank you for your reply. Here is the log captured in debug mode: 
```
[20:57:26] INFO: Preparing to start...
[20:57:26] INFO: Socat not enabled
[20:57:26] INFO: Starting Zigbee2MQTT...
Starting Zigbee2MQTT without watchdog.
[2025-01-27 20:57:28] debug: 	z2m: Removing old log directory '/config/zigbee2mqtt/log/2025-01-25.23-16-09'
[2025-01-27 20:57:28] info: 	z2m: Logging to console, file (filename: log.log)
[2025-01-27 20:57:28] debug: 	z2m: Loaded state from file /config/zigbee2mqtt/state.json
[2025-01-27 20:57:28] info: 	z2m: Starting Zigbee2MQTT version 2.0.0 (commit #unknown)
[2025-01-27 20:57:28] debug: 	z2m: sd-notify loaded
[2025-01-27 20:57:28] info: 	z2m: Starting zigbee-herdsman (3.2.1)
[2025-01-27 20:57:28] debug: 	z2m: Using zigbee-herdsman with settings: '"{\"network\":{\"panID\":12501,\"extendedPanID\":[210,217,77,119,205,169,186,182],\"channelList\":[11],\"networkKey\":\"HIDDEN\"},\"databasePath\":\"/config/zigbee2mqtt/database.db\",\"databaseBackupPath\":\"/config/zigbee2mqtt/database.db.backup\",\"backupPath\":\"/config/zigbee2mqtt/coordinator_backup.json\",\"serialPort\":{\"path\":\"/dev/ttyACM0\",\"adapter\":\"deconz\"},\"adapter\":{\"disableLED\":false}}"'
[2025-01-27 20:57:28] debug: 	zh:adapter:discovery: Connected devices: [{"path":"/dev/ttyACM0","manufacturer":"dresden elektronik ingenieurtechnik GmbH","serialNumber":"DE2470390","pnpId":"usb-dresden_elektronik_ingenieurtechnik_GmbH_ConBee_II_DE2470390-if00","vendorId":"1cf1","productId":"0030"},{"path":"/dev/ttyS0"},{"path":"/dev/ttyS1"},{"path":"/dev/ttyS2"},{"path":"/dev/ttyS3"}]
[2025-01-27 20:57:28] info: 	zh:adapter:discovery: Matched adapter: {"path":"/dev/ttyACM0","manufacturer":"dresden elektronik ingenieurtechnik GmbH","serialNumber":"DE2470390","pnpId":"usb-dresden_elektronik_ingenieurtechnik_GmbH_ConBee_II_DE2470390-if00","vendorId":"1cf1","productId":"0030"} => deconz: 4
[2025-01-27 20:57:28] debug: 	zh:deconz:driver: Set delay to 0
[2025-01-27 20:57:28] debug: 	zh:controller: Starting with options '{"network":{"networkKeyDistribute":false,"networkKey":"HIDDEN","panID":12501,"extendedPanID":[210,217,77,119,205,169,186,182],"channelList":[11]},"serialPort":{"path":"/dev/ttyACM0","adapter":"deconz"},"adapter":{"disableLED":false},"databasePath":"/config/zigbee2mqtt/database.db","databaseBackupPath":"/config/zigbee2mqtt/database.db.backup","backupPath":"/config/zigbee2mqtt/coordinator_backup.json"}'
[2025-01-27 20:57:28] debug: 	zh:deconz:driver: Opening with /dev/ttyACM0
[2025-01-27 20:57:28] debug: 	zh:deconz:driver: Serialport opened
[2025-01-27 20:57:28] debug: 	zh:deconz:driver: send read parameter request from queue. seqNr: 1 paramId: 5
[2025-01-27 20:57:28] debug: 	zh:deconz:frameparser: PANID: 30d5
[2025-01-27 20:57:28] debug: 	zh:deconz:driver: send read parameter request from queue. seqNr: 2 paramId: 11
[2025-01-27 20:57:28] debug: 	zh:deconz:frameparser: APS_EXT_PANID: 0xd2d94d77cda9bab6
[2025-01-27 20:57:28] debug: 	zh:deconz:driver: send read parameter request from queue. seqNr: 3 paramId: 28
[2025-01-27 20:57:28] debug: 	zh:deconz:frameparser: CHANNEL: 11
[2025-01-27 20:57:28] debug: 	zh:deconz:driver: send read parameter request from queue. seqNr: 4 paramId: 24
[2025-01-27 20:57:28] debug: 	zh:deconz:frameparser: NETWORK_KEY: hidden
[2025-01-27 20:57:28] debug: 	zh:deconz:driver: send write parameter request from queue. seqNr: 5 paramId: 19 param: 5,2,5,0,0,32,0,1,4,5,1,0,25,0,10,6,0,0,0,5,1,0,5,1,4,1,0
[2025-01-27 20:57:28] debug: 	zh:deconz:frameparser: write parameter response - parameter id: 19 - status: 0
[2025-01-27 20:57:28] debug: 	zh:controller: Started with result 'resumed'
[2025-01-27 20:57:28] debug: 	zh:deconz:driver: send read parameter request from queue. seqNr: 6 paramId: 5
[2025-01-27 20:57:28] debug: 	zh:deconz:frameparser: DEVICE_STATE changed: 10100010
[2025-01-27 20:57:28] debug: 	zh:deconz:driver: networkstate: 2 apsDataConfirm: 0 apsDataIndication: 0 configChanged: 0 apsRequestFreeSlots: 1
[2025-01-27 20:57:28] debug: 	zh:deconz:frameparser: PANID: 30d5
[2025-01-27 20:57:28] debug: 	zh:deconz:driver: send read parameter request from queue. seqNr: 7 paramId: 11
[2025-01-27 20:57:28] debug: 	zh:deconz:frameparser: APS_EXT_PANID: 0xd2d94d77cda9bab6
[2025-01-27 20:57:28] debug: 	zh:deconz:driver: send read parameter request from queue. seqNr: 8 paramId: 28
[2025-01-27 20:57:28] debug: 	zh:deconz:frameparser: CHANNEL: 11
[2025-01-27 20:57:28] debug: 	zh:controller: Injected database: true, adapter: true
[2025-01-27 20:57:28] debug: 	zh:controller:device: Request Queue (0x00212effff07ce25): default expiration timeout set to 0
[2025-01-27 20:57:28] debug: 	zh:controller:database: Writing database to '/config/zigbee2mqtt/database.db'
[2025-01-27 20:57:28] debug: 	zh:deconz:driver: send read parameter request from queue. seqNr: 9 paramId: 1
[2025-01-27 20:57:28] debug: 	zh:deconz:frameparser: MAC: 0x00212effff07ce25
[2025-01-27 20:57:28] info: 	z2m: zigbee-herdsman started (resumed)
[2025-01-27 20:57:28] debug: 	zh:deconz:driver: send read firmware version request from queue. seqNr: 10
[2025-01-27 20:57:28] debug: 	zh:deconz:frameparser: read firmware version response - version: 0,7,114,38
[2025-01-27 20:57:28] info: 	z2m: Coordinator firmware version: '{"meta":{"maintrel":0,"majorrel":38,"minorrel":114,"product":0,"revision":"0x26720700","transportrev":0},"type":"ConBee2/RaspBee2"}'
[... cut ...]
[2025-01-27 21:08:05] error: 	z2m: Error: waiting for response TIMEOUT
    at DeconzAdapter.checkReceivedDataPayload (/app/node_modules/.pnpm/zigbee-herdsman@3.2.1/node_modules/zigbee-herdsman/src/adapter/deconz/adapter/deconzAdapter.ts:703:28)
    at Timeout._onTimeout (/app/node_modules/.pnpm/zigbee-herdsman@3.2.1/node_modules/zigbee-herdsman/src/adapter/deconz/adapter/deconzAdapter.ts:75:18)
    at listOnTimeout (node:internal/timers:594:17)
    at processTimers (node:internal/timers:529:7)
```

Do you have any insights on why this might be happening?

giovannigenna, 2025-01-27:
I ran some tests and tried changing the USB port, but nothing happened. Then, I decided to use a different USB extension cable, and now everything works like a charm.

I repeated the process multiple times: deleting the zigbee2mqtt folder, removing the USB dongle, and trying again. Each time, it worked without any issues.

It seems like the problem might have been related to my previous USB cable, although I’m not entirely sure. Interestingly, I tested the same USB cable with a USB stick, and it worked fine.

For now, everything is working perfectly, but I still don’t understand why switching the cable resolved the issue.

Pabicchio, 2025-02-23:
HAOS on RPi 5, ConBee II
I had the same problem and error, tried to switch the usb port and it didn't solve the issue. Rebooted the Pi, uninstalled the plugin and deleted the file, configured it (maybe a little different from between but I'm not that sure even if I followed the same guide) and now it seems to be working.

closed 2025-05-03 by github-actions[bot]