Koenkk/zigbee2mqtt #25877: Müller Licht 404049D sends "action" twice since update to 2.0.0

dkneisz, 2025-01-18:
### What happened?

Since I updated to version 2.0.0 I have the issue, that my Müller Licht Tint remotes are sending the action twice (see debug log). It seems that only the value for "action" is sent twice, but not the rest of the attributes and also not the JSON in the device topic (which for now I used as a workaround). This did not happen in version 1.42.0.

### What did you expect to happen?

I expect that the "action" is only sent once.

### How to reproduce it (minimal and precise)

- enable to send attributes in separate topics (forgot what that option is called)
- press an action button on the Müller Licht Tint remote

### Zigbee2MQTT version

2.0.0

### Adapter firmware version

20221226

### Adapter

zzh!

### Setup

Running in a docker container on a x64 mini PC running Debian 12

### Debug log

```
[2025-01-18 08:23:40] debug: 	z2m: Received Zigbee message from '<Device>', type 'write', cluster 'genBasic', data '{"16389":1}' from endpoint 1 with groupID 16387
[2025-01-18 08:23:40] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/<Device>', payload '{"action":"scene_1","action_group":16387,"device":{"applicationVersion":0,"friendlyName":"<Device>","hardwareVersion":0,"ieeeAddr":"0xa4c13800aa431e4b","manufacturerID":4635,"manufacturerName":"MLI","model":"404049D","networkAddress":2663,"powerSource":"Battery","softwareBuildID":"MDRM001","stackVersion":2,"type":"EndDevice","zclVersion":3},"last_seen":"2025-01-18T08:23:40+01:00","linkquality":102}'
[2025-01-18 08:23:40] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/<Device>/last_seen', payload '2025-01-18T08:23:40+01:00'
[2025-01-18 08:23:40] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/<Device>/linkquality', payload '102'
[2025-01-18 08:23:40] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/<Device>/action', payload 'scene_1'
[2025-01-18 08:23:40] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/<Device>/action_group', payload '16387'
[2025-01-18 08:23:40] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/<Device>/device-friendlyName', payload '<Device>'
[2025-01-18 08:23:40] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/<Device>/device-model', payload '404049D'
[2025-01-18 08:23:40] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/<Device>/device-ieeeAddr', payload '0xa4c13800aa431e4b'
[2025-01-18 08:23:40] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/<Device>/device-networkAddress', payload '2663'
[2025-01-18 08:23:40] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/<Device>/device-type', payload 'EndDevice'
[2025-01-18 08:23:40] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/<Device>/device-manufacturerID', payload '4635'
[2025-01-18 08:23:40] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/<Device>/device-powerSource', payload 'Battery'
[2025-01-18 08:23:40] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/<Device>/device-applicationVersion', payload '0'
[2025-01-18 08:23:40] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/<Device>/device-stackVersion', payload '2'
[2025-01-18 08:23:40] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/<Device>/device-zclVersion', payload '3'
[2025-01-18 08:23:40] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/<Device>/device-hardwareVersion', payload '0'
[2025-01-18 08:23:40] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/<Device>/device-dateCode', payload ''
[2025-01-18 08:23:40] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/<Device>/device-softwareBuildID', payload 'MDRM001'
[2025-01-18 08:23:40] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/<Device>/device-manufacturerName', payload 'MLI'
[2025-01-18 08:23:40] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/<Device>/action', payload 'scene_1'
```

Koenkk, 2025-01-19:
Could you provide the debug log of this?

See [this](https://www.zigbee2mqtt.io/guide/configuration/logging.html) on how to enable debug logging.

dkneisz, 2025-01-20:
I already sent the debug log. There was nothing more on this topic in the log. I will try it again, but I did this multiple times and there was nothing more in the log (and I set it to debug, as there is a debug message in the first line). I set this through the UI. Maybe it's different if I set it in configuration.yaml and restart the docker container. Do I need to set anything else or perhaps I missed some entries.

dkneisz, 2025-01-20:
So I tried this again and there is not much more. There are some debug messages before and after that, but I'm not sure if they belong to other devices. Here is the whole thing from after the publish action of the device before to before the publish action of the device after the remote:
```
"<[2025-01-20 19:33:42] debug: 	zh:zstack:unpi:parser: <-- [254,29,68,129,3,64,0,0,103,10,1,1,0,105,0,215,218,38,0,0,9,20,27,18,202,2,5,64,32,1,103,10]
[2025-01-20 19:33:42] debug: 	zh:zstack:unpi:parser: --- parseNext [254,29,68,129,3,64,0,0,103,10,1,1,0,105,0,215,218,38,0,0,9,20,27,18,202,2,5,64,32,1,103,10]
[2025-01-20 19:33:42] debug: 	zh:zstack:unpi:parser: <-- [29,124]
[2025-01-20 19:33:42] debug: 	zh:zstack:unpi:parser: --- parseNext [254,29,68,129,3,64,0,0,103,10,1,1,0,105,0,215,218,38,0,0,9,20,27,18,202,2,5,64,32,1,103,10,29,124]
[2025-01-20 19:33:42] debug: 	zh:zstack:unpi:parser: --> parsed 29 - 2 - 4 - 129 - [3,64,0,0,103,10,1,1,0,105,0,215,218,38,0,0,9,20,27,18,202,2,5,64,32,1,103,10,29] - 124
[2025-01-20 19:33:42] debug: 	zh:zstack:znp: <-- AREQ: AF - incomingMsg - {"groupid":16387,"clusterid":0,"srcaddr":2663,"srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"linkquality":105,"securityuse":0,"timestamp":2546391,"transseqnumber":0,"len":9,"data":{"type":"Buffer","data":[20,27,18,202,2,5,64,32,1]}}
[2025-01-20 19:33:42] debug: 	zh:controller: Received payload: clusterID=0, address=2663, groupID=16387, endpoint=1, destinationEndpoint=1, wasBroadcast=false, linkQuality=105, frame={"header":{"frameControl":{"frameType":0,"manufacturerSpecific":true,"direction":0,"disableDefaultResponse":true,"reservedBits":0},"manufacturerCode":4635,"transactionSequenceNumber":202,"commandIdentifier":2},"payload":[{"attrId":16389,"dataType":32,"attrData":1}],"command":{"ID":2,"name":"write","parameters":[{"name":"attrId","type":33},{"name":"dataType","type":32},{"name":"attrData","type":1000}],"response":4}}
[2025-01-20 19:33:42] debug: 	zh:zstack:unpi:parser: --- parseNext []
[2025-01-20 19:33:42] debug: 	z2m: Received Zigbee message from '<device>', type 'write', cluster 'genBasic', data '{"16389":1}' from endpoint 1 with groupID 16387
[2025-01-20 19:33:42] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/<device>', payload '{"action":"scene_1","action_group":16387,"device":{"applicationVersion":0,"friendlyName":"<device>","hardwareVersion":0,"ieeeAddr":"0xa4c13800aa431e4b","manufacturerID":4635,"manufacturerName":"MLI","model":"404049D","networkAddress":2663,"powerSource":"Battery","softwareBuildID":"MDRM001","stackVersion":2,"type":"EndDevice","zclVersion":3},"last_seen":"2025-01-20T19:33:42+01:00","linkquality":105}'
[2025-01-20 19:33:42] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/<device>/last_seen', payload '2025-01-20T19:33:42+01:00'
[2025-01-20 19:33:42] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/<device>/linkquality', payload '105'
[2025-01-20 19:33:42] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/<device>/action', payload 'scene_1'
[2025-01-20 19:33:42] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/<device>/action_group', payload '16387'
[2025-01-20 19:33:42] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/<device>/device-friendlyName', payload '<device>'
[2025-01-20 19:33:42] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/<device>/device-model', payload '404049D'
[2025-01-20 19:33:42] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/<device>/device-ieeeAddr', payload '0xa4c13800aa431e4b'
[2025-01-20 19:33:42] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/<device>/device-networkAddress', payload '2663'
[2025-01-20 19:33:42] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/<device>/device-type', payload 'EndDevice'
[2025-01-20 19:33:42] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/<device>/device-manufacturerID', payload '4635'
[2025-01-20 19:33:42] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/<device>/device-powerSource', payload 'Battery'
[2025-01-20 19:33:42] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/<device>/device-applicationVersion', payload '0'
[2025-01-20 19:33:42] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/<device>/device-stackVersion', payload '2'
[2025-01-20 19:33:42] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/<device>/device-zclVersion', payload '3'
[2025-01-20 19:33:42] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/<device>/device-hardwareVersion', payload '0'
[2025-01-20 19:33:42] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/<device>/device-dateCode', payload ''
[2025-01-20 19:33:42] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/<device>/device-softwareBuildID', payload 'MDRM001'
[2025-01-20 19:33:42] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/<device>/device-manufacturerName', payload 'MLI'
[2025-01-20 19:33:42] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/<device>/action', payload 'scene_1'
[2025-01-20 19:33:42] debug: 	zh:zstack:unpi:parser: <-- [254,9,69,196,68,56,3,153,47,82,202,220,181,176,254,35,68,129,0,0,0,0,68,56,1,1,0,163,0,13,5,39]
[2025-01-20 19:33:42] debug: 	zh:zstack:unpi:parser: --- parseNext [254,9,69,196,68,56,3,153,47,82,202,220,181,176,254,35,68,129,0,0,0,0,68,56,1,1,0,163,0,13,5,39]
[2025-01-20 19:33:42] debug: 	zh:zstack:unpi:parser: --> parsed 9 - 2 - 5 - 196 - [68,56,3,153,47,82,202,220,181] - 176
[2025-01-20 19:33:42] debug: 	zh:zstack:znp: <-- AREQ: ZDO - srcRtgInd
[... cut ...]
[2025-01-20 19:33:42] debug: 	zh:zstack: sendZclFrameToEndpointInternal 0xa4c1389ad423e657:14404/1 (0,0,1)
[2025-01-20 19:33:42] debug: 	zh:zstack:znp: --> SREQ: AF - dataRequest - {"dstaddr":14404,"destendpoint":1,"srcendpoint":1,"clusterid":0,"transid":92,"options":0,"radius":30,"len":5,"data":{"type":"Buffer","data":[24,122,11,10,0]}}
[2025-01-20 19:33:42] debug: 	zh:zstack:unpi:writer: --> frame [254,15,36,1,68,56,1,1,0,0,92,0,30,5,24,122,11,10,0,114]
[2025-01-20 19:33:42] debug: 	zh:zstack:unpi:parser: --- parseNext []
```

Koenkk, 2025-01-21:
@dkneisz could you paste your `configuration.yaml` (and `devices.yaml` in case you have one) here + `database.db` file?

dkneisz, 2025-01-23:
Here you go:
[z2m-data.zip](https://github.com/user-attachments/files/18517725/z2m-data.zip)

Koenkk, 2025-01-23:
Found and fixed the issue in https://github.com/Koenkk/zigbee2mqtt/pull/25963

Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now.

closed 2025-01-23 by Koenkk