Koenkk/zigbee2mqtt #25878: Device disappears after Z2M restart

Diadlo, 2025-01-18:
### What happened?

1. Device added successfully
2. Device can be controlled in UI before restart
3. Device is present in `database.db`
4. Device disappear from UI and from `database.db` after Z2M restart

### What did you expect to happen?

1. Device added successfully
2. Device can be controlled in UI before and after restart
3. Device is present in `database.db`

### How to reproduce it (minimal and precise)

1. Add any zigbee device. I'm using Tuya TS011F
2. Make sure that device can be controlled in UI
3. Check that device added in `database.db`
4. Restart Z2M (Tools -> Restart Zigbee2MQTT)
5. Check if device exist on "Devices" tab and can be controlled

### Zigbee2MQTT version

2.0.0-dev commit: 7392557

### Adapter firmware version

0.0.0.1 (downloaded from [v2.6.0_ncp_zboss_max_mem_nrf52840_dongle.hex](https://github.com/kardia-as/nrf-zboss-ncp/blob/main/nRF52840_dongle/v2.6.0_ncp_zboss_max_mem_nrf52840_dongle.hex))

### Adapter

zboss

### Setup

Add-on on Home Assistant OS on Raspberry Pi

### Debug log

Log after reboot: [log.log](https://github.com/user-attachments/files/18464085/log.log)

Coordinator address: `0xf4ce3684e3132c9a`
Device address: `0xa4c1380637ba72a5`

Diadlo, 2025-01-19:
It looks like should be fixed with https://github.com/Koenkk/zigbee-herdsman/pull/1293

pxFIN, 2025-01-22:
@Diadlo I reckon your issue is very much related to this ?

https://github.com/Koenkk/zigbee2mqtt/issues/24705

Diadlo, 2025-01-23:
@pxFIN yep. Looks related. And now my issue is solved, so closing

closed 2025-01-23 by Diadlo