Koenkk/zigbee2mqtt #25882: SONOFF ZBMINI2 wall switch stops working after using frontend switch

takiainen, 2025-01-18:
### What happened?

Sonoff ZBMINI2 relay stops accepting physical rocker switch input after using the relay via zigbee2mqtt frontend.

### What did you expect to happen?

Normally with these types of relays you can use both physical wall switches and and "virtual" frontend switch without any issues. Once you press wall switch, the frontend switch changes accordingly and so on. Both methods work at the same time. 

Now, instead, if you use only wall switches (or even the test button on the ZBMINI2 relay) everything works and you also see the "virtual" switch on the frontend change accordingly. But once you click the virtual switch, the wall switch stops working until complete power off full reset. 

I have updated the device firmware.
I am running latest zigbee2mqtt in docker container. Everything else works fine. 

### Zigbee2MQTT version

2.0.0

### Adapter firmware version

20240710

### Adapter

Sonoff Dongle P 

### Setup

Intel NUC, Linux Mint 22, Z2M running in a container

### Debug log

z2m: Failed to configure '0x70c59cfffe492c7e', attempt 2 (Error: ZCL command 0x70c59cfffe492c7e/1 customClusterEwelink.read(["radioPower",1,20,21,22,23], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"reservedBits":0,"writeUndiv":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE') at Endpoint.checkStatus (/app/node_modules/.pnpm/zigbee-herdsman@3.2.1/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:349:28) at Endpoint.zclCommand (/app/node_modules/.pnpm/zigbee-herdsman@3.2.1/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:954:26) at Endpoint.read (/app/node_modules/.pnpm/zigbee-herdsman@3.2.1/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:445:29) at configure (/app/node_modules/.pnpm/zigbee-herdsman-converters@21.12.0/node_modules/src/devices/sonoff.ts:1394:13) at Object.configure (/app/node_modules/.pnpm/zigbee-herdsman-converters@21.12.0/node_modules/src/index.ts:257:21) at Configure.configure (/app/lib/extension/configure.ts:124:13) at EventEmitter.wrappedCallback (/app/lib/eventBus.ts:204:17))

centeljon, 2025-01-18:
to clarify
i know there are ZBMINI, ZBMINIR2 and ZBMINIL2 sonoff devices
you talk about ZBMINI2 ?

takiainen, 2025-01-18:
ZBMINIR2

takiainen, 2025-01-18:
It seems that there's no real issue after all: If I click switch between tabs (i.e. About, Exposes, Bind, Reporting etc.) after using physical wall switch and then return to "Exposes" section, the virtual switch position has updated after all to match the position of the wall switch. 

And once trying this via Home Assistant everything works perfectly fine. 

So just a minor glitch on the frontend side, but no real world effect on the functioning of the relay, it seems.  I guess this issue can be closed.

closed 2025-01-19 by Koenkk