Koenkk/zigbee2mqtt #25883: [Feature request]: Use siren für HEIMAN HS1SA / SmokeSensor-EF-3.0

skyfullofdust, 2025-01-18:
### Is your feature request related to a problem? Please describe

I recently switched to Z2M after using ZHA, unfortunately I lost the ability to use (toggle on/off) the siren of the device.

### Describe the solution you'd like

Expose the siren to Zigbee2MQTT/Home Assistant as well.

### Describe alternatives you've considered

Switching back to ZHA, which doesn't seem useful for me.

### Additional context

Found the implementation used in ZHA: https://github.com/zigpy/zha-device-handlers/blob/dev/zhaquirks/heiman/smoke.py

bobo-cher, 2025-01-19:
That's interesting. I was intrigued and configured ZHA for a test. In ZHA there is switch "Siren" and many additional options for siren sound:

[image]

None of the configuration options worked in my test, at least not when doing selftest via button on the smoke detector. I did not test with real smoke.

The switch "Siren" shows correct state during selftest. However attempts to manually activate siren using this switch do nothing.

In Z2M we have property "Test", which shows correct status during selftest. This property cannot be set by user, it's readonly.

[image]

For me it looks like the configuration of this smoke detector in ZHA is wrong. Maybe my test was not good enough and I missed something. Were you able to manually activate siren from Home Assistant in ZHA?

skyfullofdust, 2025-01-19:
Many thanks for looking into it!

The configuration options didn't seem to work for me as well. Like I could set something, but it didn't seem to have an effect on anything either.

When I tested the alarm the siren triggered in HA, as you could reproduce. I could also manually turn it off (which is basically what I wanna achieve). I also managed to trigger/active the siren once, but I think it was immediately after a self-test when the device was still in active communication with the ZigBee coordinator.
Can you reproduce/confirm those two behaviors?

bobo-cher, 2025-01-19:
Being able to mute siren from HA is an important feature. I'm not sure it can be properly tested with a selftest. The selftest is rather short for this. On my other smoke detector Aqara it can take 1-3 seconds after clicking on Mute in HA until it is really silenced.

During my test in ZHA I've managed to screw up my network. After returning to Z2M most devices were offline and needed repairing. I'm sure I did something wrong when I configured adapter in ZHA but at the moment I hesitate to do another test in ZHA 🤣

bobo-cher, 2025-01-19:
I did use a second adapter and the first (main) was disconnected from PC. ZHA was too smart and offered to migrate to the new adapter or create a new network. I'm not sure I chose the correct answer 😉.

I have specifically chosen a different Zigbee channel. As I later found out that was what screws up the network. All devices switched to that channel. I don't know why they did this, they were not listed in ZHA even. The devices were offline when I returned to Z2M. I repaired most devices but there are few behind switches in wall boxes which require opening of the wall switches and clicking on the pair button. Before doing this I remembered that I used other channel during ZHA test and changed to that channel in Z2M. That helped and devices became online again. That saved me a lot of work for repairing all the shutters, which would also require their recalibration. 🤯

skyfullofdust, 2025-01-19:
Let me know when you decided to re-try again with ZHA and could succeed with reproducing. I might try to do a temporary setup to confirm it as well.
Do you have any experiences with Log-analyzing and reverse-engineering interfaces so we can have at least the mute-feature in Z2M as well?

bobo-cher, 2025-01-19:
I repeated the test in ZHA. When using the button on device to do selftest the property "Siren" in ZHA becomes active (property "Smoke" is active too). Trying to mute the siren from ZHA didn't work for me (tested multiple times). Activating the "Siren" property just after the test completes indeed activate the test again. It doesn't make the siren to sound continuously, it adds just a few more beeps like during selftest.

I did go so far and made real tests with smoke using candle. During real smoke alarm the property "Smoke" was active and property "Siren" was NOT. It was not possible to mute alarm from HA. Only a physical button on device silenced the siren.

The same tests made in Z2M. During selftest properties "Smoke" and „Test“ are both active (they are read only). With a real smoke alarm property "Smoke" is active and property "Test" is not.

Based on these tests in my opinion Z2M works correctly. In ZHA property "Test" is shown incorrectly as "Siren". It is also changeable (not readonly) in ZHA but the changes have no effect. In addition ZHA shows many options related to siren but none of them works on this device.

skyfullofdust, 2025-01-19:
Thanks for your extended tests, so that means that ZHA just exposes the test-property as a siren and can be temporarily used to repeat the test from HA? Which other model do you have that actually allows muting them?

What I'm wondering, how did you manage to set it off with a candle? I had no luck in my tests a couple weeks ago, even not without hot water steam.

bobo-cher, 2025-01-19:
The other one is Aqara smoke detector (just search for "Aqara smoke detector", there is only one model). I've bought three of them and three Heiman a few weeks ago to test. I have installed all of them literally yesterday but now I'm having second thoughts. Not being able to mute remotely is an issue, which I somehow missed completely. I was previously concerned mainly about ability to turn on siren remotely (I want all sirens to turn on if one detector detects smoke). My thinking was that having remotely activated detectors in bed rooms would be enough and I can use cheaper Heimans in other rooms. Now I'm thinking about returning Heimans and buying additional Aqaras.

Aqaras are not perfect as well. During my tests with the first detector the battery level dropped to 75%. I don't think they last anywhere near 10 years as promised. The batteries are expensive as well. Still they are the cheapest Zigbee smoke detectors with remotely activated alarm. I've setup automations to use their alarms for water leaks too and that worked pretty well in my tests.

As for smoke test with candle, I just put out the candle and that produced a required amount of smoke (there is not much smoke when the candle is burning). Hold the detector above smoke for 30-60 seconds.

bobo-cher, 2025-01-20:
Found this issue confirming the siren cannot be muted remotely on this device: https://github.com/Koenkk/zigbee-herdsman-converters/issues/478.