Koenkk/zigbee2mqtt #25885: Installcode no more working with 2.0.0 for Bosch Twinguard

passiber, 2025-01-18:
### What happened?

I detected after updating to version 2.0.0, the install code seems not longer working for Bosch Twinguard.

I have several Bosch Twinguards in my Zigbee Network. It was possible to connect them with the old version 1.42 without problem. Some days ago I updated to version 2.0.0. Today I wanted to connect a new Bosch Twinguard,. But it's not possible. I also tried to reconnect another Bosch Twinguard, which was already in my network before. Also here I can't get a connection with the version 2.0.0.
So I think there is something wrong with the install code "adjustment" which was included with the install code fix?

I get following messages in the log:

```
2025-01-18 18:56:50] info: 	zh:controller: Install code was adjusted for reason 'missing CRC'.
[2025-01-18 18:56:50] info: 	zh:controller: Adding install code for 0x000D6F00191051B7.
[2025-01-18 18:56:50] info: 	z2m: Successfully added new install code
[2025-01-18 18:58:11] error: 	zh:controller: Interview failed for '0x000d6f00191051b7 with error 'Error: Interview failed because can not get node descriptor ('0x000d6f00191051b7')'
```

So it seems, that the Install code will be "adjusted", but it will be not successful to connect the device.
The Bosch Twinguard have a 91-digit install code. 

### What did you expect to happen?

It should work like already in the previous versions

### How to reproduce it (minimal and precise)

Everytime when I try to connect a Bosch Twinguard with 91-digit install code at the frontend with "tools->add install code"

### Zigbee2MQTT version

2.0.0

### Adapter firmware version

20210708

### Adapter

zstack

### Setup

Add-On, Pi

JohannesBY, 2025-01-26:
Same for me:
I want to set up a completely new system with raspi 4, the SONOFF ZBDongle-P and 5 Bosch Twinguards -> Interview failed

Home Assistant Core 2025.1.4
SONOFF ZBDongle-P 20240710
Zigbee2MQTT 2.0.0

raoulbhatia, 2025-02-01:
After unsuccessfully trying to pair my Twinguard using a Conbee III (see Koenkk/zigbee-herdsman#1184)
I purchased a ZBDongle-E and am also struggling right now with the exact same issue.

* Itead Sonoff Zigbee Dongle Plus V2 / ZBDongle-E
* Coordinator revision: I tried 7.4.4 [GA] and 7.4.5 [GA]
* Coordinator type: EmberZNet
* Zigbee2MQTT version: [2.0.0](https://github.com/Koenkk/zigbee2mqtt/releases/tag/2.0.0) - installed as a HAOS Addon

raoulbhatia, 2025-02-01:
FYI I've
1. downgraded to [1.42.0](https://github.com/Koenkk/zigbee2mqtt/releases/tag/1.42.0)
2. wiped all data
3. tried to pair my Twinguard
4. **BUT still get the same error** (logs below)

I successfully paired a Bosch BSD-2/Smoke alarm II under 1.42.0

```
[2025-02-01 21:45:09] error: 	zh:controller: Interview failed for '0x000d6f0019933538 with error 'Error: Interview failed because can not get node descriptor ('0x000d6f0019933538')'
[2025-02-01 21:45:09] error: 	z2m: Failed to interview '0x000d6f0019933538', device has not successfully been paired
[2025-02-01 21:45:09] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x000d6f0019933538","ieee_address":"0x000d6f0019933538","status":"failed"},"type":"device_interview"}'
[2025-02-01 21:45:37] info: 	zh:controller: Interview for '0x000d6f0019933538' started
[2025-02-01 21:45:37] info: 	z2m: Starting interview of '0x000d6f0019933538'
[2025-02-01 21:45:37] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x000d6f0019933538","ieee_address":"0x000d6f0019933538","status":"started"},"type":"device_interview"}'
[2025-02-01 21:46:06] error: 	zh:controller: Interview failed for '0x000d6f0019933538 with error 'Error: Interview failed because can not get node descriptor ('0x000d6f0019933538')'
[2025-02-01 21:46:06] error: 	z2m: Failed to interview '0x000d6f0019933538', device has not successfully been paired
[2025-02-01 21:46:06] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x000d6f0019933538","ieee_address":"0x000d6f0019933538","status":"failed"},"type":"device_interview"}'
[2025-02-01 21:46:33] info: 	zh:controller: Interview for '0x000d6f0019933538' started
[2025-02-01 21:46:33] info: 	z2m: Starting interview of '0x000d6f0019933538'
[2025-02-01 21:46:33] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x000d6f0019933538","ieee_address":"0x000d6f0019933538","status":"started"},"type":"device_interview"}'
```

kooori, 2025-02-02:
Same for me. I have spent so long troubleshooting this. So happy I came across this thread.

- I have 3x Bosch Twinguard Device ID: 8750001215 and the install code is 91 characters long.
- using a: SLZB-06M Coordinator (checked on V2.8.0.dev, 2.7.1, 2.3.6)

[image]

[image]

passiber, 2025-02-02:
I downgraded to version 1.42.0 again. So I was able to connect my Bosch Twinguards again. 
After all Twinguards are connected they are also working fine with version 2.0.0. In my case the problem is, that I need to reconnect some Twinguards after 2-3 months, because they sometimes loose the connection. And this was not possible with version 2.0.0. So for me it makes no sense to use the version 2.0.0 with this kind of bug.

@raoulbhatia I think you propably have another problem. Did you read this thread [https://github.com/Koenkk/zigbee2mqtt/issues/20169](url)? I had the same problem like you at the beginning. I was not able to connect the Bosch Twinguards with ZBDongle-E. Then I switched to ZBDongle-**P**, where I successfully could connect the Bosch Twinguards until zigbee version 2.0.0.

nerdiges, 2025-02-02:
Same for me. I have also spent some time troubleshooting.

-  I have a Bosch Twinguard Device ID 8750001213 and  install code is 91 characters long.
- Z2M is Version 2.1.0-1
- Coordinator is SLZB-06 with zigbee fw 2.7.1  

FYI: Install code for Bosch Thermostat II worked flawless

runningman84, 2025-02-04:
> Same for me. I have also spent some time troubleshooting.
> 
> * I have a Bosch Twinguard Device ID 8750001213 and  install code is 91 characters long.
> * Z2M is Version 2.1.0-1
> * Coordinator is SLZB-06 with zigbee fw 2.7.1
> 
> FYI: Install code for Bosch Thermostat II worked flawless

I can confirm both observations with my setup, Thermostat works with install code, Twinguard does not work and needs inital pairing with 1.42. Once it is paired it works just fine with 2.0.0.

kooori, 2025-02-04:
> I can confirm both observations with my setup, Thermostat works with install code, Twinguard does not work and needs inital pairing with 1.42. Once it is paired it works just fine with 2.0.0.

Do you mean running zigbee2mqtt as version 1.42 and then it works? I'm abit of a newbie, how would I do that.

w0nk1, 2025-02-05:
Any news regarding this topic? I have exactly the same issue. Worked fine with 1.42 but with 2.1.0-1 it is not working anymore. How can I downgrade without a backup?

JohannesBY, 2025-02-05:
> Any news regarding this topic? I have exactly the same issue. Worked fine with 1.42 but with 2.1.0-1 it is not working anymore. How can I downgrade without a backup?

I found the solution here: https://community.home-assistant.io/t/how-to-downgrade-addons/331223/20

1. Download the desired HA Addon release: https://github.com/zigbee2mqtt/hassio-zigbee2mqtt
2. Save the unpacked source to the .\addons folder of your HA instance
3. In your add-on store shows up "local add-ons"

runningman84, 2025-02-05:
@Koenkk would you debug this issue if somebody would donate you a device?

Koenkk, 2025-02-05:
Could you provide the debug log when adding the install code on both 1.42 (works) and on 2.0.0 (doesn't work)?

See [this](https://www.zigbee2mqtt.io/guide/configuration/logging.html) on how to enable debug logging.

kooori, 2025-02-06:
> Could you provide the debug log when adding the install code on both 1.42 (works) and on 2.0.0 (doesn't work)?
> 
> See [this](https://www.zigbee2mqtt.io/guide/configuration/logging.html) on how to enable debug logging.

Thanks @JohannesBY for posting that.

@Koenkk I tried running it on both versions and it wont add. Although on version 1.4.1 and 1.4.2 (I tested both) I do NOT get the missing CRC error. Logs below.

Although my twinguard does not run as I get the node descriptor does not exist.

I have the bosch twinguard model 8750001215

--v2.1.0-1 msg
`
[2025-02-06 20:29:21] debug: 	z2m:mqtt: Received MQTT message on 'zigbee2mqtt/bridge/request/install_code/add' with data '{"transaction":"8jnch-1","value":"RB01SG0D836591B3CC0010000000000000000000000D6F00188EF870DLKCB602F27AD11661D5DA8F9E84B500795"}'
[2025-02-06 20:29:21] info: 	zh:controller: Install code was adjusted for reason 'missing CRC'.
[2025-02-06 20:29:21] info: 	zh:controller: Adding install code for 0x000D6F00188EF870.
[2025-02-06 20:29:21] debug: 	zh:ember:ezsp: ===> [FRAME: ID=273:"IMPORT_TRANSIENT_KEY" Seq=73 Len=30]
[2025-02-06 20:29:21] debug: 	zh:ember:uart:ash: ---> [FRAME type=DATA frmTx=1 frmRx=0](ackRx=1)
[2025-02-06 20:29:21] debug: 	zh:ember:uart:ash: <--- [FRAME type=DATA]
[2025-02-06 20:29:21] debug: 	zh:ember:uart:ash: <--- [FRAME type=DATA ackNum=2](ackRx=1 frmTx=2)
[2025-02-06 20:29:21] debug: 	zh:ember:uart:ash: <--- [FRAME type=DATA ackNum=2 frmNum=0](frmRx=0) Added to rxQueue
[2025-02-06 20:29:21] debug: 	zh:ember:uart:ash: ---> [FRAME type=ACK frmRx=1](ackRx=2)
[2025-02-06 20:29:21] debug: 	zh:ember:ezsp: <=== [FRAME: ID=273:"IMPORT_TRANSIENT_KEY" Seq=73 Len=9]
[2025-02-06 20:29:21] debug: 	zh:ember: [ADD INSTALL CODE] Success for '0x000D6F00188EF870'.
[2025-02-06 20:29:21] info: 	z2m: Successfully added new install code
[2025-02-06 20:29:21] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/response/install_code/add', payload '{"data":{"value":"RB01SG0D836591B3CC0010000000000000000000000D6F00188EF870DLKCB602F27AD11661D5DA8F9E84B500795"},"status":"ok","transaction":"8jnch-1"}'
[2025-02-06 20:29:21] debug: 	z2m:mqtt: Received MQTT message on 'zigbee2mqtt/bridge/request/permit_join' with data '{"device":null,"time":254,"transaction":"8jnch-2"}'
[2025-02-06 20:29:21] info: 	z2m: Zigbee: allowing new devices to join.
[2025-02-06 20:29:21] debug: 	zh:ember:ezsp: ===> [FRAME: ID=273:"IMPORT_TRANSIENT_KEY" Seq=74 Len=30]
[2025-02-06 20:29:21] debug: 	zh:ember:uart:ash: ---> [FRAME type=DATA frmTx=2 frmRx=1](ackRx=2)
[2025-02-06 20:29:21] debug: 	zh:ember:uart:ash: <--- [FRAME type=DATA]
[2025-02-06 20:29:21] debug: 	zh:ember:uart:ash: <--- [FRAME type=DATA ackNum=3](ackRx=2 frmTx=3)
[2025-02-06 20:29:21] debug: 	zh:ember:uart:ash: <--- [FRAME type=DATA ackNum=3 frmNum=1](frmRx=1) Added to rxQueue
[2025-02-06 20:29:21] debug: 	zh:ember:uart:ash: ---> [FRAME type=ACK frmRx=2](ackRx=3)
[2025-02-06 20:29:21] debug: 	zh:ember:ezsp: <=== [FRAME: ID=273:"IMPORT_TRANSIENT_KEY" Seq=74 Len=9]
[2025-02-06 20:29:21] debug: 	zh:ember:ezsp: ===> [FRAME: ID=85:"SET_POLICY" Seq=75 Len=7]
[2025-02-06 20:29:21] debug: 	zh:ember:uart:ash: ---> [FRAME type=DATA frmTx=3 frmRx=2](ackRx=3)
[2025-02-06 20:29:21] debug: 	zh:ember:uart:ash: <--- [FRAME type=DATA]
[2025-02-06 20:29:21] debug: 	zh:ember:uart:ash: <--- [FRAME type=DATA ackNum=4](ackRx=3 frmTx=4)
[2025-02-06 20:29:21] debug: 	zh:ember:uart:ash: <--- [FRAME type=DATA ackNum=4 frmNum=2](frmRx=2) Added to rxQueue
[2025-02-06 20:29:21] debug: 	zh:ember:uart:ash: ---> [FRAME type=ACK frmRx=3](ackRx=4)
[2025-02-06 20:29:21] debug: 	zh:ember:ezsp: <=== [FRAME: ID=85:"SET_POLICY" Seq=75 Len=6]
[2025-02-06 20:29:21] debug: 	zh:ember: [EzspPolicyId] SET 'TRUST_CENTER_POLICY' TO '3' with status=OK.
[2025-02-06 20:29:21] debug: 	zh:ember:ezsp: ===> [FRAME: ID=34:"PERMIT_JOINING" Seq=76 Len=6]
[2025-02-06 20:29:21] debug: 	zh:ember:uart:ash: ---> [FRAME type=DATA frmTx=4 frmRx=3](ackRx=4)
[2025-02-06 20:29:21] debug: 	zh:ember:uart:ash: <--- [FRAME type=DATA]
[2025-02-06 20:29:21] debug: 	zh:ember:uart:ash: <--- [FRAME type=DATA ackNum=5](ackRx=4 frmTx=5)
[2025-02-06 20:29:21] debug: 	zh:ember:uart:ash: <--- [FRAME type=DATA ackNum=5 frmNum=3](frmRx=3) Added to rxQueue
[2025-02-06 20:29:21] debug: 	zh:ember:uart:ash: ---> [FRAME type=ACK frmRx=4](ackRx=5)
[2025-02-06 20:29:21] debug: 	zh:ember:ezsp: <=== [FRAME: ID=34:"PERMIT_JOINING" Seq=76 Len=6]
[2025-02-06 20:29:21] debug: 	zh:ember: Permit joining on coordinator for 254 sec.
[2025-02-06 20:29:21] debug: 	zh:ember: ~~~> [ZDO PERMIT_JOINING_REQUEST BROADCAST to=65532 messageTag=7 payload=07fe01]
[2025-02-06 20:29:21] debug: 	zh:ember:ezsp: ===> [FRAME: ID=54:"SEND_BROADCAST" Seq=77 Len=24]
[2025-02-06 20:29:21] debug: 	zh:ember:uart:ash: ---> [FRAME type=DATA frmTx=5 frmRx=4](ackRx=5)
[2025-02-06 20:29:21] debug: 	zh:ember:uart:ash: <--- [FRAME type=DATA]
[2025-02-06 20:29:21] debug: 	zh:ember:uart:ash: <--- [FRAME type=DATA ackNum=5](ackRx=5 frmTx=6)
[2025-02-06 20:29:21] debug: 	zh:ember:uart:ash: <--- [FRAME type=DATA ackNum=5 frmNum=4](frmRx=4) Added to rxQueue
[2025-02-06 20:29:21] debug: 	zh:ember:uart:ash: ---> [FRAME type=ACK frmRx=5](ackRx=5)
[2025-02-06 20:29:21] debug: 	zh:ember:ezsp: <=== [CBFRAME: ID=25:"STACK_STATUS_HANDLER" Seq=76 Len=6]
[2025-02-06 20:29:21] debug: 	zh:ember:ezsp: ezspStackStatusHandler(): callback called with: [status=ZIGBEE_NETWORK_OPENED]
[... cut ...]
[2025-02-06 21:18:05] debug: 	zh:controller:device: Interview - failed for device '0x000d6f00189b9d8f' with error 'Error: Interview failed because can not get node descriptor ('0x000d6f00189b9d8f')'
[2025-02-06 21:18:05] debug: 	zh:controller:database: Writing database to '/config/zigbee2mqtt/database.db'
[2025-02-06 21:18:05] error: 	zh:controller: Interview failed for '0x000d6f00189b9d8f with error 'Error: Interview failed because can not get node descriptor ('0x000d6f00189b9d8f')'
[2025-02-06 21:18:05] error: 	z2m: Failed to interview '0x000d6f00189b9d8f', device has not successfully been paired`

jeremyghys, 2025-02-07:
Hi +1

Exactly the same problem since >=v2.0

w0nk1, 2025-02-07:
@kooori It will only work in version 1.4.1 if you use the ZBDongle-P. Unfortunately, it does not work with the ZBDongle-E.

w0nk1, 2025-02-07:
> thanks for coming back to me. It's the first time I am going Zigbee. Would this TubesZB device work (https://tubeszb.com/product/cc2652p2-based-zigbee-to-poe-coordinator-2023/) with the Twinguard(I need something POE based)? If so I will go ahead and get this and keep the other one for down the track.

I can't say for sure, but the chipset looks very similar, so chances are good, I guess. Maybe someone else can provide more insights

> I'm paranoid that the model I have is different to the supported model on the z2m website. The last digit is different. 8750001215 is the model I just received from Amazon UK (8750001213 is supported on the website).

[image]

This is one of the Twinguards I paired yesterday with version 1.4.1 and the ZBDongle-P. No problems!

kooori, 2025-02-07:
@w0nk1 I just ordered the SLZB-06 adapter (as it has POE). It has the same CC2652P chipset so theoretically should work.

I'll report back here on how I go with the new device and fingers crossed it connects otherwise I might have to buy the bosch smart home controller.

runningman84, 2025-02-07:
In 2.1.0 I get this error

```
/data/log $ grep -r 0x000d6f0019b76ec1 .
./2025-02-05.21-15-14/log.log:[2025-02-07 21:09:22] debug:      zh:controller: Device '0x000d6f0019b76ec1' joined
./2025-02-05.21-15-14/log.log:[2025-02-07 21:09:22] debug:      zh:controller: Device '0x000d6f0019b76ec1' accepted by handler
./2025-02-05.21-15-14/log.log:[2025-02-07 21:09:22] debug:      zh:controller: New device '0x000d6f0019b76ec1' joined
./2025-02-05.21-15-14/log.log:[2025-02-07 21:09:22] debug:      zh:controller: Creating device '0x000d6f0019b76ec1'
./2025-02-05.21-15-14/log.log:[2025-02-07 21:09:22] info:       z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/0x000d6f0019b76ec1/availability', payload '{"state":"online"}'
./2025-02-05.21-15-14/log.log:[2025-02-07 21:09:22] info:       z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/0x000d6f0019b76ec1', payload '{"last_seen":"2025-02-07T20:09:22.984Z"}'
./2025-02-05.21-15-14/log.log:[2025-02-07 21:09:22] info:       zh:controller: Interview for '0x000d6f0019b76ec1' started
./2025-02-05.21-15-14/log.log:[2025-02-07 21:09:22] debug:      zh:controller:device: Interview - start device '0x000d6f0019b76ec1'
./2025-02-05.21-15-14/log.log:[2025-02-07 21:09:22] info:       z2m: Device '0x000d6f0019b76ec1' joined
./2025-02-05.21-15-14/log.log:[2025-02-07 21:09:23] info:       z2m: Starting interview of '0x000d6f0019b76ec1'
./2025-02-05.21-15-14/log.log:[2025-02-07 21:09:23] info:       z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x000d6f0019b76ec1","ieee_address":"0x000d6f0019b76ec1"},"type":"device_joined"}'
./2025-02-05.21-15-14/log.log:[2025-02-07 21:09:23] info:       z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x000d6f0019b76ec1","ieee_address":"0x000d6f0019b76ec1","status":"started"},"type":"device_interview"}'
./2025-02-05.21-15-14/log.log:[2025-02-07 21:09:29] debug:      zh:controller: Device '0x000d6f0019b76ec1' joined
./2025-02-05.21-15-14/log.log:[2025-02-07 21:09:29] debug:      zh:controller: Device '0x000d6f0019b76ec1' accepted by handler
./2025-02-05.21-15-14/log.log:[2025-02-07 21:09:29] debug:      zh:controller: Device '0x000d6f0019b76ec1' is already in database with different network address, updating network address
./2025-02-05.21-15-14/log.log:[2025-02-07 21:09:29] info:       z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/0x000d6f0019b76ec1', payload '{"last_seen":"2025-02-07T20:09:29.630Z"}'
./2025-02-05.21-15-14/log.log:[2025-02-07 21:09:29] debug:      zh:controller: Not interviewing '0x000d6f0019b76ec1', completed 'false', in progress 'true'
./2025-02-05.21-15-14/log.log:[2025-02-07 21:09:33] debug:      zh:controller:device: Interview - node descriptor request failed for '0x000d6f0019b76ec1', attempt 1
./2025-02-05.21-15-14/log.log:[2025-02-07 21:09:36] debug:      zh:controller: Device '0x000d6f0019b76ec1' joined
./2025-02-05.21-15-14/log.log:[2025-02-07 21:09:36] debug:      zh:controller: Device '0x000d6f0019b76ec1' accepted by handler
./2025-02-05.21-15-14/log.log:[2025-02-07 21:09:36] debug:      zh:controller: Device '0x000d6f0019b76ec1' is already in database with different network address, updating network address
./2025-02-05.21-15-14/log.log:[2025-02-07 21:09:36] info:       z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/0x000d6f0019b76ec1', payload '{"last_seen":"2025-02-07T20:09:36.261Z"}'
./2025-02-05.21-15-14/log.log:[2025-02-07 21:09:36] debug:      zh:controller: Not interviewing '0x000d6f0019b76ec1', completed 'false', in progress 'true'
./2025-02-05.21-15-14/log.log:[2025-02-07 21:09:43] debug:      zh:controller:device: Interview - node descriptor request failed for '0x000d6f0019b76ec1', attempt 2
./2025-02-05.21-15-14/log.log:[2025-02-07 21:09:49] debug:      zh:controller: Device '0x000d6f0019b76ec1' joined
./2025-02-05.21-15-14/log.log:[2025-02-07 21:09:49] debug:      zh:controller: Device '0x000d6f0019b76ec1' accepted by handler
./2025-02-05.21-15-14/log.log:[2025-02-07 21:09:49] debug:      zh:controller: Device '0x000d6f0019b76ec1' is already in database with different network address, updating network address
./2025-02-05.21-15-14/log.log:[2025-02-07 21:09:49] info:       z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/0x000d6f0019b76ec1', payload '{"last_seen":"2025-02-07T20:09:49.556Z"}'
./2025-02-05.21-15-14/log.log:[2025-02-07 21:09:49] debug:      zh:controller: Not interviewing '0x000d6f0019b76ec1', completed 'false', in progress 'true'
./2025-02-05.21-15-14/log.log:[2025-02-07 21:09:53] debug:      zh:controller:device: Interview - node descriptor request failed for '0x000d6f0019b76ec1', attempt 3
./2025-02-05.21-15-14/log.log:[2025-02-07 21:10:03] debug:      zh:controller:device: Interview - node descriptor request failed for '0x000d6f0019b76ec1', attempt 4
./2025-02-05.21-15-14/log.log:[2025-02-07 21:10:13] debug:      zh:controller:device: Interview - node descriptor request failed for '0x000d6f0019b76ec1', attempt 5
./2025-02-05.21-15-14/log.log:[2025-02-07 21:10:16] debug:      zh:controller: Device '0x000d6f0019b76ec1' joined
./2025-02-05.21-15-14/log.log:[2025-02-07 21:10:16] debug:      zh:controller: Device '0x000d6f0019b76ec1' accepted by handler
./2025-02-05.21-15-14/log.log:[2025-02-07 21:10:16] debug:      zh:controller: Device '0x000d6f0019b76ec1' is already in database with different network address, updating network address
[... cut ...]
./2025-02-05.21-15-14/log.log:[2025-02-07 21:11:29] error:      zh:controller: Interview failed for '0x000d6f0019b76ec1 with error 'Error: Interview failed because can not get node descriptor ('0x000d6f0019b76ec1')'
./2025-02-05.21-15-14/log.log:[2025-02-07 21:11:29] error:      z2m: Failed to interview '0x000d6f0019b76ec1', device has not successfully been paired
./2025-02-05.21-15-14/log.log:[2025-02-07 21:11:29] info:       z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x000d6f0019b76ec1","ieee_address":"0x000d6f0019b76ec1","status":"failed"},"type":"device_interview"}'

```

I am using zstack with a zigstar CC2652P controller.

runningman84, 2025-02-07:
For some reason I cannot downgrade to 1.42.0 anymore it crashes after a few seconds. So I cannot join the 2 remaining twinguards or provide any addtional logs at the moment from 1.42.0 :(

runningman84, 2025-02-07:
okay I got 1.42.0 running again this is the full log of a twingaud joining:
```
/data/log # grep -r 0x000d6f0019b76662 *
2025-02-07.21-39-19/log.log:[2025-02-07 21:57:41] debug:        zh:controller: Device '0x000d6f0019b76662' joined
2025-02-07.21-39-19/log.log:[2025-02-07 21:57:41] debug:        zh:controller: Device '0x000d6f0019b76662' accepted by handler
2025-02-07.21-39-19/log.log:[2025-02-07 21:57:41] debug:        zh:controller: New device '0x000d6f0019b76662' joined
2025-02-07.21-39-19/log.log:[2025-02-07 21:57:41] debug:        zh:controller: Creating device '0x000d6f0019b76662'
2025-02-07.21-39-19/log.log:[2025-02-07 21:57:41] info:         z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/0x000d6f0019b76662/availability', payload 'online'
2025-02-07.21-39-19/log.log:[2025-02-07 21:57:41] info:         z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/0x000d6f0019b76662', payload '{"last_seen":"2025-02-07T20:57:41.763Z"}'
2025-02-07.21-39-19/log.log:[2025-02-07 21:57:41] info:         zh:controller: Interview for '0x000d6f0019b76662' started
2025-02-07.21-39-19/log.log:[2025-02-07 21:57:41] debug:        zh:controller:device: Interview - start device '0x000d6f0019b76662'
2025-02-07.21-39-19/log.log:[2025-02-07 21:57:41] info:         z2m: Device '0x000d6f0019b76662' joined
2025-02-07.21-39-19/log.log:[2025-02-07 21:57:41] info:         z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":{"friendly_name":"0x000d6f0019b76662"},"type":"device_connected"}'
2025-02-07.21-39-19/log.log:[2025-02-07 21:57:41] info:         z2m: Starting interview of '0x000d6f0019b76662'
2025-02-07.21-39-19/log.log:[2025-02-07 21:57:42] info:         z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"interview_started","meta":{"friendly_name":"0x000d6f0019b76662"},"type":"pairing"}'
2025-02-07.21-39-19/log.log:[2025-02-07 21:57:42] debug:        zh:controller: Received ZDO response: clusterId=END_DEVICE_ANNOUNCE, status=SUCCESS, payload={"nwkAddress":4471,"eui64":"0x000d6f0019b76662","capabilities":{"alternatePANCoordinator":0,"deviceType":0,"powerSource":0,"rxOnWhenIdle":0,"reserved1":0,"reserved2":0,"securityCapability":0,"allocateAddress":1}}
2025-02-07.21-39-19/log.log:[2025-02-07 21:57:42] debug:        zh:controller: Device announce from '0x000d6f0019b76662:4471'
2025-02-07.21-39-19/log.log:[2025-02-07 21:57:42] info:         z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/0x000d6f0019b76662', payload '{"last_seen":"2025-02-07T20:57:42.162Z"}'
2025-02-07.21-39-19/log.log:[2025-02-07 21:57:42] debug:        z2m: Device '0x000d6f0019b76662' announced itself
2025-02-07.21-39-19/log.log:[2025-02-07 21:57:42] info:         z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"announce","meta":{"friendly_name":"0x000d6f0019b76662"},"type":"device_announced"}'
2025-02-07.21-39-19/log.log:[2025-02-07 21:57:42] info:         z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x000d6f0019b76662","ieee_address":"0x000d6f0019b76662"},"type":"device_joined"}'
2025-02-07.21-39-19/log.log:[2025-02-07 21:57:42] info:         z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x000d6f0019b76662","ieee_address":"0x000d6f0019b76662","status":"started"},"type":"device_interview"}'
2025-02-07.21-39-19/log.log:[2025-02-07 21:57:42] info:         z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x000d6f0019b76662","ieee_address":"0x000d6f0019b76662"},"type":"device_announce"}'
2025-02-07.21-39-19/log.log:[2025-02-07 21:57:42] debug:        zh:controller:device: Interview - got node descriptor for device '0x000d6f0019b76662'
2025-02-07.21-39-19/log.log:[2025-02-07 21:57:42] info:         zh:controller:device: Device '0x000d6f0019b76662' is only compliant to revision 'pre-21' of the ZigBee specification (current revision: 23).
2025-02-07.21-39-19/log.log:[2025-02-07 21:57:42] debug:        zh:controller:device: Interview - got active endpoints for device '0x000d6f0019b76662'
2025-02-07.21-39-19/log.log:[2025-02-07 21:57:43] debug:        zh:controller:device: Interview - got simple descriptor for endpoint '1' device '0x000d6f0019b76662'
2025-02-07.21-39-19/log.log:[2025-02-07 21:57:43] debug:        zh:controller:endpoint: ZCL command 0x000d6f0019b76662/1 genBasic.read(["modelId"], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"reservedBits":0,"writeUndiv":false,"sendPolicy":"immediate"})
2025-02-07.21-39-19/log.log:[2025-02-07 21:57:43] debug:        zh:zstack: sendZclFrameToEndpointInternal 0x000d6f0019b76662:4471/1 (0,0,1)
2025-02-07.21-39-19/log.log:[2025-02-07 21:57:43] debug:        zh:controller:endpoint: ZCL command 0x000d6f0019b76662/1 genBasic.defaultRsp({"cmdId":1,"statusCode":0}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":1,"reservedBits":0,"transactionSequenceNumber":126,"writeUndiv":false})
2025-02-07.21-39-19/log.log:[2025-02-07 21:57:43] debug:        z2m: Received Zigbee message from '0x000d6f0019b76662', type 'readResponse', cluster 'genBasic', data '{"modelId":"Champion"}' from endpoint 1 with groupID 0
2025-02-07.21-39-19/log.log:[2025-02-07 21:57:43] info:         z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/0x000d6f0019b76662', payload '{"last_seen":"2025-02-07T20:57:43.949Z","linkquality":123}'
[... cut ...] mqtt_bridge_0x00124b001caaaa48"},"device_class":"timestamp","enabled_by_default":false,"entity_category":"diagnostic","icon":"mdi:clock","json_attributes_topic":"zigbee2mqtt/kids_room/alice/smoke","name":"Last seen","object_id":"kids_room/alice/smoke_last_seen","origin":{"name":"Zigbee2MQTT","sw":"1.42.0","url":"https://www.zigbee2mqtt.io"},"state_topic":"zigbee2mqtt/kids_room/alice/smoke","unique_id":"0x000d6f0019b76662_last_seen_zigbee2mqtt","value_template":"{{ value_json.last_seen }}"}'

```

JohannesBY, 2025-02-08:
@Koenkk I did a little investigation: Maybe the CRC is the problem

Since version 2.0.0 a missing crc will be calculated and the install code will be adjusted. Till version 1.42.0 this doesn't happen and the 'installCodeFormat' is '1' instead of '2'

Logfiles working and failing:

[45df7312_zigbee2mqtt_2025-02-08T10-11-09.807Z.log](https://github.com/user-attachments/files/18718078/45df7312_zigbee2mqtt_2025-02-08T10-11-09.807Z.log)
[local_zigbee2mqtt_2025-02-08T10-25-27.217Z.log](https://github.com/user-attachments/files/18718077/local_zigbee2mqtt_2025-02-08T10-25-27.217Z.log)
[local_zigbee2mqtt_2025-02-08T10-30-11.284Z.log](https://github.com/user-attachments/files/18718076/local_zigbee2mqtt_2025-02-08T10-30-11.284Z.log)

Koenkk, 2025-02-09:
I'm not sure how you are running z2m, but could you try making the following change and see if that works?

```bash
docker exec -it CONTAINER_NAME sh
apk add nano
nano /app/node_modules/zigbee-herdsman/dist/adapter/z-stack/adapter/zStackAdapter.js
# Change: const payload = { installCodeFormat: key.length === 18 ? 1 : 2, ieeeaddr: ieeeAddress, installCode: key };
# To: const payload = { installCodeFormat: 2, ieeeaddr: ieeeAddress, installCode: key };
exit
docker restart CONTAINER_NAME
```

nerdiges, 2025-02-09:
> I'm paranoid that the model I have is different to the supported model on the z2m website. The last digit is different. 8750001215 is the model I just received from Amazon UK (8750001213 is supported on the website).

@kooori I was able to pair three 8750001213 twinguards with Z2M < 1.4.2. Thus, I assume that it will work once it is fixed for version 2.x

nerdiges, 2025-02-09:
> I'm not sure how you are running z2m, but could you try making the following change and see if that works?
> 
> docker exec -it CONTAINER_NAME sh
> apk add nano
> nano /app/node_modules/zigbee-herdsman/dist/adapter/z-stack/adapter/zStackAdapter.js
> # Change: const payload = { installCodeFormat: key.length === 18 ? 1 : 2, ieeeaddr: ieeeAddress, installCode: key };
> # To: const payload = { installCodeFormat: 2, ieeeaddr: ieeeAddress, installCode: key };
> exit
> docker restart CONTAINER_NAME

@Koenkk I was able to join a Twinguard 8750001213 (91 character install code) with these changes in Z2M 2.1.1-1 (Coordinator is SLZB-06 with zigbee fw 2.7.1)

Thank you very much for the quick support!

Koenkk, 2025-02-10:
@nerdiges should be fixed by https://github.com/Koenkk/zigbee-herdsman/pull/1317, could you try again without this change on the latest dev branch?

Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now and in the next release which is every 1st of the month.

closed 2025-02-10 by Koenkk

kooori, 2025-02-11:
I can confirm this fix works on the 8750001215 model aswell with the SLZB-06M.

Thank you to everyone that helped! 

(Now  to return the SLZB-06 that I just ordered...  oooops)

RN-Say, 2025-02-12:
Another successful Dongle E pairing!!!! I've got 4 of these Twinguards and I've been right on the edge of starting over with an SLZB-06 and re-pairing my 100 Zigbee devices. I can't tell you how happy I am that I waited!!

Maybe your SLZB-06 can still be used as a Zigbee router or a Thread border router instead of returning it.

raoulbhatia, 2025-02-12:
As a follow up, I am happy
to confirm a successful pairing
of a Bosch Twinguard
via Sonoff ZigBee Dongle E
running z2m commit 30d13d6

kooori, 2025-07-03:
Both the SLZB-06 and SLZB-06M worked for me.

S474N, 2025-07-17:
I have also SLZB-06 and problem persists (I have 9 pcs of BOSCH, 8 works and 1 no).

Muhviehstar, 2025-09-03:
> Do I understand correctly that for successful pairing I need to buy a Sonof ZigBee dongle E?
> 
> I currently have a SLZB-MR1 and unfortunately I can't pair it successfully.
> 
> [image]

I have the same issue, do you have a solution yet?

STaegtmeier, 2025-10-23:
Same issue with the "missing CRC" with the CZC coordinator from cod.m (CC2652P7, firmware 20250321, Zigbee2mqtt 2.6.2 in kubernetes).

```
[23/10/2025, 10:00:27] frontend:api:bridge: Sending {"topic":"bridge/request/install_code/add","payload":{"value":"THISISMYUNIQUECODE","transaction":"r2rc4-3"}}
[23/10/2025, 10:00:27] zh:controller: Install code was adjusted for reason 'missing CRC'.
[23/10/2025, 10:00:27] zh:controller: Adding install code for 0x000a1f0012a12345.
[23/10/2025, 10:00:27] z2m: Successfully added new install code
[23/10/2025, 10:00:27] z2m:mqtt: MQTT publish: topic 'zigbee2mqtt-czcec13/bridge/response/install_code/add', payload '{"data":{"value":"THISISMYUNIQUECODE"},"status":"ok","transaction":"r2rc4-3"}'
```

burmistrzak, 2026-02-06:
@Nerivec Got a Sonoff Ti-based dongle as well. Pairing Twinguard(s) always worked for me. Granted, this was before https://github.com/Koenkk/zigbee-herdsman/pull/1317 was merged, IIRC.

@itkama Ok, thanks. Please reset the Twinguard fully and ensure fresh batteries are installed.
Before making another pairing attempt, enable **debug** logging for all Z2M components.
Trying pairing once more and attach the complete log file.

itkama, 2026-02-06:
Might have found the problem:
https://community.simon42.com/t/bosch-smart-home-z2m-kanaele-erfahrungen-rauchmelder-2-twinguard/64361
https://community.home-assistant.io/t/bosch-twinguard-in-z2m-install-code-adjusted-for-missing-crc/864517

Looks like the device does not work on all possible zigbee channels. I'm using channel 18 which is not one of the channels the original bosch controller supports.

burmistrzak, 2026-02-06:
> Might have found the problem: https://community.simon42.com/t/bosch-smart-home-z2m-kanaele-erfahrungen-rauchmelder-2-twinguard/64361 [...]

@itkama Welp.. Just saw your comment. 😬
It's best practice to use ZLL channels. Some devices (especially older or uncertified ones, i.e. Twinguard) don't support all Zigbee channels.

Quoting the docs:

> Optional: Zigbee channel, changing might require re-pairing of some devices (see docs below). (Note: use a ZLL channel: 11, 15, 20, or 25 to avoid problems)
    # (default: 11)

Depending on your network size, it might be sensible to change your channel.

raoulbhatia, 2026-02-06:
> Looking at the threads above, I wasn't the first person to buy it because the wiki says it works. Should I open a PR to add a line about the channels?

Yes please!  I still follow this thread from my previous issue and would have appreciated such information!

burmistrzak, 2026-02-06:
@itkama Absolutely! And thanks for confirming. 👌

dannypv05261, 2026-02-15:
I also get the same missing CRC error like @itkama when I paste install code. Not sure if it doesn't support channel 20 as well. It would be great if anyone can help me take a look. Thanks.

Zigbee2mqtt version: 2.8.0
Hardware: CC2652P7, firmware 20250321, name: UZG-01
Channel: 20

I follow the doc of  Bosch 8750001213 to join device to z2m. Not sure if anything I got missing
1. Screen QR code and get the install code with QR read of my Pixel default camera apps (RB...... with 91 characters as log below)
2. Hold Bosch Twinguard's button on top few seconds to enter into pairing mode (yellow light keeps flashing constantly)
3. Go to Z2M Settings" --> "Tools" and click on "Add install code", paste the code (check there is no spcae as well), and click OK.

Below is the log after adding install code
```
[2026-02-15 14:45:41] [32minfo[39m: 	zh:controller: Install code was adjusted for reason 'missing CRC'.
[2026-02-15 14:45:41] [32minfo[39m: 	zh:controller: Adding install code for 0x000d6f0019b9bacc.
[2026-02-15 14:45:41] [32minfo[39m: 	z2m: Successfully added new install code
[2026-02-15 14:45:41] [32minfo[39m: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt_ugz01/bridge/response/install_code/add', payload '{"data":{"value":"RB01SG0D836591B3CC0010000000000000000000000D6F0019B9BACCDLKA3EB588A3D32CB94CE8FD98EF4F92E0A"},"status":"ok","transaction":"srycm-4"}'
```

dannypv05261, 2026-02-21:
> [@dannypv05261](https://github.com/dannypv05261) were you able to successfully pair in the end? I just got a new Twinguard and the pairing keeps failing. z2m thinks the interview is complete, but the device keeps blinking its orange light for pairing mode, and eventually just goes to a red light and then goes offline.

Yes, I paired successfully with below steps. Plus, Step 4 enable permit join to pair the device.

> z2m. Not sure if anything I got missing
> 
> 1. Screen QR code and get the install code with QR read of my Pixel default camera apps (RB...... with 91 characters as log below)
> 2. Hold Bosch Twinguard's button on top few seconds to enter into pairing mode (orange light keeps flashing constantly)
> 3. Go to Z2M Settings" --> "Tools" and click on "Add install code", paste the code (check there is no spcae as well), and click OK.