Koenkk/zigbee2mqtt #25894: Was the last_seen attribute for connected devices removed with the 2.0 update?

jrhansen, 2025-01-19:
### What happened?

After updating to Z2M 2.0.2 it seems that the last_seen attribute for the connected devices are no longer there.

Is that a bug or intentional? If the latter, why?

### Zigbee2MQTT version

2.0.2

### Adapter firmware version

20240710

### Adapter

Sonoff USB dongle (ZStack)

### Setup

Supervised Home Assistant on RPi4

mrmaximas, 2025-01-19:
Have you tried looking at the zigbee2mqtt log?

jrhansen, 2025-01-19:
@mrmaximas I’m not sure I follow you. What should I look for in the log files? I need the information the be in HA.

However, I realize that maybe I posted this in the wrong repository… maybe I should move it to hassio-zigbee2mqtt (?)

mrmaximas, 2025-01-19:
> hassio-zigbee2mqtt

i'm not use HA

closed 2025-03-28 by github-actions[bot]