Koenkk/zigbee2mqtt #25899: update.sh and git pull complains over configuration.yaml

Mirarkitty, 2025-01-19:
### What happened?

Running update.sh, 

[...]
error: Your local changes to the following files would be overwritten by merge:
        data/configuration.yaml
Please commit your changes or stash them before you merge.
Aborting

same with running git pull manually. Tried on two different instances.

### What did you expect to happen?

Smooth upgrade, or instructions found somewhere on how to overcome this issue. 

Can probably find a workaround, just wanted to collect this in case other people also run into this issue.

### How to reproduce it (minimal and precise)

./update.sh from 1.39.1

### Zigbee2MQTT version

1.39.1 commit: 5bdf90df upgrade to 2.0.0

### Adapter firmware version

20210708

### Adapter

N/A

### Setup

Running on raspberry pi Raspbian GNU/Linux 10

### Debug log

pi@rpi3z1:/opt/zigbee2mqtt$ ./update.sh 
Checking Zigbee2MQTT status...
Creating backup of configuration...
Checking out changes to package-lock.json...
Updating...
remote: Enumerating objects: 2161, done.
remote: Counting objects: 100% (1383/1383), done.
remote: Compressing objects: 100% (400/400), done.
remote: Total 2161 (delta 1253), reused 983 (delta 983), pack-reused 778 (from 3)
Receiving objects: 100% (2161/2161), 1.67 MiB | 4.24 MiB/s, done.
Resolving deltas: 100% (1549/1549), completed with 83 local objects.
From https://github.com/Koenkk/zigbee2mqtt
   5bdf90df..060ae99c  master          -> origin/master
 * [new branch]        deps/all        -> origin/deps/all
   61b08c94..3d3c68b5  dev             -> origin/dev
 * [new branch]        feat/ota-update-check-all -> origin/feat/ota-update-check-all
 * [new branch]        fix/alpine.3.21 -> origin/fix/alpine.3.21
 * [new branch]        release-please--branches--dev--components--zigbee2mqtt -> origin/release-please--branches--dev--components--zigbee2mqtt
 * [new branch]        release-please--branches--dev--components--zigbee2mqtt--release-notes -> origin/release-please--branches--dev--components--zigbee2mqtt--release-notes
 * [new tag]           1.34.0          -> 1.34.0
 * [new tag]           1.35.0          -> 1.35.0
 * [new tag]           1.35.1          -> 1.35.1
 * [new tag]           1.35.2          -> 1.35.2
 * [new tag]           1.35.3          -> 1.35.3
 * [new tag]           1.36.0          -> 1.36.0
 * [new tag]           1.36.1          -> 1.36.1
 * [new tag]           1.37.0          -> 1.37.0
 * [new tag]           1.37.1          -> 1.37.1
 * [new tag]           1.38.0          -> 1.38.0
 * [new tag]           1.39.0          -> 1.39.0
 * [new tag]           1.39.1          -> 1.39.1
 * [new tag]           2.0.0           -> 2.0.0
 * [new tag]           1.40.0          -> 1.40.0
 * [new tag]           1.40.1          -> 1.40.1
 * [new tag]           1.40.2          -> 1.40.2
 * [new tag]           1.41.0          -> 1.41.0
 * [new tag]           1.42.0          -> 1.42.0
Updating 5bdf90df..060ae99c
error: Your local changes to the following files would be overwritten by merge:
        data/configuration.yaml
Please commit your changes or stash them before you merge.
Aborting
pi@rpi3z1:/opt/zigbee2mqtt$

Mirarkitty, 2025-01-19:
Possibly related: #25865

mrmaximas, 2025-01-19:
i have same issue with the package.json

Koenkk, 2025-01-19:
See the "Bare-metal installations (no Docker/Home Assistant add-on)" of https://github.com/Koenkk/zigbee2mqtt/discussions/24198

mrmaximas, 2025-01-19:
@Koenkk what to do with this messages? Please help.
sudo /Users/max/zigbee2mqtt/update.sh

Skipped stopping Zigbee2MQTT, no systemctl found
Updating...
remote: Enumerating objects: 24, done.
remote: Counting objects: 100% (23/23), done.
remote: Compressing objects: 100% (13/13), done.
remote: Total 24 (delta 12), reused 10 (delta 10), pack-reused 1 (from 1)
Unpacking objects: 100% (24/24), 108.45 KiB | 1.41 MiB/s, done.
From https://github.com/Koenkk/zigbee2mqtt
   b21167b3..8da5fa66  dev                                                    -> origin/dev
 + 4f8fc415...a41843bf release-please--branches--dev--components--zigbee2mqtt -> origin/release-please--branches--dev--components--zigbee2mqtt  (forced update)
Updating b21167b3..8da5fa66
error: Your local changes to the following files would be overwritten by merge:
	package.json
Please commit your changes or stash them before you merge.
Aborting

Koenkk, 2025-01-20:
Try `git checkout package.json`

Koenkk, 2025-01-20:
Not sure why that happens, with `pnpm i --frozen-lockfile` it shouldn't update the `package.json`

mrmaximas, 2025-01-20:
Next update with the same error
```
sudo /Users/m/zigbee2mqtt/update.sh

Skipped stopping Zigbee2MQTT, no systemctl found
Updating...
remote: Enumerating objects: 12, done.
remote: Counting objects: 100% (12/12), done.
remote: Compressing objects: 100% (9/9), done.
remote: Total 12 (delta 5), reused 5 (delta 3), pack-reused 0 (from 0)
Unpacking objects: 100% (12/12), 90.58 KiB | 1.24 MiB/s, done.
From https://github.com/Koenkk/zigbee2mqtt
   8da5fa66..2567b4c3  dev                                                    -> origin/dev
 + a41843bf...f65b8a78 release-please--branches--dev--components--zigbee2mqtt -> origin/release-please--branches--dev--components--zigbee2mqtt  (forced update)
Updating 8da5fa66..2567b4c3
error: Your local changes to the following files would be overwritten by merge:
	package.json
Please commit your changes or stash them before you merge.
Aborting
```

mrmaximas, 2025-01-21:
It might make sense to add a line 
`git checkout package.json 
`to the update.sh ?

Koenkk, 2025-01-21:
What is the output of `git diff package.json` when this happens?

mrmaximas, 2025-01-22:
nothing's happening, but I did the last update yesterday.

```
zigbee2mqtt % git diff package.json
zigbee2mqtt %
```

flummy1978, 2025-01-22:
I join the issue here, and can confirm, that i got the same problem with upgrading from 1.39 to 2.0.0

> Try `git checkout package.json`

This doesn't work for me. 

On every single Update i got the same:

```
pi@fhem:/opt/zigbee2mqtt $ ./update.sh
Checking Zigbee2MQTT status...
Creating backup of configuration...
Checking out changes to package-lock.json...
Updated 0 paths from the index
Updating...
Updating 03269267..060ae99c
error: Your local changes to the following files would be overwritten by merge:
        data/configuration.yaml
Please commit your changes or stash them before you merge.
```

I'm not sure if it is important but when i try to start the z2mqtt i get an error like: 

```
Thu 2024-06-13 15:27:36 CEST. --
Jan 22 14:20:49 fhem node[1864]:     at Znp.onPortClose (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/adapter/z-stack/znp/znp.ts:113:16)
Jan 22 14:20:49 fhem node[1864]:     at Object.onceWrapper (node:events:633:26)
Jan 22 14:20:49 fhem node[1864]:     at Socket.emit (node:events:530:35)
Jan 22 14:20:49 fhem node[1864]:     at TCP.<anonymous> (node:net:343:12)
Jan 22 14:20:49 fhem systemd[1]: zigbee2mqtt.service: Main process exited, code=exited, status=1/FAILURE
Jan 22 14:20:49 fhem systemd[1]: zigbee2mqtt.service: Failed with result 'exit-code'.
Jan 22 14:20:49 fhem systemd[1]: Stopped zigbee2mqtt.
```
After this error, it start normaly and work fine, but still in version 1.39 

Any ideas what i can do? 

I've made the same update from 1.4->2.0 on a docker - and after editing the configuration.yaml it worked fine 

kind regards
Andreas

Koenkk, 2025-01-22:
@mrmaximas so does it work now?

@flummy1978 see https://github.com/Koenkk/zigbee2mqtt/issues/25899#issuecomment-2600996989

mrmaximas, 2025-01-22:
@Koenkk not work w/o git checkout package.json
==
git diff package.json
diff --git a/package.json b/package.json
index 182e0815..980de032 100644
--- a/package.json
+++ b/package.json
@@ -98,5 +98,6 @@
     },
     "optionalDependencies": {
         "sd-notify": "^2.8.0"
-    }
+    },
+    "packageManager": "pnpm@9.15.4+sha512.b2dc20e2fc72b3e18848459b37359a32064663e5627a51e4c74b2c29dd8e8e0491483c3abb40789cfd578bf362fb6ba8261b05f0387d76792ed6e23ea3b1b6a0"
 }
==
sudo /Users/m/zigbee2mqtt/update.sh
Password:
Skipped stopping Zigbee2MQTT, no systemctl found
Updating...
Updating 2567b4c3..e9f74f09
error: Your local changes to the following files would be overwritten by merge:
	package.json
Please commit your changes or stash them before you merge.
Aborting
==
git checkout package.json
Updated 1 path from the index
==
sudo /Users/m/zigbee2mqtt/update.sh

Skipped stopping Zigbee2MQTT, no systemctl found
Updating...
Updating 2567b4c3..e9f74f09
Fast-forward
 package.json   |  2 +-
 pnpm-lock.yaml | 10 +++++-----
 2 files changed, 6 insertions(+), 6 deletions(-)
Installing dependencies...
! The local project doesn't define a 'packageManager' field. Corepack will now add one referencing pnpm@9.15.4+sha512.b2dc20e2fc72b3e18848459b37359a32064663e5627a51e4c74b2c29dd8e8e0491483c3abb40789cfd578bf362fb6ba8261b05f0387d76792ed6e23ea3b1b6a0.
! For more details about this field, consult the documentation at https://nodejs.org/api/packages.html#packagemanager

Lockfile is up to date, resolution step is skipped
Packages: +1
+
Progress: resolved 1, reused 0, downloaded 1, added 1, done

dependencies:
- zigbee-herdsman-converters 21.22.0
+ zigbee-herdsman-converters 21.23.0

Done in 1.4s
Building...

> zigbee2mqtt@2.0.0-dev build /Users/m/zigbee2mqtt
> tsc && node index.js writehash

Done!

flummy1978, 2025-01-22:
> [@flummy1978](https://github.com/flummy1978) see [#25899 (comment)](https://github.com/Koenkk/zigbee2mqtt/issues/25899#issuecomment-2600996989)

@Koenkk sorry... I dont understand what u mean there (what i should do )

Koenkk, 2025-01-23:
@mrmaximas 

When this happens: 

> error: Your local changes to the following files would be overwritten by merge:
>	package.json

what is the output of `git status; git diff`?

@flummy1978 could you provide the complete debug log? now it only shows the end.

flummy1978, 2025-01-23:
> uld you provide the complete debug log? now it only shows the end.

@Koenkk sure ... but unfortunatelly thats all whats in the log. I've tried it one more time and changed the log level to debug stopped the service, then start this log with sudo journalctl -u zigbee2mqtt.service -f > full_log1.txt && tail full_log1.txt 
then i started the service again and got this: 

```
-- Journal begins at Thu 2024-06-13 15:27:36 CEST. --
Jan 23 22:26:26 fhem node[135019]:     at Logger.log (/opt/zigbee2mqtt/lib/util/logger.ts:188:25)
Jan 23 22:26:26 fhem node[135019]:     at Logger.info (/opt/zigbee2mqtt/lib/util/logger.ts:201:14)
Jan 23 22:26:26 fhem node[135019]:     at Znp.onPortClose (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/adapter/z-stack/znp/znp.ts:113:16)
Jan 23 22:26:26 fhem node[135019]:     at Object.onceWrapper (node:events:633:26)
Jan 23 22:26:26 fhem node[135019]:     at Socket.emit (node:events:530:35)
Jan 23 22:26:26 fhem node[135019]:     at TCP.<anonymous> (node:net:343:12)
Jan 23 22:26:26 fhem systemd[1]: zigbee2mqtt.service: Main process exited, code=exited, status=1/FAILURE
Jan 23 22:26:26 fhem systemd[1]: zigbee2mqtt.service: Failed with result 'exit-code'.
Jan 23 22:26:26 fhem systemd[1]: Stopped zigbee2mqtt.
Jan 23 22:26:26 fhem systemd[1]: zigbee2mqtt.service: Consumed 32.330s CPU time.
Jan 23 22:28:19 fhem systemd[1]: Starting zigbee2mqtt...
Jan 23 22:28:19 fhem node[135678]: Starting Zigbee2MQTT without watchdog.
Jan 23 22:28:29 fhem node[135678]: [2025-01-23 22:28:29] info:         z2m: Logging to console, file (filename: log.log)
Jan 23 22:28:29 fhem node[135678]: [2025-01-23 22:28:29] debug:         z2m: Loaded state from file /opt/zigbee2mqtt/data/state.json
Jan 23 22:28:29 fhem node[135678]: [2025-01-23 22:28:29] info:         z2m: Starting Zigbee2MQTT version 1.39.0 (commit #03269267)
```

When its possible to help on other way, just let me know, what i ve to do. Sadly i m not that familiar with linux / command line

Koenkk, 2025-01-25:
@mrmaximas and we exactly do you see the package.json change? After a `git pull && pnpm install --frozen-lockfile`?

@flummy1978 you should find more in the `data/log` folder.

mrmaximas, 2025-01-25:
> pnpm install --frozen-lockfile

```
m@Homebridge zigbee2mqtt % sudo launchctl bootout system/zigbee2mqtt
Password:
m@Homebridge zigbee2mqtt % git pull
Already up to date.
m@Homebridge zigbee2mqtt % pnpm install --frozen-lockfile
✔ The modules directory at "/Users/max/zigbee2mqtt/node_modules" will be removed and reinstalled from scratch. Proceed? (Y/n) · true
Recreating /Users/m/zigbee2mqtt/node_modules
 EACCES  EACCES: permission denied, rmdir '/Users/m/zigbee2mqtt/node_modules/.bin'
m@Homebridge zigbee2mqtt % sudo pnpm install --frozen-lockfile
Lockfile is up to date, resolution step is skipped
Already up to date
Done in 506ms
m@Homebridge zigbee2mqtt % sudo launchctl bootstrap system /Library/LaunchDaemons/zigbee2mqtt.plist
m@Homebridge zigbee2mqtt %
```

mrmaximas, 2025-01-26:
@Koenkk today's log
zigbee2mqtt/update.sh

Skipped stopping Zigbee2MQTT, no systemctl found
Updating...
remote: Enumerating objects: 17, done.
remote: Counting objects: 100% (11/11), done.
remote: Compressing objects: 100% (3/3), done.
remote: Total 17 (delta 8), reused 9 (delta 8), pack-reused 6 (from 2)
Unpacking objects: 100% (17/17), 54.68 KiB | 1.16 MiB/s, done.
From https://github.com/Koenkk/zigbee2mqtt
   2af65cf2..420a9f2f  dev                                                    -> origin/dev
 + 309c55c5...d87d3352 deps/all                                               -> origin/deps/all  (forced update)
 + 6bb2803d...d1137124 release-please--branches--dev--components--zigbee2mqtt -> origin/release-please--branches--dev--components--zigbee2mqtt  (forced update)
Updating 2af65cf2..420a9f2f
error: Your local changes to the following files would be overwritten by merge:
	package.json
Please commit your changes or stash them before you merge.
Aborting

Koenkk, 2025-01-27:
@flummy1978 can you try without journalctl and just do `pnpm start` in the z2m folder?

@mrmaximas can you do the steps of [`update.sh`](https://github.com/Koenkk/zigbee2mqtt/blob/master/update.sh) 1 by 1 and see when the `package.json` get's updated? (which you can see by executing `git status`)

mrmaximas, 2025-01-27:
@Koenkk I updated a few hours ago.
git status now looks:
```
git status
On branch dev
Your branch is up to date with 'origin/dev'.

Changes not staged for commit:
  (use "git add <file>..." to update what will be committed)
  (use "git restore <file>..." to discard changes in working directory)
	modified:   package.json

no changes added to commit (use "git add" and/or "git commit -a")
```
but
```
sudo git pull --no-rebase
remote: Enumerating objects: 19, done.
remote: Counting objects: 100% (19/19), done.
remote: Compressing objects: 100% (16/16), done.
remote: Total 19 (delta 4), reused 4 (delta 3), pack-reused 0 (from 0)
Unpacking objects: 100% (19/19), 61.77 KiB | 1.87 MiB/s, done.
From https://github.com/Koenkk/zigbee2mqtt
   420a9f2f..7b199f89  dev                                                    -> origin/dev
 + d1137124...1cfb74ad release-please--branches--dev--components--zigbee2mqtt -> origin/release-please--branches--dev--components--zigbee2mqtt  (forced update)
Updating 420a9f2f..7b199f89
error: Your local changes to the following files would be overwritten by merge:
	package.json
Please commit your changes or stash them before you merge.
Aborting
```

flummy1978, 2025-01-28:
@Koenkk 
```
pi@fhem:/opt/zigbee2mqtt $ pnpm start
! The local project doesn't define a 'packageManager' field. Corepack will now add one referencing pnpm@9.15.4+sha512.b2dc20e2fc72b3e18848459b37359a32064663e5627a51e4c74b2c29dd8e8e0491483c3abb40789cfd578bf362fb6ba8261b05f0387d76792ed6e23ea3b1b6a0.
! For more details about this field, consult the documentation at https://nodejs.org/api/packages.html#packagemanager

> zigbee2mqtt@1.39.0 start /opt/zigbee2mqtt
> node index.js

```

mrmaximas, 2025-01-31:
@Koenkk todays' log:
```
% sudo /Users/m/zigbee2mqtt/update.sh

Password:
Skipped stopping Zigbee2MQTT, no systemctl found
Updating...
Updating 32329695..0706a5e0
error: Your local changes to the following files would be overwritten by merge:
	package.json
Please commit your changes or stash them before you merge.
Aborting

 % git status
On branch dev
Your branch is behind 'origin/dev' by 1 commit, and can be fast-forwarded.
  (use "git pull" to update your local branch)

Changes not staged for commit:
  (use "git add <file>..." to update what will be committed)
  (use "git restore <file>..." to discard changes in working directory)
	modified:   package.json

no changes added to commit (use "git add" and/or "git commit -a")

% git pull
Updating 32329695..0706a5e0
error: Your local changes to the following files would be overwritten by merge:
	package.json
Please commit your changes or stash them before you merge.
Aborting
 
% git checkout package.json
Updated 1 path from the index

 % sudo /Users/m/zigbee2mqtt/update.sh

Skipped stopping Zigbee2MQTT, no systemctl found
Updating...
Updating 32329695..0706a5e0
Fast-forward
 package.json   |  2 +-
 pnpm-lock.yaml | 10 +++++-----
 2 files changed, 6 insertions(+), 6 deletions(-)
Installing dependencies...
! The local project doesn't define a 'packageManager' field. Corepack will now add one referencing pnpm@9.15.4+sha512.b2dc20e2fc72b3e18848459b37359a32064663e5627a51e4c74b2c29dd8e8e0491483c3abb40789cfd578bf362fb6ba8261b05f0387d76792ed6e23ea3b1b6a0.
! For more details about this field, consult the documentation at https://nodejs.org/api/packages.html#packagemanager

Lockfile is up to date, resolution step is skipped
Packages: +1
+
Progress: resolved 1, reused 0, downloaded 1, added 1, done

dependencies:
- zigbee-herdsman-converters 21.26.0
+ zigbee-herdsman-converters 21.27.0

Done in 1.2s
Building...

> zigbee2mqtt@2.0.0-dev build /Users/m/zigbee2mqtt
> tsc && node index.js writehash

Done!

```

Koenkk, 2025-02-01:
@mrmaximas try to execute every step in `update.sh` manually to figure out when exactly the file changes.

@flummy1978 added it in https://github.com/Koenkk/zigbee2mqtt/pull/26106

mrmaximas, 2025-02-02:
@Koenkk i'm using os x
% sudo #!/bin/bash -e
sudo #/bin/bash -c "$(curl -fsSL https://raw.githubusercontent.com/Homebrew/install/HEAD/install.sh)" -e
sudo: #/bin/bash: command not found

Koenkk, 2025-02-02:
With the next update, what is the output of:

```bash
git diff
git pull --no-rebase
git diff
pnpm i --frozen-lockfile
git diff
```

mrmaximas, 2025-02-02:
@Koenkk thanks! I always run update.sh with sudo. I don't know how correct this is, but I have permission errors before. maybe I need to fix something in this. when I wait for the next dev I will run these commands first without sudo. ok?

mrmaximas, 2025-02-02:
% git diff
% git pull --no-rebase
remote: Enumerating objects: 23, done.
remote: Counting objects: 100% (15/15), done.
remote: Compressing objects: 100% (4/4), done.
remote: Total 23 (delta 11), reused 11 (delta 11), pack-reused 8 (from 1)
error: insufficient permission for adding an object to repository database .git/objects
fatal: failed to write object
fatal: unpack-objects failed
% sudo git pull --no-rebase
remote: Enumerating objects: 23, done.
remote: Counting objects: 100% (15/15), done.
remote: Compressing objects: 100% (4/4), done.
remote: Total 23 (delta 11), reused 11 (delta 11), pack-reused 8 (from 1)
Unpacking objects: 100% (23/23), 51.34 KiB | 1.05 MiB/s, done.
From https://github.com/Koenkk/zigbee2mqtt
   98b5a608..75c3c5e5  dev                                                    -> origin/dev
 * [new branch]        Koenkk-patch-1                                         -> origin/Koenkk-patch-1
   5e9d53f0..83ff4a76  master                                                 -> origin/master
 + 9656d7d4...51f7eba0 release-please--branches--dev--components--zigbee2mqtt -> origin/release-please--branches--dev--components--zigbee2mqtt  (forced update)
Updating 98b5a608..75c3c5e5
Fast-forward
 .github/workflows/release_please.yml |   2 +
 package.json                         |  10 +--
 pnpm-lock.yaml                       | 413 ++++++++++++++++++++++++++++++++++++++++++++++++++++++++++---------------------------------------------------------
 3 files changed, 214 insertions(+), 211 deletions(-)
% git diff
pnpm i --frozen-lockfile
✔ The modules directory at "/Users/m/zigbee2mqtt/node_modules" will be removed and reinstalled from scratch. Proceed? (Y/n) · true
Recreating /Users/m/zigbee2mqtt/node_modules
 EACCES  EACCES: permission denied, rmdir '/Users/m/zigbee2mqtt/node_modules/.bin'

 % sudo pnpm i --frozen-lockfile
Lockfile is up to date, resolution step is skipped
Packages: +39
+++++++++++++++++++++++++++++++++++++++
Progress: resolved 39, reused 18, downloaded 21, added 26, done

dependencies:
- semver 7.6.3
+ semver 7.7.0
- zigbee-herdsman-converters 21.27.1
+ zigbee-herdsman-converters 21.28.0

devDependencies:
- @eslint/core 0.10.0
+ @eslint/core 0.11.0
- @types/node 22.10.10
+ @types/node 22.13.0
- typescript-eslint 8.21.0
+ typescript-eslint 8.22.0

Done in 2.6s
% git diff
%

Koenkk, 2025-02-03:
So `git diff` never outputs something? Then I don't understand how the next `git pull` generates conflicts.

mrmaximas, 2025-02-03:
@Koenkk If you don't understand it, how can I understand it ;) I have much less knowledge in this.
now i run again  `git diff` - same empty answer from command.

Koenkk, 2025-02-04:
Try with `sudo pnpm install -g pnpm@10.2.0` instead

mrmaximas, 2025-02-04:
> Try with `sudo pnpm install -g pnpm@10.2.0` instead

% sudo pnpm install -g pnpm@10.2.0
 ERR_PNPM_NO_GLOBAL_BIN_DIR  Unable to find the global bin directory

Run "pnpm setup" to create it automatically, or set the global-bin-dir setting, or the PNPM_HOME env variable. The global bin directory should be in the PATH.

mrmaximas, 2025-02-04:
@Koenkk I think I found the problem. I'll tell you tomorrow.

njordan77, 2025-02-06:
i'm struggling with the same issue.....

`Update available! 9.15.4 → 10.2.0. `

trying: 
```
 sudo pnpm add -g pnpm
Nothing to stop. No server is running for the store at /root/.local/share/pnpm/store/v3
 ERR_PNPM_NO_GLOBAL_BIN_DIR  Unable to find the global bin directory
```

ANY FIX available to make the  update working?
Thankss

mrmaximas, 2025-02-07:
@flummy1978 Most of the commands in the file are unnecessary for os x. the first important is `git pull --no-rebase` with sudo in my case.

njordan77, 2025-03-01:
any update on this?

tried all commented potential solutions, but not working....

`Starting Zigbee2MQTT...
Done!
pi@SmartPi:/opt/zigbee2mqtt $ sudo pnpm add -g pnpm
Nothing to stop. No server is running for the store at /root/.local/share/pnpm/store/v3
 ERR_PNPM_NO_GLOBAL_BIN_DIR  Unable to find the global bin directory
`

njordan77, 2025-05-01:
nobody else having this issue?

pcousin1, 2025-05-06:
I’m having the same issue here, my ability is very limited, have read through what I can, but still not sure how to resolve. 
I use an rpi4, domoticz, zigbee2MQTT was installed following Linux, I didn’t use docker. 
Any help would be appreciated. 
Thank you. 

[image]

haruzept, 2025-07-06:
I have the same issue. Is there a fix or a workaround?

Checking Zigbee2MQTT status...
Creating backup of configuration...
Checking out changes to package-lock.json...
Updated 0 paths from the index
Updating...
Updating bc6eccf6..5063c7d3
error: Your local changes to the following files would be overwritten by merge:
        data/configuration.yaml
Please commit your changes or stash them before you merge.
Aborting

rallegade, 2025-07-25:
> i'm struggling with the same issue.....
>
> `Update available! 9.15.4 → 10.2.0. `
>
> trying:
>
> ```
>  sudo pnpm add -g pnpm
> Nothing to stop. No server is running for the store at [...]

I had the same issue and was able to fix it by running `corepack install`

`root@z2m:/home/pi# ./update_z2m.sh
Script is running as root. Proceeding with the tasks...
Checking for updates...
Update available!
Checking Zigbee2MQTT status...
Stopping Zigbee2MQTT...
Resetting local changes to package.json and pnpm-lock.yaml...
Updating...
Updating 5063c7d..ffdd353
Fast-forward
 .github/ISSUE_TEMPLATE/problem_report.yaml | 6 ++++++
 1 file changed, 6 insertions(+)
Installing dependencies...
Lockfile is up to date, resolution step is skipped
Already up to date

   ╭───────────────────────────────────────────────────────────────────╮
   │                                                                                                                                                                                │
   │                Update available! 9.15.4 → 10.13.1.                                                                                                        │
   │   Changelog: https://github.com/pnpm/pnpm/releases/tag/v10.13.1                                                                  │
   │                 Run "pnpm add -g pnpm" to update.                                                                                                   │
   │                                                                                                                                                                                │
   ╰───────────────────────────────────────────────────────────────────╯

Done in 4.9s
Building...

> zigbee2mqtt@2.5.1 build /opt/zigbee2mqtt
> tsc && node index.js writehash

Starting Zigbee2MQTT...
Done!
`

```
root@z2m:/home/pi/z2m# corepack install
Adding pnpm@10.12.1 to the cache...
```

closed 2025-10-01 by github-actions[bot]