Koenkk/zigbee2mqtt #25900: Ocasionally, Zigbee2mqtt restarts

arthurperezbessa, 2025-01-19:
### What happened?

My Zigbee2Mqtt sometimes restarts

socat:
enabled: false
master: pty,raw,echo=0,link=/tmp/ttyZ2M,mode=777
slave: tcp-listen:8485,keepalive,nodelay,reuseaddr,keepidle=1,keepintvl=1,keepcnt=5
options: "-d -d"
log: false

serial:
port: >-
  /dev/serial/by-id/usb-Itead_Sonoff_Zigbee_3.0_USB_Dongle_Plus_V2_d4efa90de512ef118fe66db8bf9df066-if00-port0
adapter: ember
baudrate: 115200
rtscts: false

### Zigbee2MQTT version

2.0.0-2

### Adapter firmware version

8.0.2 [GA]

### Adapter

EmberZNet

### Setup

Home Assistant on Raspberry Pi4 4GB

### Debug log

[2025-01-15 19:31:43] error: 	z2m: Publish 'set' 'state' to 'Interruptor Banheiro' failed: 'Error: ZCL command 0xa4c1382d5c6d1e64/2 genOnOff.off({}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"reservedBits":0,"writeUndiv":false}) failed ({"target":725,"apsFrame":{"profileId":260,"clusterId":6,"sourceEndpoint":1,"destinationEndpoint":2,"options":4416,"groupId":0,"sequence":44},"zclSequence":171,"commandIdentifier":11} timed out after 10000ms)'
[2025-01-15 19:31:43] error: 	z2m: Publish 'set' 'state' to 'Interruptor Quarto' failed: 'Error: ZCL command 0xa4c138755f7f93c6/1 genOnOff.off({}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"reservedBits":0,"writeUndiv":false}) failed ({"target":49281,"apsFrame":{"profileId":260,"clusterId":6,"sourceEndpoint":1,"destinationEndpoint":1,"options":4416,"groupId":0,"sequence":45},"zclSequence":172,"commandIdentifier":11} timed out after 10000ms)'
[2025-01-15 19:31:43] error: 	z2m: Publish 'set' 'state' to 'Interruptor Cozinha' failed: 'Error: ZCL command 0xa4c1384d83df3d8b/1 genOnOff.off({}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"reservedBits":0,"writeUndiv":false}) failed ({"target":42558,"apsFrame":{"profileId":260,"clusterId":6,"sourceEndpoint":1,"destinationEndpoint":1,"options":4416,"groupId":0,"sequence":49},"zclSequence":177,"commandIdentifier":11} timed out after 10000ms)'
[2025-01-15 19:31:43] error: 	z2m: Publish 'set' 'state' to 'Interruptor Serviço' failed: 'Error: ZCL command 0xa4c138d41d8700b5/1 genOnOff.off({}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"reservedBits":0,"writeUndiv":false}) failed ({"target":15457,"apsFrame":{"profileId":260,"clusterId":6,"sourceEndpoint":1,"destinationEndpoint":1,"options":4416,"groupId":0,"sequence":51},"zclSequence":179,"commandIdentifier":11} timed out after 10000ms)'
[2025-01-15 19:32:21] error: 	z2m: Publish 'set' 'state' to 'Interruptor Banheiro' failed: 'Error: ZCL command 0xa4c1382d5c6d1e64/2 genOnOff.off({}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"reservedBits":0,"writeUndiv":false}) failed ({"target":725,"apsFrame":{"profileId":260,"clusterId":6,"sourceEndpoint":1,"destinationEndpoint":2,"options":4416,"groupId":0,"sequence":67},"zclSequence":183,"commandIdentifier":11} timed out after 10000ms)'
[2025-01-15 19:32:21] error: 	z2m: Publish 'set' 'state' to 'Interruptor Cozinha' failed: 'Error: ZCL command 0xa4c1384d83df3d8b/1 genOnOff.off({}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"reservedBits":0,"writeUndiv":false}) failed ({"target":42558,"apsFrame":{"profileId":260,"clusterId":6,"sourceEndpoint":1,"destinationEndpoint":1,"options":4416,"groupId":0,"sequence":75},"zclSequence":189,"commandIdentifier":11} timed out after 10000ms)'
[2025-01-15 19:32:21] error: 	z2m: Publish 'set' 'state' to 'Interruptor Quarto' failed: 'Error: ZCL command 0xa4c138755f7f93c6/2 genOnOff.off({}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"reservedBits":0,"writeUndiv":false}) failed ({"target":49281,"apsFrame":{"profileId":260,"clusterId":6,"sourceEndpoint":1,"destinationEndpoint":2,"options":4416,"groupId":0,"sequence":76},"zclSequence":192,"commandIdentifier":11} timed out after 10000ms)'
[2025-01-15 19:36:09] error: 	z2m: Publish 'set' 'state' to 'Interruptor Banheiro' failed: 'Error: ZCL command 0xa4c1382d5c6d1e64/2 genOnOff.off({}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"reservedBits":0,"writeUndiv":false}) failed ({"target":725,"apsFrame":{"profileId":260,"clusterId":6,"sourceEndpoint":1,"destinationEndpoint":2,"options":4416,"groupId":0,"sequence":112},"zclSequence":199,"commandIdentifier":11} timed out after 10000ms)'
[2025-01-15 19:36:09] error: 	z2m: Publish 'set' 'state' to 'Interruptor Cozinha' failed: 'Error: ZCL command 0xa4c1384d83df3d8b/1 genOnOff.off({}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"reservedBits":0,"writeUndiv":false}) failed ({"target":42558,"apsFrame":{"profileId":260,"clusterId":6,"sourceEndpoint":1,"destinationEndpoint":1,"options":4416,"groupId":0,"sequence":117},"zclSequence":205,"commandIdentifier":11} timed out after 10000ms)'
[2025-01-15 19:36:09] error: 	z2m: Publish 'set' 'state' to 'Interruptor Varanda' failed: 'Error: ZCL command 0xa4c13805f5573159/1 genOnOff.off({}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"reservedBits":0,"writeUndiv":false}) failed ({"target":27414,"apsFrame":{"profileId":260,"clusterId":6,"sourceEndpoint":1,"destinationEndpoint":1,"options":4416,"groupId":0,"sequence":118},"zclSequence":206,"commandIdentifier":11} timed out after 10000ms)'
[2025-01-15 19:36:09] error: 	z2m: Publish 'set' 'state' to 'Interruptor Serviço' failed: 'Error: ZCL command 0xa4c138d41d8700b5/1 genOnOff.off({}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"reservedBits":0,"writeUndiv":false}) failed ({"target":15457,"apsFrame":{"profileId":260,"clusterId":6,"sourceEndpoint":1,"destinationEndpoint":1,"options":4416,"groupId":0,"sequence":119},"zclSequence":207,"commandIdentifier":11} timed out after 10000ms)'
[2025-01-16 05:54:50] error: 	z2m: Publish 'set' 'state' to 'Interruptor Varanda' failed: 'Error: ZCL command 0xa4c13805f5573159/1 genOnOff.off({}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"reservedBits":0,"writeUndiv":false}) failed ({"target":27414,"apsFrame":{"profileId":260,"clusterId":6,"sourceEndpoint":1,"destinationEndpoint":1,"options":4416,"groupId":0,"sequence":225},"zclSequence":14,"commandIdentifier":11} timed out after 10000ms)'
[2025-01-16 05:54:50] error: 	z2m: Publish 'set' 'state' to 'Interruptor Cozinha' failed: 'Error: ZCL command 0xa4c1384d83df3d8b/1 genOnOff.off({}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"reservedBits":0,"writeUndiv":false}) failed ({"target":42558,"apsFrame":{"profileId":260,"clusterId":6,"sourceEndpoint":1,"destinationEndpoint":1,"options":4416,"groupId":0,"sequence":226},"zclSequence":15,"commandIdentifier":11} timed out after 10000ms)'
[2025-01-16 05:54:50] error: 	z2m: Publish 'set' 'state' to 'Interruptor Serviço' failed: 'Error: ZCL command 0xa4c138d41d8700b5/1 genOnOff.off({}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"reservedBits":0,"writeUndiv":false}) failed ({"target":15457,"apsFrame":{"profileId":260,"clusterId":6,"sourceEndpoint":1,"destinationEndpoint":1,"options":4416,"groupId":0,"sequence":227},"zclSequence":16,"commandIdentifier":11} timed out after 10000ms)'
[2025-01-16 11:07:10] error: 	z2m: Publish 'set' 'state' to 'Interruptor Closet' failed: 'Error: ZCL command 0xa4c1387ca0a0afbb/1 genOnOff.off({}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"reservedBits":0,"writeUndiv":false}) failed ({"target":46168,"apsFrame":{"profileId":260,"clusterId":6,"sourceEndpoint":1,"destinationEndpoint":1,"options":4416,"groupId":0,"sequence":95},"zclSequence":73,"commandIdentifier":11} timed out after 10000ms)'
[2025-01-16 12:55:59] error: 	z2m: Publish 'set' 'state' to 'Interruptor Varanda' failed: 'Error: ZCL command 0xa4c13805f5573159/1 genOnOff.off({}, {"timeout":10000,"disableResponse":false,"disableReco [... cut ...] 7,109,34,58,49,44,34,102,114,101,113,34,58,51,56]},"msgpartcrc":207}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"reservedBits":0,"writeUndiv":false}) failed (~x~> [ZCL to=0xb43522fffe123fc2:61959 apsFrame={"profileId":260,"clusterId":60672,"sourceEndpoint":1,"destinationEndpoint":1,"options":4352,"groupId":0,"sequence":0}] Failed to send request with status=FAIL.)}
[2025-01-17 20:14:14] error: 	z2m: Adapter disconnected, stopping

bobo-cher, 2025-01-19:
> [2025-01-17 20:13:42] error: zh:ember:uart:ash: Received ERROR from adapter, with code=ERROR_EXCEEDED_MAXIMUM_ACK_TIMEOUT_COUNT.
> [2025-01-17 20:13:42] error: zh:ember:uart:ash: ASH disconnected | Adapter status: ASH_NCP_FATAL_ERROR

See #23861. There are firmwares in testing which should solve the problem or at least improve the situation. See the last 10 days of the discussion for links to firmware downloads.

Nerivec, 2025-01-21:
Here's a link to said firmware builds that should hopefully fix your troubles. Check out the linked post above for more details.
https://github.com/Nerivec/silabs-firmware-builder/releases/tag/v2024.6.2-update5

closed 2025-01-21 by Nerivec