Koenkk/zigbee2mqtt #25901: [4512772] First button press after a while is lost by Z2M - sniffer reports traffic

seime, 2025-01-19:
### What happened?

Problematic device: https://www.zigbee2mqtt.io/devices/4512772.html#namron-4512772

After device has been unused for some time (plus/minus 30 minutes, far from exact measurement), the first button press is not acted upon by Z2M. The next button press is registered, and also so apparently ones that are received shortly (seconds, minutes) after the initial press.

Attached is a wireshark capture using another Dongle-E showing 3 button presses. 

### What did you expect to happen?

An MQTT message with the button action is sent

### How to reproduce it (minimal and precise)

1. Pair a new device
2. Leave it unused for some time
3. Push a button -> no mqtt message with action
4. Push again -> mqtt message produced

### Zigbee2MQTT version

 2.0.0 commit: 060ae99cff715cc9ef6a0cfa7cae0cdcb17f0c8a

### Adapter firmware version

8.0.2 [GA] (also tested with 7.4.4)

### Adapter

EmberZNet / ZBDogle-E

### Setup

Docker, Mosquitto

### Debug log

Wireshark capture of this device
[entre.pcapng.gz](https://github.com/user-attachments/files/18469232/entre.pcapng.gz)

Z2M log
[entre.log](https://github.com/user-attachments/files/18469231/entre.log)

Koenkk, 2025-01-19:
Do you see any `Received zigbee message` in the debug log for this first button press?

datbilling, 2025-01-19:
Could be the same issue as this for Aqara devices?

[https://github.com/home-assistant/core/issues/128653](https://github.com/home-assistant/core/issues/128653)

seime, 2025-01-20:
> Do you see any `Received zigbee message` in the debug log for this first button press?

I think this is the first message; 
```
[2025-01-19 15:58:02] debug: 	z2m: Received Zigbee message from 'Bryter_6', type 'attributeReport', cluster 'genPowerCfg', data '{"batteryPercentageRemaining":200}' from endpoint 1 with groupID 0
[2025-01-19 15:58:02] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Bryter_6', payload '{"battery":100,"linkquality":144,"update":{"installed_version":26,"latest_version":26,"state":"idle"}}'
```

So battery etc, but no action. Just to be clear, it is a bit hard to get a clear log due to other traffic on the network. So I cannot guarantee that a message is  always received.

Maybe in the future the log format could be prefixed with the device MAC or similar to make it easy to filter for specific devices. Could be that there is some config for this already, did not check.

> Could be the same issue as this for Aqara devices?

Looks similar, but since I run openHAB and not HA, IMHO the common denominator is Z2M.
The earliest Z2M I've tested is 1.4.2 (this is a new installation)

I've got 13 of these button panels, and all behave the same (so presumably no faulty hardware).

Koenkk, 2025-01-20:
Can you share your network key, what is the packet no. of the first button press?

seime, 2025-01-21:
Wireshark packet filter: `zbee_nwk.src == 0x2842`
Network k: `0e:61:ac:52:9e:42:8c:12:77:92:9e:60:77:cc:e2:bb`

Koenkk, 2025-01-21:
I meant the first column in Wireshark, with `zbee_nwk.src == 0x2842` I cannot find a packet with seq number 186 (sniff contains 61 to 65)

seime, 2025-01-22:
We might have different Wireshark column configuration, so I'll just post a screenshot of mine with some annotations. You should be able to get the exact same information by opening the capture file.

I'm assuming annotation 2 and 4 are the same and what you are asking for.
I was referring to the value of annotation 3 :)

So first button press is 61 as shown in the screenshot is the first button press.

[image]

I'm sorry if I'm causing confusion or not understanding exactly what you are asking for, I'm a newbie to Zigbee protocol details.

seime, 2025-01-24:
If necessary I will try to get another packet capture, but for sure there is another identical button panel located 2m from the coordinator (no obstacles), and it has the exact same issue.

seime, 2025-01-26:
Hi @Koenkk, thanks for looking into this! I'm at the installation site and have done 2 packet captures:

All captures done at the same location (2 meter direct line of sight to controller)

Repairing device 
[entre_pair_z.pcapng.gz](https://github.com/user-attachments/files/18550561/entre_pair_z.pcapng.gz)

Long press off, then on again (a few minutes after repairing)
[entre_off_on_close_too_coordinator_2.pcapng.gz](https://github.com/user-attachments/files/18550562/entre_off_on_close_too_coordinator_2.pcapng.gz)

Waiting an hour. 2 button presses, the first one ignored.
[entre_after_some_waiting_time_2_presses_only_1_acted_upon_by_z2m_2.pcapng.gz](https://github.com/user-attachments/files/18550564/entre_after_some_waiting_time_2_presses_only_1_acted_upon_by_z2m_2.pcapng.gz)

BlairC1, 2025-01-28:
I have 5 aqara buttons and they’ve all been doing the same for a while now. I never looked into it, figured it was something on my end.

kirashogun, 2025-01-31:
Just FYI, this issue occurs in HA from 2024.10.x. Everything is fine in 2024.9.x.  Z2M 1.42.0-2 as an add-on within HA. 

i.e. Without changing Z2M version, this issue doesn't exist in HA 2024.9.3 but as soon as you upgrade to 2024.10.0 and all subsequent versions, this issue occurs.

ExquisiteMetropolis, 2025-02-14:
This feels the same as my reported issue?

https://github.com/Koenkk/zigbee2mqtt/issues/26227

seime, 2025-02-15:
@ExquisiteMetropolis no this is about button presses being lost, not "HA is not listing all possible actions for the Hue Dimmerswitch & Smartbutton."

@kirashogun This issue has nothing to do with HA, so it appears that you have a different issue?

FWIW I've also tested with 2.1.1, bug appears to still be present.

seime, 2025-04-17:
This still is a problem. I manually added bindings; that helped. Not verified that it solves everything though.

For anyone else with the problem, this is what I added: 

[image]

BlairC1, 2025-06-18:
This is still and issue but there is a workaround. It works fine  for mine if I enable legacy action sensor in Z2M and then use the action sensor as the trigger for my automations. eg.

trigger: state
entity_id:
  - sensor.bedroom_button_balcony_door_action
id: SinglePress
to: single

soonic6, 2025-08-01:
same here.
since i changed all my automations to sensor actions, because of the deprecated legacy actions.

now, every time z2m is restarted i have to use all my switches and buttons once, before my automations work.
doesn't matter which hardware i try.

seime, 2025-10-01:
Not stale.

a-roussel, 2025-10-28:
I am having the exact same issue: after some time, the first action of the remote (https://www.zigbee2mqtt.io/devices/E2213.html) is lost.

> This still is a problem. I manually added bindings; that helped. Not verified that it solves everything though.
> 
> For anyone else with the problem, this is what I added:
> 
> [image]

This unfortunately did not help, still looking for a reliable workaround.

andrei-lazarov, 2025-12-28:
I've had this problem with IKEA STYRBAR and SLZB-06M iirc. 
Re-pairing the remote fixed it.

seime, 2026-03-13:
Still a problem

soonic6, 2026-09-27:
still present.

seime, 2026-11-13:
Claude did an analysis of this issue and the packet capture: 

---

I think I've found the cause. It's in the zigbee-herdsman Ember adapter, not in the adapter firmware or the radio.

### What the panels send
The capture `entre_after_some_waiting_time_2_presses_only_1_acted_upon_by_z2m_2.pcapng` (filter `zbee_nwk.src == 0xfff6`) shows that the panel sends the button press as a **group message**: NWK dst `0xFFFD`, APS delivery mode Group, group `0xff61`, `OnOff: Off`. The two presses are identical apart from the counters (NWK frame counters 20536 → 20537, so nothing was missed in between).

- The coordinator relays both presses. That's what @Koenkk saw.
- Only for the **second** press does the coordinator send a ZCL Default Response back to the panel (frame 384). So Z2M only ever received the second press.

### What the log shows
In `entre.log`, all within the same second:
    15:58:02 ezspMessageSentHandler(): [type=MULTICAST] … "groupId":33825,"sequence":72   <- 1st preonly relayed by the NCP    15:58:02 ===> SET_MULTICAST_TABLE_ENTRY                                                             15:58:02 zh:ember: Registered multilticastId":33825,…}    15:58:03 ezspIncomingMessageHandler(): [type=MULTICAST] … groupId 33825, sequence 74     <- 2nd press, delivered -> action                                                                                                    ### WhyIn `src/adapter/ember/adapter/emberAdapter.ts`, in the `SLStatus.OK` branch of the message-sent hand(the "workaround for devices using mult                                                                                                    - The NCP only passes a groupcast up toin its multicast table.- At startup, only the fixed groups from `endpoints.ts` are registered (`0`, `901`, `0x0b84`).      - Any other group is added only *after*elayed a message to it. That report hasno source address and can't be turned into a ZCL message, so **the first groupcast to each group is always dropped**.
- The learned groups are kept only in memory (`this.multicastTable = []` in the constructor). **After every Z2M restart, every remote loses iat matches the comments above about
having to press each switch once after a restart, and explains why this happens with many different
remotes and adapters.

### Workaround (confirmed)
Bind the remote's OnOff, LevelCtrl and Scenes clusters, on every endpoint, to the **Coordinator**. The
remote then sends to the coordinator diup. I've done this on all 13 of my
4512772 panels. After a Z2M restart, the first press is received.

### Suggested fix
- At startup, register in the NCP multi known devices are likely to send to:
Z2M groups, groups in device binding tables, and groups learned at runtime, saved to the database so they
survive a restart.
- Set `MULTICAST_TABLE_SIZE` explicitly to something larger than the firmware default, and log a clear
warning when the table is full. Right neps growing (`tableIdx =
this.multicastTable.length`).

### Still open
This explains the lost first press afte confirmed whether a press can also belost after a long idle period *without* a restart. Once a group has been learned it should stay learned. If someone sees that, a debug log showiticast table entry` or `Failed to
register group` appears at that moment would help.