Koenkk/zigbee2mqtt #25902: Tuya sensors not in "breaking changes" list missing illuminance after upgrade to 2.0.0-2

Lurker00, 2025-01-19:
### What happened?

I have two Tuya sensors with illuminance:
1. Human presence sensor
TS0601/_TZE204_e5m9c5hl
https://www.zigbee2mqtt.io/devices/WZ-M100-W.html#wenzhi-wz-m100-w

2. PIR 24Ghz human presence sensor
TS0601/_TZE200_kb5noeto
https://www.zigbee2mqtt.io/devices/ZG-204ZM.html#tuya-zg-204zm

Both are working fine with 1.42.0-2. Both are not listed in the ["breaking changes" for illuminance](https://github.com/Koenkk/zigbee2mqtt/discussions/24198). After the upgrade, both have "illuminance" attribute "null", and they are showed as Unavailable in Home Assistant. The status of a working device is:

```
{
    "detection_delay": 0,
    "fading_time": 45,
    "illuminance_lux": 5,
    "linkquality": 127,
    "maximum_range": 2.8,
    "minimum_range": 0.3,
    "presence": true,
    "radar_sensitivity": 3,
    "target_distance": 1.68
}
```
Its status after the upgrade is like:
```
{
    "detection_delay": 0,
    "fading_time": 45,
    "linkquality": 127,
    "maximum_range": 2.8,
    "minimum_range": 0.3,
    "presence": true,
    "radar_sensitivity": 3,
    "target_distance": 1.68
    "illuminance": null,
}
```
Note, that `illuminance_lux` -> `illuminance` and moved to the end, with always `null` value.

I found no new entities to enable, as written in the "breaking changes", and people reports around.

Restoring just Z2M to 1.42.0-2 didn't help: the sensors remain unavailable. Restoring to full nightly backup resolved the issue.

### What did you expect to happen?

Illuminance sensors working as before upgrade. Possibly via enable/diable of entities, as described in the "breaking change", but working and providing measurements.

### Zigbee2MQTT version

2.0.0-2

### Adapter firmware version

20240316

### Adapter

zStack3x0, HG POE Plus

### Setup

Add-on in Home Assistant OS under ProxMox

jsschmid, 2025-02-06:
same problem with ZG-204ZL

BeFygo, 2025-03-30:
This is still so for the wz-m100 (_TZE200_clrdrnya) it seems that the illuminance value is mapped to  'Interval time'

Koenkk, 2025-03-30:
Could you provide the debug log when the device should report `illuminance`?

BeFygo, 2025-03-30:
Unfortunately I cannot, the issue I had I fixed it just a moment ago with using `https://github.com/wzwenzhi/Wenzhi-ZigBee2mqtt/blob/main/mtg035_250205ext.js`

Koenkk, 2025-03-30:
Integrated that!

Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now and in the next release which is every 1st of the month.

closed 2025-03-30 by Koenkk

kingp0dd, 2025-04-03:
I have the same device, but this is the external converter that worked for me before 2.0

I don't know where to put it now, is it here:

`/homeassistant/zigbee2mqtt/external_converters/cntop_zigbee_sensor.js`

```
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const legacy = require('zigbee-herdsman-converters/lib/legacy');
const extend = require('zigbee-herdsman-converters/lib/extend');
const ota = require('zigbee-herdsman-converters/lib/ota');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const utils = require('zigbee-herdsman-converters/lib/utils');
const globalStore = require('zigbee-herdsman-converters/lib/store');
const e = exposes.presets;
const ea = exposes.access;

const tzDatapoints = {
    ...tuya.tz.datapoints,
    key: [...tuya.tz.datapoints.key,  'alarm', 'alarm_time', 'alarm_volume', 'type', 'volume', 'ringtone', 'duration', 'medium_motion_detection_distance',
            'large_motion_detection_distance', 'large_motion_detection_sensitivity', 'small_motion_detection_distance',
            'small_motion_detection_sensitivity', 'static_detection_distance', 'static_detection_sensitivity', 'keep_time', 'indicator',
            'motion_sensitivity', 'detection_distance_max', 'detection_distance_min', 'presence_sensitivity', 'sensitivity', 'illuminance_interval',
            'medium_motion_detection_sensitivity', 'small_detection_distance', 'small_detection_sensitivity',]
}

const definition = [

    {
        fingerprint: tuya.fingerprint('TS0225', ['_TZE200_hl0ss9oa']),
        model: 'ZG-205ZL',
        vendor: 'TuYa',
        description: '24Ghz human presence sensor',
        fromZigbee: [tuya.fz.datapoints],
        toZigbee: [tzDatapoints],
        exposes: [
            e.presence(),
            e.enum('motion_state', ea.STATE, ['none', 'large', 'small', 'static']).withDescription('Motion state'),
            e.illuminance_lux(),
            e.numeric('fading_time', ea.STATE_SET).withValueMin(0).withValueMax(600).withValueStep(1).withUnit('s')
                .withDescription('Presence keep time'),
            e.numeric('large_motion_detection_distance', ea.STATE_SET).withValueMin(0).withValueMax(10).withValueStep(0.01).withUnit('m')
                .withDescription('Large motion detection distance'),
            e.numeric('large_motion_detection_sensitivity', ea.STATE_SET).withValueMin(0).withValueMax(10).withValueStep(1).withUnit('x')
                .withDescription('Large motion detection sensitivity'),
            e.numeric('small_motion_detection_distance', ea.STATE_SET).withValueMin(0).withValueMax(6).withValueStep(0.01).withUnit('m')
                .withDescription('Small motion detection distance'),
            e.numeric('small_motion_detection_sensitivity', ea.STATE_SET).withValueMin(0).withValueMax(10).withValueStep(1).withUnit('x')
                .withDescription('Small motion detection sensitivity'),
            e.numeric('static_detection_distance', ea.STATE_SET).withValueMin(0).withValueMax(6).withValueStep(0.01).withUnit('m')
                .withDescription('Static detection distance'),
            e.numeric('static_detection_sensitivity', ea.STATE_SET).withValueMin(0).withValueMax(10).withValueStep(1).withUnit('x')
                .withDescription('Static detection sensitivity'),
            e.enum('mode', ea.STATE_SET, ['off', 'arm', 'alarm']).withDescription('Alarm mode'),
            e.enum('alarm_volume', ea.STATE_SET, ['mute', 'low', 'medium', 'high']).withDescription('Alarm volume'),
            e.numeric('alarm_time', ea.STATE_SET).withValueMin(1).withValueMax(60).withValueStep(1).withUnit('m').withDescription('Alarm time'),
            e.binary('indicator', ea.STATE_SET, 'ON', 'OFF').withDescription('LED indicator mode'),
        ],
        meta: {
            tuyaDatapoints: [
                [1, 'presence', tuya.valueConverter.trueFalse1],
                [20, 'illuminance_lux', tuya.valueConverter.raw],
                [11, 'motion_state', tuya.valueConverterBasic.lookup({
                    'none': tuya.enum(0), 'large': tuya.enum(1), 'small': tuya.enum(2), 'static': tuya.enum(3),
                })],
                [12, 'fading_time', tuya.valueConverter.raw],
                [13, 'large_motion_detection_distance', tuya.valueConverter.divideBy100],
[... 206 lines cut ...]

module.exports = definition;
```

Zigbee Model
TS0601
Zigbee Manufacturer
_TZE200_kb5noeto
Description
PIR 24Ghz human presence sensor
Support status
Supported

IEEE Address
0xa4c1387b390c8a69
Network address
0x86E0 / 34528
Firmware version
0122052017
Manufacturer
[Tuya](https://www.zigbee2mqtt.io/supported-devices/#v=Tuya)
Model
[ZG-204ZM](https://www.zigbee2mqtt.io/devices/ZG-204ZM.html#tuya-zg-204zm)

```

BeFygo, 2025-04-03:
> I have the same device, but this is the external converter that worked for me before 2.0
>
> I don't know where to put it now, is it here: [...]

The external_converters setting is no longer used. Instead all external converters inside 
data/external_converters directory are now automatically loaded. Make sure to move all your external converters to this directory.

External extensions are now loaded from data/external_extensions instead of data/extension. Make sure to rename your data/extension directory (if present) to data/external_extensions.