Koenkk/zigbee2mqtt #25904: [Wrong device]: CK-BL702-AL-01(7009_Z102LG03-1) recognized as TS0505B

ETE-Design, 2025-01-19:
### Link

https://www.aliexpress.com/item/1005007014259661.html

### Model

CK-BL702-AL-01(7009_Z102LG03-1)

### Description

Zigbee LED bulb

### Vendor

eWeLink

### Picture (link)

https://www.zigbee2mqtt.io/images/devices/CK-BL702-AL-01.png

### Database entry

{"id":3,"type":"Router","ieeeAddr":"0xa4c1386b763216ef","nwkAddr":5546,"manufId":4417,"manufName":"_TZ3210_bfwvfyx1","powerSource":"DC Source","modelId":"TS0505B","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":269,"inClusterList":[0,3,4,5,6,8,768,4096,61184],"outClusterList":[10,25],"clusters":{"genBasic":{"attributes":{}},"lightingColorCtrl":{"attributes":{"57344":0,"61447":[],"61448":[],"61449":[],"61453":88,"colorCapabilities":29,"tuyaRgbMode":0}},"genOnOff":{"attributes":{"onOff":1}},"genLevelCtrl":{"attributes":{"61440":149}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":162,"stackVersion":2,"hwVersion":1,"swBuildId":"0122052017","zclVersion":3,"interviewCompleted":true,"meta":{"configured":-585015193},"lastSeen":1737308898498}

### Notes

I have two eWeLink Zigbee LED bulb - One is found as it should and the other is found as an TS0505B.

yaminahacs, 2025-01-25:
I have same problem, this is eWeLink led bulb but shows as Tuya bulb

Underctrl2020, 2025-02-01:
I have same issue @ETE-Design . I bought 5 bulbs and 4 are discovered as CK-BL702-AL-01 and one as TS0505B_4_1. The box is completely same, even the bulb.

ETE-Design, 2025-02-09:
@Koenkk Can you help?

Underctrl2020, 2025-02-10:
I have reached out to seller to see whether these are indeed two different types.

Underctrl2020, 2025-02-13:
It seems they are indeed two different devices, despite having same body and same box, but I haven't got exact confirmation. Maybe some chip inside is different. I noticed they both behave slightly differently.

ETE-Design, 2025-02-13:
> It seems they are indeed two different devices, despite having same body and same box, but I haven't got exact confirmation. Maybe some chip inside is different. I noticed they both behave slightly differently.

They should behave equal since they are the same devices from eWeLink... So seems like the chip should just be added to the eWeLink - But I have no knowledge to do it :-(

ETE-Design, 2025-04-15:
@Koenkk Can you help with this so it is recognised as the right device?

Koenkk, 2025-04-16:
Fixed!

Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now and in the next release which is every 1st of the month.

closed 2025-04-16 by Koenkk

ESP4Ever, 2025-07-13:
Hello!
This bug fix just killed the older implementation for the bulbs with same ID! As I already mentioned earlier somewhere, the fingerprint function is badly needed one more field "FW version" because even if the ID is the same, different FW makes them incompatible.

This code was working relatively fine with my bulbs:
```

        fingerprint: tuya.fingerprint("TS0505B", ["_TZ3210_bfwvfyx1"]),
        model: "TS0505B_4",
        vendor: "Tuya",
        description: "Zigbee RGB+CCT light",
        whiteLabel: [tuya.whitelabel("Tuya", "TS0505B_4_1", "Zigbee 3.0 15W led light bulb E27 RGBCW", ["_TZ3210_bfwvfyx1"])],
        toZigbee: [tz.on_off, tzLocal.led_control, tuya.tz.do_not_disturb],
        fromZigbee: [fz.on_off, fz.tuya_led_controller, fz.brightness, fz.ignore_basic_report],
        exposes: [e.light_brightness_colortemp_colorhs([50, 1000]).removeFeature("color_temp_startup"), tuya.exposes.doNotDisturb()],
        configure: (device, coordinatorEndpoint) => {
            device.getEndpoint(1).saveClusterAttributeKeyValue("lightingColorCtrl", {
                colorCapabilities: 29,
            });
        },
```

while the new one doesn't:

```
        fingerprint: tuya.fingerprint("TS0505B", ["_TZ3210_bfwvfyx1"]),
        model: "CK-BL702-AL-01_1",
        vendor: "eWeLink",
        description: "Zigbee 3.0 15W led light bulb E27 RGBCW",
        toZigbee: [tuya.tz.do_not_disturb],
        extend: [m.light({colorTemp: {range: [50, 500], startup: false}, effect: false, powerOnBehavior: false, color: {modes: ["xy", "hs"]}})],
        exposes: [tuya.exposes.doNotDisturb()],
        configure: (device, coordinatorEndpoint) => {
            device.getEndpoint(1).saveClusterAttributeKeyValue("lightingColorCtrl", {colorCapabilities: 29});
        },
```