Koenkk/zigbee2mqtt #25905: [Wrong device]: ZG-101Z/D recognized as TS004F - _TZ3000_abrsvsou

ETE-Design, 2025-01-19:
### Link

https://www.aliexpress.com/item/1005007549384535.html

### Model

ZG-101Z/D

### Description

Smart Knob Dimmer

### Vendor

Tuya

### Picture (link)

https://5.imimg.com/data5/SELLER/Default/2024/12/477444349/CD/CA/BY/88125121/cw-1000x1000.webp

### Database entry

{"id":5,"type":"EndDevice","ieeeAddr":"0xa4c138e22f43ae6b","nwkAddr":52701,"manufId":4742,"manufName":"_TZ3000_abrsvsou","powerSource":"Battery","modelId":"TS004F","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":260,"inClusterList":[0,1,3,4,6,4096],"outClusterList":[25,10,3,4,5,6,8,4096],"clusters":{"genBasic":{"attributes":{}},"genOnOff":{"attributes":{"tuyaOperationMode":1}},"genPowerCfg":{"attributes":{"batteryPercentageRemaining":200,"batteryVoltage":30}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":145,"stackVersion":2,"hwVersion":1,"swBuildId":"0122052017","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1737307931357}

ETE-Design, 2025-02-09:
@Koenkk Can you help here?

Ibraheem277, 2025-04-13:
I have the issue but it's the other way around 

[image]

Mr-Beat, 2025-04-21:
> It seems this device has lost the possibility of change mode from command to event as it was capable in previous versions (I don't know what exactly version was previously when it was working as expected). Is there any way to bring back this possibility? Currently is working only in command mode.

I can confirm this. I use the “Tuya ZigBee Smart Knob” device. I bought it here: https://de.aliexpress.com/item/1005006765315209.html

Until now, this switch was always recognized as "Tuya ERS-10TZBVK-AA". Details: https://www.zigbee2mqtt.io/devices/ERS-10TZBVK-AA.html

Now the switch is recognized as "Tuya ZG-101Z/D" and many functions are missing. https://www.zigbee2mqtt.io/devices/ZG-101Z_D.html

For example, the following are no longer available: action, action_step_size, action_transition_time, action_rate, battery, operation_mode.

How can I get the old functions back?

Translated with [DeepL.com](https://www.deepl.com/?utm_campaign=product&utm_source=web_translator&utm_medium=web&utm_content=copy_free_translation) (free version)

ETE-Design, 2025-04-21:
> It seems this device has lost the possibility of change mode from command to event as it was capable in previous versions (I don't know what exactly version was previously when it was working as expected). Is there any way to bring back this possibility? Currently is working only in command mode.

These functions need to be implemented in the "New Device"
I don't know how to do it sadly :-(

ETE-Design, 2025-04-21:
> I can confirm this. I use the “Tuya ZigBee Smart Knob” device. I bought it here: https://de.aliexpress.com/item/1005006765315209.html
>
> Until now, this switch was always recognized as "Tuya [...]

Someone need to implement these functions to the "New" Knop as it was with the wrong recognized device... I'm not up for that task since I don't know how to do it :-(

josteinh, 2025-04-22:
> I have the issue but it's the other way around
> 
> [image]

I have the same problem. Two devices previously configured as ERS-10TZBVK-AA, suddenly stopped working.
I now see they are recogniced as ZG-101Z/D. (They are not working.)

Have tried reconfigure, removing/re-adding, but no success...

mikimause9, 2025-04-25:
> I can confirm this. I use the “Tuya ZigBee Smart Knob” device. I bought it here: https://de.aliexpress.com/item/1005006765315209.html
>
> Until now, this switch was always recognized as "Tuya [...]

I have the exact same issue. Device previously recognised as knoob (ERS-10TZBVK-AA) now gets recognised as a "poorer" knoob (ZG-101Z_D). What's more confusing is that now, for me, no functionality of the device works except battery percentage and lq. I see no state change when clicking/turning, etc.

JensvanZutphen, 2025-05-02:
did Zigbee2MQTT 2.3.0 fix this? im wondering if i can update.

djrm05, 2025-05-02:
I've just upgraded and yes, this is now working as before! thanks

closed 2025-05-02 by ETE-Design

paulomac1000, 2025-05-06:
It seems that the issue still persists. When I set the device to "action" operation mode, I can see the values for Action Step Size and Action Transition Time. However, when using the "action" or "event" mode, the "action" property does not appear in homeassistant. The value is visible in zigbee2mqtt, but it is not being sent to Hass.io.

For example, here's the state of the device in zigbee2mqtt:

``` json
{
    "battery": 78,
    "linkquality": 51,
    "operation_mode": "event",
    "voltage": 2800,
    "action_rate": null,
    "action_step_size": null,
    "action_transition_time": null,
    "action": "double"
}
```
Could you please check on your end if you're experiencing a similar issue? Any insights would be appreciated.

[image]