Koenkk/zigbee2mqtt #25906: Tuya barometric sensor needs support (Z2M not supported yet)

ftz19, 2025-01-19:
### Link

https://de.aliexpress.com/item/1005007783768055.html?spm=a2g0o.productlist.main.27.48eb37d6oNc9YJ&algo_pvid=b8a2e53a-0c43-4ba1-a759-2c145392cfca&algo_exp_id=b8a2e53a-0c43-4ba1-a759-2c145392cfca-13&pdp_npi=4%40dis%21EUR%2128.60%2120.59%21%21%2128.79%2120.73%21%40211b65de17373265052111317e03d6%2112000042189578730%21sea%21DE%216170548212%21X&curPageLogUid=FxFtlitLMrSm&utparam-url=scene%3Asearch%7Cquery_from%3A

### Database entry

{"id":46,"type":"Router","ieeeAddr":"0xa4c1380cc7d6592b","nwkAddr":2779,"manufId":4417,"manufName":"_TZE204_w2vunxzm","powerSource":"Mains (single phase)","modelId":"TS0601","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[4,5,61184,0],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"65503":"Q5 /iR5 /i","65506":56,"65508":0,"65534":0,"stackVersion":0,"dateCode":"","appVersion":74,"manufacturerName":"_TZE204_w2vunxzm","modelId":"TS0601","powerSource":1,"zclVersion":3,"hwVersion":1}}},"binds":[{"cluster":0,"type":"endpoint","deviceIeeeAddress":"0x00212effff0e567e","endpointID":1}],"configuredReportings":[{"cluster":0,"attrId":4,"minRepIntval":60,"maxRepIntval":3600,"repChange":0}],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":74,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":332242049},"lastSeen":1737327034843}

### Zigbee2MQTT version

2.0.0-2

### Comments

New Tuya device. For barometric and temprature messurement.

### External definition

```shell
const definition = {
    zigbeeModel: ['TS0601'],
    model: 'TS0601',
    vendor: '_TZE204_w2vunxzm',
    description: 'Automatically generated definition',
    extend: [],
    meta: {},
};

module.exports = definition;
```

### What does/doesn't work with the external definition?

New device, is not recognized.

nmaster2042, 2025-01-22:
Have this device too, need support.

lujuan43, 2025-01-28:
Have this device too, need support.

closed 2025-04-06 by github-actions[bot]